epics-base/epics-base #400: RTEMS Posix has large thread stack sizes

kiwichris, 2023-07-06:
Inspecting the stack size of an RTEMS POSIX build I see:
```
(gdb) info thread
  Id   Target Id                                                                                                   Frame 
  1    Thread 1.167837711 (DHCP (0a01000f), priority(c:254 r:254), stack(s: 16384 a:0x109118e0), state(WK:SUSP))   _User_extensions_Thread_switch (executing=0x0, heir=0x10732510 <_RTEMS_tasks_Objects+22512>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
* 2    Thread 1.184614913 (     (0b010001), priority(c:254 r:254), stack(s: 65536 a:0x10742de0), state(CV))        rsrv_init () at ../rsrv/caservertask.c:475
  3    Thread 1.184614915 (     (0b010003), priority(c:229 r:229), stack(s:524288 a:0x107699a0), state(SEM:SUSP))  _User_extensions_Thread_switch (executing=0x0, heir=0x1071ec60 <_POSIX_Threads_Objects+3216>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  4    Thread 1.184614916 (     (0b010004), priority(c:229 r:229), stack(s:524288 a:0x11428360), state(SEM:SUSP))  _User_extensions_Thread_switch (executing=0x0, heir=0x1071f2a8 <_POSIX_Threads_Objects+4824>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  5    Thread 1.184614917 (     (0b010005), priority(c: 77 r: 77), stack(s:1048576 a:0x114a9a10), state(SEM:SUSP)) _User_extensions_Thread_switch (executing=0x0, heir=0x1071f8f0 <_POSIX_Threads_Objects+6432>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  6    Thread 1.184614918 (     (0b010006), priority(c:105 r:105), stack(s:2097152 a:0x115ae550), state(SEM:SUSP)) _User_extensions_Thread_switch (executing=0x0, heir=0x1071ff38 <_POSIX_Threads_Objects+8040>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  7    Thread 1.184614919 (     (0b010007), priority(c: 93 r: 93), stack(s:2097152 a:0x117b3110), state(SEM:SUSP)) _User_extensions_Thread_switch (executing=0x0, heir=0x10720580 <_POSIX_Threads_Objects+9648>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  8    Thread 1.184614920 (     (0b010008), priority(c: 75 r: 75), stack(s:2097152 a:0x119b7cd0), state(SEM:SUSP)) _User_extensions_Thread_switch (executing=0x0, heir=0x10720bc8 <_POSIX_Threads_Objects+11256>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  9    Thread 1.184614921 (     (0b010009), priority(c:128 r:128), stack(s:2097152 a:0x11bb8bb0), state(SEM:SUSP)) _User_extensions_Thread_switch (executing=0x0, heir=0x10721210 <_POSIX_Threads_Objects+12864>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  10   Thread 1.184614922 (     (0b01000a), priority(c: 85 r: 85), stack(s:2097152 a:0x11dcf200), state(SEM:SUSP)) _User_extensions_Thread_switch (executing=0x0, heir=0x10721858 <_POSIX_Threads_Objects+14472>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  11   Thread 1.184614923 (     (0b01000b), priority(c:103 r:103), stack(s:2097152 a:0x11fd01f0), state(SEM:SUSP)) _User_extensions_Thread_switch (executing=0x0, heir=0x10721ea0 <_POSIX_Threads_Objects+16080>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  12   Thread 1.184614924 (     (0b01000c), priority(c:100 r:100), stack(s:2097152 a:0x121d0cf0), state(SEM:SUSP)) _User_extensions_Thread_switch (executing=0x0, heir=0x107224e8 <_POSIX_Threads_Objects+17688>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  13   Thread 1.184614925 (     (0b01000d), priority(c: 98 r: 98), stack(s:2097152 a:0x123d17f0), state(SUSP))     _User_extensions_Thread_switch (executing=0x0, heir=0x10722b30 <_POSIX_Threads_Objects+19296>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  14   Thread 1.184614926 (     (0b01000e), priority(c: 95 r: 95), stack(s:2097152 a:0x125d22f0), state(SUSP))     _User_extensions_Thread_switch (executing=0x0, heir=0x10723178 <_POSIX_Threads_Objects+20904>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  15   Thread 1.184614927 (     (0b01000f), priority(c: 93 r: 93), stack(s:2097152 a:0x127d2df0), state(SUSP))     _User_extensions_Thread_switch (executing=0x0, heir=0x107237c0 <_POSIX_Threads_Objects+22512>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  16   Thread 1.184614928 (     (0b010010), priority(c: 90 r: 90), stack(s:2097152 a:0x129d38f0), state(SUSP))     _User_extensions_Thread_switch (executing=0x0, heir=0x10723e08 <_POSIX_Threads_Objects+24120>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  17   Thread 1.184614929 (     (0b010011), priority(c: 88 r: 88), stack(s:2097152 a:0x12bd43f0), state(SUSP))     _User_extensions_Thread_switch (executing=0x0, heir=0x10724450 <_POSIX_Threads_Objects+25728>) at ../../../cpukit/include/rtems/score/userextimpl.h:401
  ```
The stack sizes are the `s:` fields of the `stack()` data and are large. The EPICS [Score](https://github.com/epics-base/epics-base/blob/7.0/modules/libcom/src/osi/os/RTEMS-score/osdThread.c#L140) build uses smaller stacks. Is the Score stack sizes still applicable to the RTEMS POSIX build?

mdavidsaver, 2023-07-06:
wrt. the RTEMS-score stack sizes, epicsThreads tend to use `epicsThreadStackBig` in part as a result of past misadventures with troubleshooting stack overflows with c++ code running on RTEMS.

> Is the Score stack sizes still applicable to the RTEMS POSIX build?

Nope.

https://github.com/epics-base/epics-base/blob/42604fc794b6f26b60db953a230b52c1118d3039/modules/libcom/src/osi/os/RTEMS-score/osdThread.c#L143-L147

I can't immediately say whether `STACK_SIZE` is being overwritten for RTEMS-posix.  I think not.  Although what you report is off from that I calculate by a factor of 2.  This could be my mistake.

Perhaps @anjohnson or @hjunkes can say something about how this configuration is meant to be tuned.

https://github.com/epics-base/epics-base/blob/42604fc794b6f26b60db953a230b52c1118d3039/modules/libcom/src/osi/os/posix/osdThread.c#L492-L501

anjohnson, 2023-07-06:
If the size of an `epicsThreadStackBig` on RTEMS-6 is 1 or 2MB that might explain why my students were unable to run a standard EPICS example IOC on the RTEMS-MVME2100 CPU, which has 32MB of RAM. It was dying at the end of `iocInit` in a pva2pva initHook routine with an error like "c++ exception: bad_alloc" and no other explanation. Removing PVA from the build made booting possible and leaves about 13MB free. VxWorks on my MVME2100 running the example IOC leaves around 18MB free, and if I remove the PVA code I have 25MB free.

Chris' output appears to show using about 26MB total for stacks. I calculate that the Big threads should only be asking for 4*64K*4=1024KB=1MB but maybe the Posix code is doubling the number it was asked for. However that looks rather too large for an embedded IOC; the VxWorks PowerPC numbers are 8000, 12000 and 22000 bytes, and I guess the RTEMS-score values of 5000, 8000 and 11000 were picked by Eric Norum for a 68K-family CPU, probably from the VxWorks-68k numbers of 4000, 6000 and 11000. Most of the numbers for embedded systems were probably determined by looking at the high-water mark on a series of long-running production IOCs to see how much they were actually using.

I don't know how or who determined that Posix multiplier of 0x10000 = 64K, but I would definitely recommend reducing that for RTEMS-posix. @kiwichris does RTEMS or GDB have any way to report the high-water mark of the amount of stack used by a thread?

Also from examining the code Michael linked to I'm not actually finding anywhere in the EPICS build system (going back to the 3.14 branch) that sets `OSITHREAD_USE_DEFAULT_STACK=YES`, and my Linux builds don't show a `-DOSITHREAD_USE_DEFAULT_STACK` flag at all so the use of the OS default stack size has been disabled for all architectures. I found that Marty changed the default to `NO` back in 2004 (see 5c99cf24f13f72fa81077e3f4aed5890ca11dc40). I suspect that was to allow building Linux CA applications that could connect to more IOCs — we were hitting the virtual memory limit on a 32-bit CPU where Linux only lets user-space use up to half the virtual address space. The RHEL default is 8MB per stack, and that gives under 256 threads or 128 CA server connections on a 32-bit system.

kiwichris, 2023-07-06:
These stack sizes seem oversized for me and any change in the C++ compiler for RTEMS that required sizes of that magnitude should be considered a regression. There maybe some differences in usage across the architectures but this should be small. 

@anjohnson sorry, I should have mentioned I am on an `aarch64` so `sizeof(void*)` is 8.

kiwichris, 2023-07-06:
> @kiwichris does RTEMS or GDB have any way to report the high-water mark of the amount of stack used by a thread?

Not at the moment but it should be possible to add. You would need stack checking enabled which you have. Great idea, I will look into this.

[ FYI I am currently blocked on the `aarch64` `iocInit` completing because the RTEMS allocator mutex appears locked on entry to a call to `malloc_into()` via the pool status call after mounting an export with NFSv4 ]

anjohnson, 2023-07-06:
Thanks Chris, those numbers make sense now. I suggest we provide a way for the EPICS target OS configuration files to override that 0x10000 multiplier used in the `STACK_SIZE()` macro, 64K is fine for clients running on a workstation OS with virtual memory, but not for embedded systems. Something like this in osi/os/posix/osdThread.c should work, although @mdavidsaver might have other ideas:

```C
#ifndef OSITHREAD_STACK_FACTOR
#define OSITHREAD_STACK_FACTOR 0x10000
#endif

#define STACK_SIZE(f) (f * OSITHREAD_STACK_FACTOR * sizeof(void *))
```

mdavidsaver, 2023-07-07:
> ... @mdavidsaver might have other ideas:

@anjohnson Since you ask...  When the libc provides a default stack size, I have long been skeptical of our second guessing this in all cases.  So I would actually be in favor of `OSITHREAD_USE_DEFAULT_STACK` defaulting to `YES` except on RTEMS/vxworks.  And even then, maybe only for targets with eg. total system RAM is less than 128MB.

@kiwichris With RTEMS pthread, what is the default stack size?

kiwichris, 2023-07-07:
> With RTEMS pthread, what is the default stack size?

The default is the minimum which is a configuration value called [CONFIGURE_MINIMUM_POSIX_THREAD_STACK_SIZE](https://docs.rtems.org/branches/master/c-user/config/posix-api.html#configure-minimum-posix-thread-stack-size). A review of [rtems_config.c](https://github.com/epics-base/epics-base/blob/7.0/modules/libcom/RTEMS/posix/rtems_config.c) shows it is not defined by EPICS.

mdavidsaver, 2023-07-07:
> > @kiwichris does RTEMS or GDB have any way to report the high-water mark of the amount of stack used by a thread?
> 
> Not at the moment but it should be possible to add. You would need stack checking enabled which you have. Great idea, I will look into this.

Isn't this information already available from the RTEMS shell `stackuse` command?

kiwichris, 2023-07-07:
> Isn't this information already available from the RTEMS shell `stackuse` command?

Yes but you need to be running to see. RTEMS's `libdebugger` provides debugger access to the threads. Andrew would like to see the stack check data in `gdb` when debugging and I think it is a great idea.

My first post on this issue shows you the `info thread` output in `gdb`.

hjunkes, 2023-07-07:
A just booted system (MVME6100)

(0x0b010003 == 184614915)

```
> rt stackuse
                             STACK USAGE BY THREAD
ID         NAME                  LOW        HIGH       CURRENT     AVAIL   USED
0x09010001 IDLE                  0x007ff5a8 0x0080151f 0x00801250   8056   1344
0x0a010001 TIME                  0x008622a8 0x0086a22f 0x0086a120  32648   1536
0x0a010002 IRQS                  0x0086aa88 0x00872a0f 0x00872900  32648    336
0x0a010003 kqueue_ctx task       0x00883270 0x0088b1ef 0x0088b0c0  32640    368
0x0a010004 swi5: fast task       0x0088ba50 0x008939cf 0x00893890  32640    384
0x0a010005 thread taskq          0x00894230 0x0089c1af 0x0089c080  32640    368
0x0a010006 swi6: Giant tas       0x0089ca10 0x008a498f 0x008a4850  32640    384
0x0a010007 swi6: task queu       0x008a51f0 0x008ad16f 0x008ad030  32640   1600
0x0a010008 config_0              0x008ad9d0 0x008b594f 0x008b5820  32640    368
0x0a010009 swi1: netisr 0        0x008b61f0 0x008be16f 0x008be030  32640    800
0x0a01000a MVE                   0x008c5e98 0x008c7e1f 0x008c7cd0   8072   1728
0x0a01000b bufdaemon             0x008e0090 0x008e800f 0x008e7e90  32640   1296
0x0a01000c vnlru                 0x008e88d0 0x008f084f 0x008f0670  32640    544
0x0a01000d syncer                0x008f1110 0x008f908f 0x008f8ee0  32640    880
0x0a01000e softirq_0             0x008f99c0 0x0090193f 0x00901810  32640    368
0x0a01000f bufspacedaemon-       0x00902200 0x0090a17f 0x0090a020  32640    416
0x0a010010 nfscl                 0x0091a3e0 0x0092235f 0x00922150  32640   2336
0x0a010011 TNTa                  0x00922c80 0x0092abff 0x0092aad0  32640    688
0x0a010012 TNTb                  0x0092b460 0x009333df 0x009332b0  32640    688
0x0a010013 TNTc                  0x009342e8 0x0093c26f 0x0093c140  32648    688
0x0a010014 TNTd                  0x0093d198 0x0094511f 0x00944ff0  32648    688
0x0a010015 TNTe                  0x00945e30 0x0094ddaf 0x0094dc80  32640    688
0x0a010016 TNTD                  0x0094ece0 0x00950c5f 0x009509c0   8064    736
0x0b010001                       0x00801ff0 0x00811f6f 0x008110f0  65408   9840
0x0b010003                       0x00821ac8 0x00861a4f 0x008618d0 262024   1408
0x0b010004                       0x00a898a0 0x00b0981f 0x00b09670 524160    512
0x0b010005                       0x00b0a080 0x00b89fff 0x00b89e50 524160    512
0x0b010007                       0x00ca3ff0 0x00ce3f6f 0x00ce3dc0 262016    720
0x0b010008                       0x00ce47d0 0x00d6474f 0x00d645a0 524160    608
0x0b010009                       0x00d64fb0 0x00e64f2f 0x00e64dd0 1048448   1024
0x0b01000a                       0x00e65790 0x00f6570f 0x00f655b0 1048448    432
0x0b01000b                       0x00f67ec0 0x01067e3f 0x01067ce0 1048448    432
0x0b01000c                       0x010686a0 0x0116861f 0x01168490 1048448   1088
0x0b01000d                       0x01229448 0x013293cf 0x01329260 1048456    448
0x0b01000e                       0x01335d08 0x01435c8f 0x01435ad0 1048456    528
0x0b01000f                       0x01436608 0x0153658f 0x015363d0 1048456    528
0x0b010010                       0x01536f08 0x01636e8f 0x01636cd0 1048456    528
0x0b010011                       0x01637808 0x0173778f 0x017375d0 1048456    528
0x0b010012                       0x01738108 0x0183808f 0x01837ed0 1048456    992
0x0b010013                       0x01838a08 0x0193898f 0x019387d0 1048456   1072
0x0b010014                       0x01939308 0x01a3928f 0x01a390d0 1048456   1008
0x0b010015                       0x01a3ec58 0x01abebdf 0x01abe840 524168    992
0x0b010016                       0x01abf438 0x01b3f3bf 0x01b3f010 524168   1584
0x0b010017                       0x01b50038 0x01b8ffbf 0x01b8fd50 262024   1712
0x0b010018                       0x01b91f38 0x01c91ebf 0x01c918e0 1048456   3072
0x00000000 Interrupt Stack       0x007fb5a0 0x007ff51f 0x00000000  16256   1072
> epicsThreadShowAll
            NAME       EPICS ID   PTHREAD ID   OSIPRI  OSSPRI  STATE
          _main_       0x81bf10            0      0       0       OK
          errlog       0x821948    184614915     10      26       OK
      timerQueue       0xa89718    184614916     60     152       OK
      timerQueue       0x952270    184614917     60     152       OK
          taskwd       0xc75418    184614919     10      26       OK
      timerQueue       0xc40858    184614920     70     178       OK
           cbLow       0xc42890    184614921     59     150       OK
        cbMedium       0xc44928    184614922     64     162       OK
          cbHigh       0xc44ad0    184614923     71     180       OK
        dbCaLink       0xc44d10    184614924     50     127       OK
        scanOnce      0x12292c8    184614925     67     170       OK
         scan-10      0x1335b88    184614926     60     152       OK
          scan-5      0x1436488    184614927     61     155       OK
          scan-2      0x1536d88    184614928     62     157       OK
          scan-1      0x1637688    184614929     63     160       OK
        scan-0.5      0x1737f88    184614930     64     162       OK
        scan-0.2      0x1838888    184614931     65     165       OK
        scan-0.1      0x1939188    184614932     66     167       OK
         CAS-TCP      0x1a3e068    184614933     18      46       OK
         CAS-UDP      0x1a3e188    184614934     16      41       OK
      CAS-beacon      0x1a3eaa0    184614935     17      44       OK
    save_restore      0x1b91db0    184614936     20      51       OK
```

joelsherrill, 2023-07-07:
The default stack size from RTEMS should be the same for Classic API tasks and POSIX threads based on the code on the RTEMS master. I looked back to 4.11 and don't see any difference in this area. I don't remember any difference in RTEMS between Classic API tasks and POSIX thread stacks ever being needed. The default minimum is usually 4-8k based on the architecture.  _ even checked those values and only the PowerPC and MIPS appear to be 8k. The other architectures are smaller. This fragment is from the pthread attributes default structure and means use the minimum recommended for the processor architecture:

 ```
 .stackaddr       = NULL,                       /* stackaddr */
 .stacksize       = 0,                          /* stacksize -- will be adjusted to minimum */
```

You can configure the default minimum size for all tasks/threads but that doesn't appear to be the cause based on the reports. Looks like somehow EPICS just the idea of "big stack" being monstrous.

anjohnson, 2023-07-07:
@joelsherril EPICS has code for the RTEMS-score version of epicsThread that provides the stack sizes, but our Posix implementation of epicsThread uses a much larger set of sizes, and we're discovering that they are too big for low-RAM embedded systems.

@kiwichris actually I said "RTEMS or GDB" — from @hjunkes output the `stackuse` command shows the numbers so could be used to check how much stack each thread really needs.

BTW the epicsThread code has a plugin for Linux that tells the OS what our thread names are when we create them, can we do a similar thing on RTEMS so our thread names appear in the `stackuse` output too?

kiwichris, 2023-07-07:
> @kiwichris actually I said "RTEMS or GDB" — from @hjunkes output the `stackuse` command shows the numbers so could be used to check how much stack each thread really needs.

Ah sorry and yes the [stackuse](https://docs.rtems.org/branches/master/shell/rtems_specific_commands.html#stackuse) command from the shell is the one to use.

> BTW the epicsThread code has a plugin for Linux that tells the OS what our thread names are when we create them, can we do a similar thing on RTEMS so our thread names appear in the `stackuse` output too?

RTEMS supporst the "non-portable" (NP) POSIX call `pthread_setname_np`. We do not have this call documented but it is the same as the Linux one ... https://man7.org/linux/man-pages/man3/pthread_setname_np.3.html.

hjunkes, 2023-07-09:
Danke,
looks better ;-)
> rt stackuse
                             STACK USAGE BY THREAD
ID         NAME                  LOW        HIGH       CURRENT     AVAIL   USED
0x09010001 IDLE                  0x008005a8 0x0080251f 0x00802250   8056   1344
0x0a010001 TIME                  0x00863320 0x0086b29f 0x0086b190  32640   1536
0x0a010002 IRQS                  0x0086bb00 0x00873a7f 0x00873970  32640    336
0x0a010003 kqueue_ctx task       0x008842e8 0x0088c26f 0x0088c140  32648    368
0x0a010004 swi5: fast task       0x0088cac8 0x00894a4f 0x00894910  32648    384
0x0a010005 thread taskq          0x008952a8 0x0089d22f 0x0089d100  32648    368
0x0a010006 swi6: Giant tas       0x0089da88 0x008a5a0f 0x008a58d0  32648    384
0x0a010007 swi6: task queu       0x008a6268 0x008ae1ef 0x008ae0b0  32648   1600
0x0a010008 config_0              0x008aea48 0x008b69cf 0x008b68a0  32648    368
0x0a010009 swi1: netisr 0        0x008b7268 0x008bf1ef 0x008bf0b0  32648    560
0x0a01000a MVE                   0x008c6f10 0x008c8e8f 0x008c8d40   8064   1792
0x0a01000b bufdaemon             0x008e1108 0x008e908f 0x008e8f10  32648   1296
0x0a01000c vnlru                 0x008e9948 0x008f18cf 0x008f16f0  32648    544
0x0a01000d syncer                0x008f2188 0x008fa10f 0x008f9f60  32648    912
0x0a01000e softirq_0             0x008faa38 0x009029bf 0x00902890  32648    368
0x0a01000f bufspacedaemon-       0x00903278 0x0090b1ff 0x0090b0a0  32648    416
0x0a010010 nfscl                 0x0091b458 0x009233df 0x009231d0  32648   2288
0x0a010011 TNTa                  0x00923cf8 0x0092bc7f 0x0092bb50  32648    688
0x0a010012 TNTb                  0x0092c4d8 0x0093445f 0x00934330  32648    688
0x0a010013 TNTc                  0x00935360 0x0093d2df 0x0093d1b0  32640    688
0x0a010014 TNTd                  0x0093e210 0x0094618f 0x00946060  32640    688
0x0a010015 TNTe                  0x00946ea8 0x0094ee2f 0x0094ed00  32648    688
0x0a010016 TNTD                  0x0094fd58 0x00951cdf 0x00951a40   8072    736
0x0b010001                       0x00802ff0 0x00812f6f 0x008120f0  65408   9840
0x0b010003 errlog                0x00822b40 0x00862abf 0x00862940 262016   1344
0x0b010004 timerQueue            0x00a89f98 0x00b09f1f 0x00b09d70 524168    512
0x0b010005 timerQueue            0x00b0a778 0x00b8a6ff 0x00b8a550 524168    512
0x0b010007 sampleTask            0x00c0bd18 0x00c8bc9f 0x00c8bab0 524168   2144
0x0b010008 updateTask            0x00c8c600 0x00d0c57f 0x00d0c340 524160    656
0x0b010009 taskwd                0x00d894f8 0x00dc947f 0x00dc92d0 262024    768
0x0b01000a timerQueue            0x00dc9cd8 0x00e49c5f 0x00e49ab0 524168    608
0x0b01000b cbLow                 0x00e4a4b8 0x00f4a43f 0x00f4a2e0 1048456   1280
0x0b01000c cbMedium              0x00f4ac98 0x0104ac1f 0x0104aac0 1048456    432
0x0b01000d cbHigh                0x0104d3c8 0x0114d34f 0x0114d1f0 1048456    432
0x0b01000e dbCaLink              0x0114dba8 0x0124db2f 0x0124d9a0 1048456   1088
0x0b01000f scanOnce              0x0132f508 0x0142f48f 0x0142f320 1048456    448
0x0b010010 scan-10               0x0143b420 0x0153b39f 0x0153b1e0 1048448    528
0x0b010011 scan-5                0x0153bd20 0x0163bc9f 0x0163bae0 1048448    528
0x0b010012 scan-2                0x0163c620 0x0173c59f 0x0173c3e0 1048448    672
0x0b010013 scan-1                0x0173cf20 0x0183ce9f 0x0183cce0 1048448    528
0x0b010014 scan-0.5              0x0183d820 0x0193d79f 0x0193d5e0 1048448    528
0x0b010015 scan-0.2              0x0193e120 0x01a3e09f 0x01a3dee0 1048448    528
0x0b010016 scan-0.1              0x01a3ea20 0x01b3e99f 0x01b3e7e0 1048448    528
0x0b010017 CAS-TCP               0x01b44370 0x01bc42ef 0x01bc3f50 524160    992
0x0b010018 CAS-UDP               0x01bc4b50 0x01c44acf 0x01c44720 524160   1328
0x0b010019 CAS-beacon            0x01c55750 0x01c956cf 0x01c95460 262016   2112
0x0b01001a save_restore          0x01c97670 0x01d975ef 0x01d97010 1048448   6816
0x00000000 Interrupt Stack       0x007fc5a0 0x0080051f 0x00000000  16256   1072

hjunkes, 2023-08-23:
I'm just playing around with the StackSize and have lowered the StackSize for RTEMS in 
src/osi/os/posix/osdThread.c:
```
#if defined (OSITHREAD_USE_DEFAULT_STACK)
#define STACK_SIZE(f) (0)
#elif defined(_POSIX_THREAD_ATTR_STACKSIZE) && _POSIX_THREAD_ATTR_STACKSIZE > 0
#ifdef __rtems__
    #define STACK_SIZE(f) (f * 0x4000 * sizeof(void *))
#else
    #define STACK_SIZE(f) (f * 0x10000 * sizeof(void *))
#endif
    static const unsigned stackSizeTable[epicsThreadStackBig+1] = {
        STACK_SIZE(1), STACK_SIZE(2), STACK_SIZE(4)
    };
```

And that already looks reasonably neat now:
```
rizzo> rt stackuse
                             STACK USAGE BY THREAD
ID         NAME                  LOW        HIGH       CURRENT     AVAIL   USED
0x09010001 IDLE                  0x00489608 0x0048b57f 0x0048b2b0   8056   1344
0x0a010001 ntwk                  0x00bb8998 0x00bba91f 0x00bba7e0   8072   1136
0x0a010002 MVEd                  0x00bbab58 0x00bbcadf 0x00bbc970   8072    992
0x0a010003 RPCd                  0x00bbf150 0x00bc17df 0x00bc1660   9872   1392
0x0a010004 TNTa                  0x00bc4ba8 0x00bccb2f 0x00bcca00  32648    688
0x0a010005 TNTb                  0x00bccd68 0x00bd4cef 0x00bd4bc0  32648    688
0x0a010006 TNTc                  0x00bd4f28 0x00bdceaf 0x00bdcd80  32648    688
0x0a010007 TNTd                  0x00bdd0e8 0x00be506f 0x00be4f40  32648    688
0x0a010008 TNTe                  0x00be52a8 0x00bed22f 0x00bed100  32648    688
0x0a010009 TNTD                  0x00bed8d8 0x00bef85f 0x00bef6a0   8072    512
0x0b010001 _main_                0x0048b9d8 0x0049b95f 0x0049ad90  65416   6768
0x0b010003 errlog                0x004a7f48 0x004b7ecf 0x004b7d50  65416   1408
0x0b010004 timerQueue            0x00c6bc40 0x00c8bbbf 0x00c8ba10 130944    512
0x0b010005 timerQueue            0x00c8c320 0x00cac29f 0x00cac0f0 130944    512
0x0b010007 sampleTask            0x00cd4f20 0x00cf4e9f 0x00cf4cb0 130944   1312
0x0b010008 updateTask            0x01013068 0x01032fef 0x01032db0 130952    656
0x0b01000a awgport update        0x010732e0 0x0108325f 0x01082eb0  65408   1504
0x0b01000b taskwd                0x01164098 0x0117401f 0x01173e70  65416    512
0x0b01000c timerQueue            0x01174258 0x011941df 0x01194030 130952    608
0x0b01000d cbLow                 0x01194418 0x011d439f 0x011d4240 262024   1840
0x0b01000e cbMedium              0x011d45d8 0x0121455f 0x01214400 262024    432
0x0b01000f cbHigh                0x01214798 0x0125471f 0x012545c0 262024    432
0x0b010010 dbCaLink              0x01254958 0x012948df 0x01294750 262024   2336
0x0b010011 scanOnce              0x01605288 0x0164520f 0x016450a0 262024   2992
0x0b010012 scan-10               0x0165e460 0x0169e3df 0x0169e220 262016    528
0x0b010013 scan-5                0x0169e740 0x016de6bf 0x016de500 262016    528
0x0b010014 scan-2                0x016dea20 0x0171e99f 0x0171e7e0 262016    528
0x0b010015 scan-1                0x0171ed00 0x0175ec7f 0x0175eac0 262016   1264
0x0b010016 scan-0.5              0x0175efe0 0x0179ef5f 0x0179eda0 262016    528
0x0b010017 scan-0.2              0x0179f2c0 0x017df23f 0x017df080 262016    528
0x0b010018 scan-0.1              0x017df5a0 0x0181f51f 0x0181f360 262016    528
0x0b010019 CAS-TCP               0x018249c8 0x0184494f 0x01844690 130952    768
0x0b01001a CAS-UDP               0x01844b88 0x01864b0f 0x018647b0 130952   1520
0x0b01001b CAS-beacon            0x018753c8 0x0188534f 0x018850e0  65416   1392
0x0b01001c CAC-event             0x018a63f0 0x018c636f 0x018c61f0 130944   2368
0x0b01001d save_restore          0x0193cf60 0x0197cedf 0x0197c900 262016   2960
0x00000000 Interrupt Stack       0x00485600 0x0048957f 0x00000000  16256   1072
```
I also made changes in modules/libcom/RTEMS/posix/rtems_config.c:

```
#define CONFIGURE_MAXIMUM_POSIX_THREADS 1000
// #define CONFIGURE_EXTRA_TASK_STACKS         (4000 * RTEMS_MINIMUM_STACK_SIZE)
```
So, instead of an extra task stack for all threads, set the number of maximum Posix threads.

mdavidsaver, 2025-02-19:
I think this issue is still relevant, and remains unaddressed.