epics-base/epics-base #405: CVE-2017-16516 and CVE-2022-24795 of yajl

picca, 2023-07-26:
Hello, I got this bug report from Debian

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1040159

It seems indeed that your embeded version is affected. I am trying to  use the Debian system library but I saw that you made some modification to the official version which is not maintained anymore

I am wondering if we should not integrate your changed in this fork of yajl

https://github.com/jstamp/yajl

This is the repository of the Debian packaging...

This way it will be easier to fix these security issue and provide a way to compile epics-base with the system library yajl on Debian.

what is your opinion ?

mdavidsaver, 2023-08-04:
I for one am not happy that our bundled yajl has effectively become a fork, and would welcome efforts to sync up with some upstream.

That said, there is likely to be a moderate amount of work involved.  At a glance, https://github.com/jstamp/yajl does not include JSON5 support (https://github.com/epics-base/epics-base/pull/86).  Since it isn't mentioned on that PR, perhaps @anjohnson can remind me whether that code change was from some other yajl fork, or his original work?

In the short term, maybe we just pick up the specific CVE patches in https://github.com/jstamp/yajl/tree/master/debian/patches ?

anjohnson, 2023-08-04:
Unfortunately there is no newer upstream, the original author just stopped maintaining yajl and responding to issues and PRs. My PR to add JSON5 support (#86 here for epics-base, [#211](https://github.com/lloyd/yajl/pull/211) in the upstream repo) was all my own original work which we wanted for EPICS anyway. Nobody else had published a comprehensive fork when I last looked at this, so I started collecting and applying patches and PRs in my [master branch](https://github.com/anjohnson/yajl/tree/master), and that's where the version we're using in Base came from.

I started to look at the CVE patches that @picca linked to above when this came in last week and do plan to apply them here (and to my yajl master branch) if appropriate.

anjohnson, 2023-12-31:
Having studied the code and the available patches for this CVE at length, I don't believe any patch is needed for the EPICS version of yajl. The errors that the CVE's Ruby reports were giving and the subsequent crashes don't occur when I feed those strings into our yajl parser. The parser won't read beyond the end of an input string that ends with a short `\uXXXX` escape because the lexer which runs before the `yajl_string_decode()` routine rejects those inputs first. I will add the CVE's bad string as a test case to the yajl tests to demonstrate.

I have looked at yajl's handling of handling of surrogate pairs in that area though (e.g. `\uD800\uDC00` which is one way that JSON/JSON5 accepts the codepoint `U+10000`), and found the code for parsing these to be safe but incorrect as it doesn't reject bad inputs. I now have fixes that parse bad surrogate codepoints into question marks `?`, which is how yajl was already handling similar errors.

I will also look into the other issues that @mdavidsaver mentioned.