epics-base/epics-base #406: Set GNU_DIR to /usr on FreeBSD

kiwichris, 2023-07-27:
FreeSBD installs the OS released commands and tools directly under `/usr`. Packages are installed under `/usr/local`. FreeBSD maintains a clear separation between the OS release commands and packages users can install.

PR #393 provides a patch to use the OS released `cc` and `c++` compilers using `CONFIG.gnuCommon`. Using that file requires `GNU_DIR` is set to `/usr`.

The core developers rejected adding `CONFIG.clangCommon` because the contents would be almost the same and the GNU version of the same file.

anjohnson, 2023-07-27:
The recently merged commit 17ad04505e52f1d712c9073c66cfb6ae77ff47ac set `CC` and related variables to use clang, but it also set `GNU_DIR = /usr/local`. @tboegi Can you explain why you used that instead of `/usr`, do you have clang installed in a non-standard location in your filesystem?

tboegi, 2023-07-27:
@anjohnson @kiwichris kiwichris
The GNU_DIR /usr/local was what was working in my system.
I will set up a clean system in a couple of days (=weeks) an re-test

tboegi, 2023-08-07:
@anjohnson @kiwichris
The GNU_DIR /usr/local belongs to gcc (which I used to compile
before switching to clang)
However cherry-picking, 
https://github.com/epics-base/epics-base/pull/393/commits/a7258d5ab42eb8abf6e849a9ccfc02df473158c9#diff-481e7e5d59bcde509cb723c523a2a509b77d851c241f15f8ad1bd47e349545bb
from
https://github.com/epics-base/epics-base/pull/393
looks good (tested on a really fresh installed system this morning).

So I think that this issue can be closed, after https://github.com/epics-base/epics-base/pull/393
has been merged