epics-base/epics-base #411: [7.0.7][armel] wrong number of  planned tests

picca, 2023-08-02:
Here the error reported  on armel

```
Test Summary Report
-------------------
testpvif.t       (Wstat: 0 Tests: 98 Failed: 0)
  TODO passed:   40
testpvalink.t    (Wstat: 0 Tests: 15 Failed: 0)
  Parse errors: Bad plan.  You planned 20 tests but ran 15.
Files=7, Tests=358,  4 wallclock secs ( 0.22 usr  0.00 sys +  0.60 cusr  0.10 csys =  0.92 CPU)
Result: FAIL
-------------------
```

the full log is here

https://buildd.debian.org/status/fetch.php?pkg=epics-base&arch=armel&ver=7.0.7%2Bdfsg1-1&stamp=1690951975&raw=0

picca, 2023-08-02:
Here the output of the failing test

```
testpvalink.t ...... 
1..20
Warning: IOC is booting with TOP = "/<<PKGBUILDDIR>>/modules/pva2pva"
          but was built with TOP = "/<<PKGBUILDDIR>>"
############################################################################
## EPICS R7.0.7
## Rev. 2023-08-02T04:49+0000
## Rev. Date build date/time: 
############################################################################
# ==== testGet ====
ok  1 - dbGetField("target:i.VAL", 7) -> 42 == 42
ok  2 - dbGetField("src:i1.VAL", 7) -> 0 == 0
ok  3 - dbGetField("src:i1.INP", 0) -> "{"pva":"target:i"}" == "{"pva":"target:i"}"
ok  4 - dbPutField("src:i1.PROC", 7, ...) -> 0 (Success)
ok  5 - dbGetField("src:i1.VAL", 7) -> 42 == 42
ok  6 - dbPutField("src:i1.INP", 0, ...) -> 0 (Success)
ok  7 - dbGetField("src:i1.VAL", 7) -> 42 == 42
ok  8 - dbPutField("src:i1.PROC", 7, ...) -> 0 (Success)
ok  9 - dbGetField("src:i1.VAL", 7) -> 4 == 4
# ==== testPut ====
ok 10 - dbGetField("target:i2.VAL", 7) -> 43 == 43
ok 11 - dbGetField("src:o2.VAL", 7) -> 0 == 0
ok 12 - dbGetField("src:o2.OUT", 0) -> "{"pva":"target:i2"}" == "{"pva":"target:i2"}"
ok 13 - dbPutField("src:o2.VAL", 7, ...) -> 0 (Success)
ok 14 - dbGetField("target:i2.VAL", 7) -> 14 == 14
ok 15 - dbGetField("src:o2.VAL", 7) -> 14 == 14
# ==== testPutAsync ====
Failed 5/20 subtests 
```

picca, 2023-08-02:
the code is here, it seems that  ```USE_MULTILOCK`` is available on th armel arch but nothing happend during the test.

```
void testPutAsync()
{
#ifdef USE_MULTILOCK
    testDiag("==== testPutAsync ====");

    int64outRecord *trig = (int64outRecord*)testdbRecordPtr("async:trig");

    while(!dbIsLinkConnected(&trig->out))
        testqsrvWaitForLinkEvent(&trig->out);

    testMonitor* done = testMonitorCreate("async:after", DBE_VALUE, 0);

    testdbPutFieldOk("async:trig.PROC", DBF_LONG, 1);
    testMonitorWait(done);

    testdbGetFieldEqual("async:trig",  DBF_LONG, 1);
    testdbGetFieldEqual("async:slow",  DBF_LONG, 1); // pushed from async:trig                                                                                                                                                                 
    testdbGetFieldEqual("async:slow2", DBF_LONG, 2);
    testdbGetFieldEqual("async:after", DBF_LONG, 3);

#else
    testSkip(5, "Not USE_MULTILOCK");
#endif
```

picca, 2023-08-02:
> Are you aware of the EPICS Debian packaging project?
> https://github.com/epicsdeb

not at all

thanks for the link.

It seems that there is no more activity since a long time...

the dh_epics_helpers seems interesting.

Did you thought about sub-contracting the epics Debian packaging ?

It would be nice to be able to install epics out of the box on Debian/Ubuntu/etc...

Cheers

-----

ralphlange, 2023-08-02:
Long story short:
Back then, most of the packages were brought to distribution-quality level. All policies applied, manpages, even adapting the unusual EPICS directory layout to the standards. (The debhelpers were key to that.)
For many years, things were hosted at BNL [1] and there were thoughts about moving them to official Debian.

Meanwhile, in the EPICS collaboration, most of the labs that were supporting (and using) the Debian packages have moved to Red Hat based distributions, use non-Distro packaging (conda) or base their deployment on containers.
There's not enough manpower left to keep things up. During a few recent EPICS collaboration meetings, there were discussions and workshops targeting the issue, with no tangible outcome. [2, 3, 4]

[1] https://epicsdeb.bnl.gov/debian/
[2] https://indico.cern.ch/event/766611/timetable/#20190606.detailed
[3] https://indico.lightsource.ca/event/2/timetable/#b-37-future-of-epics-talks
[4] https://indico.cern.ch/event/1173788/sessions/459213/#20220920

picca, 2023-08-02:
> Long story short:
> Back then, most of the packages were brought to distribution-quality level. All
> policies applied, manpages, even adapting the unusual EPICS directory layout to
> the standards. (The [...]

Great :))

> Meanwhile, in the EPICS collaboration, most of the labs that were supporting
> (and using) the Debian packages have moved to Red Hat based distributions, use
> non-Distro packaging (conda) or base their [...]

Most of the time the container world rely on system libraries installed via yum or apt.
so there is still traction to have proper packages in order to ease the container production.

But I do not know why instituts do not take advantage of a clean integration in Distro.

Maybe sub-contracting is the best solution in your case. The amount of money is not that great... for such a big piece of software it could be a good investment.

https://www.debian.org/consultants/

Cheers

picca, 2023-08-02:
during the build there is no internet conenction. Do you think that his should  be an explaination of the failure ?

ralphlange, 2023-08-02:
BTW. There is very recent activity on epics-debhelper. See the current thread on the Tech-Talk mail exploder. [1]
The people involved are from the remaining EPICS installations that use Debian EPICS packages.

[1] https://epics.anl.gov/tech-talk/2023/msg01157.php

ralphlange, 2023-08-02:
I don't think the missing internet connection is related.

EPICS has an elaborated Make-based build system with very clearly defined supported host architectures. (See the file [.../configure/CONFIG_SITE](https://github.com/epics-base/epics-base/blob/7.0/configure/CONFIG_SITE) for the list.) Building it for any other architecture will need upstream work first. (Effort varies greatly.)

mdavidsaver, 2023-08-02:
> Meanwhile, in the EPICS collaboration, most of the labs that were supporting (and using) the Debian packages have moved to Red Hat based distributions ...

Isn't NSLS2 the only one to switch?  Most of the big US labs have long used RHEL (or at least RHEL clones).  FRIB being an exception.  I (originator of epicsdeb) still run debian on my personal systems, but have stepped away from epicsdeb.  There are still a couple of folks active there, but not enough to keep up with the volume of potential work (epicsdeb covers way more than just epics-base).

mdavidsaver, 2023-08-02:
> Here the error reported on armel

To address the original report, the `testpvif` and `testpvalink` tests are known to be flaky.  They have value to developers, but perhaps not as much for a packager.

fyi. a discussion of known problematic tests https://github.com/epics-base/epics-base/discussions/162

closed 2024-06-19 by mdavidsaver