epics-base/epics-base #414: iocsh and trailing comments

minijackson, 2023-08-07:
Found out due to this issue: https://github.com/epics-extensions/tree-sitter-epics/issues/2 and @anjohnson suggested we open an issue.

Apparently, this command line:

```bash
dbpf toto 8 # this is a comment
```

is equivalent to this:

```csh
dbpf("toto", "8", "#", "this", "is", "a", "comment")
```

You can try it yourself by running:

```bash
echo # this is a comment
# ^ this prints '#'
```

I think this is confusing, and can lead to some weird errors, if someone uses trailing comments and forgets an argument, or use a command with optional arguments:

```bash
dbpf "hello" # maybe I mistyped "dbpr", but somehow I put '#' inside hello
```

This error is also confusing because iocsh doesn't warn or error out if more arguments than needed by a command are passed.

Here are several options (suggested by @anjohnson):

- Allow trailing comments

This means the iocsh parser's comment handling needs rework. Currently it only supports '#' if it's the first non-whitespace character, maybe expanded from a macro.

This is also technically a breaking change, but I hope nobody's using this behavior to actually pass arguments.

There's also this question:

```bash
echo 8# hello
# Does this print 8
# Or does it print 8#
```

Note: Bash does not parse any comment in the line `echo 8# hello`

- Warn on extraneous arguments

This should probably be done in addition to supporting trailing comments, else we could add a lot of warnings for `.cmd` files using trailing comments.

I personally think both are good to have.

simon-ess, 2024-05-22:
One possible quick fix that seems to work is something like the following: https://github.com/epics-base/epics-base/commit/30c77c4793040cda9f9ad4d73c8f63afb280cd29

However, there is a bit of a hiccup with this in general. PV names are pretty unrestricted, so for example the following is a valid PV:
```
record(ai, "#") {}
```
But then how should we interpret the following:
```
dbpf # 3
```
?

mdavidsaver, 2024-06-19:
> record(ai, "#") {}

I would be OK with adding `#` to the restricted list in `dbRecordNameValidate()`, or for that matter adding minimum record name length of say 3 characters.

anjohnson, 2024-06-24:
We have 13 records in the database unit tests whose names are only 2 characters long, and 3 with only 1 character names. I use short record names quite frequently for experiments and wouldn't be in favor of imposing a minimum name length.

Warning on extra arguments should be possible, but omitting the warning if they start with `#` as an attempt to allow trailing comments would trigger the confusion @minijackson reported when that is combined with omitting trailing arguments. Commands defined to take `iocshArgArgv` can also accept an unlimited number of arguments (IMHO `echo` should be rewritten to use that) and I don't think we should require those commands to check for and ignore trailing comments.

I'm not so sure if this is an iocsh bug now as I was in my epics-extensions/tree-sitter-epics#2 comment.

simon-ess, 2024-06-24:
I agree with @anjohnson here. I don't want to force record names to have a minimum length, but I would be happy to consider restricting `#` as the first character of a PV; we use it _in_ records, but never at the start of one.

mdavidsaver, 2025-02-19:
> I don't want to force record names to have a minimum length

> happy to consider restricting `#` as the first character of a PV; we use it _in_ records, but never at the start of one.

Ok, this seems actionable.

(Anyone who tried `caget #special#record` would be disappointed)

anjohnson, 2025-02-19:
I'm not keen on the idea of changing the set of allowable PV names just to allow trailing comments to be properly supported by iocsh. At minimum I'd want us to ask the community if anyone has a naming convention that permits or even requires PV names to start with `#`.

`caget #special#record` becomes just `caget` with some shells, I have to quote the PV name for Bash to treat it as an argument instead of a comment. That might be a solution for iocsh, but I'm not sure if I like it.

simon-ess, 2025-02-20:
Should we send out a request to the community about whether or not anyone uses PVs whose names start with a `#`? Perhaps it isn't worth doing this just to allow trailing comments, but maybe if no one is actually using this in any naming convention it wouldn't be a bad idea to rein this freedom in _slightly_.

ralphlange, 2025-02-20:
> `caget #special#record` becomes just `caget` with some shells, I have to quote the PV name for Bash to treat it as an argument instead of a comment. That might be a solution for iocsh, but I'm not sure if I like it.

Couldn't that end up as one more case where \\# removes the special meaning of the # character?

(I am aware that our handling of escapes - especially though setting and using variables inside `iocsh` is badly broken.)