epics-base/epics-base #416: epicsErrlog does not longer print to console

dirk-zimoch, 2023-08-11:
Since the major errorlog code refactoring, errlog messages are no longer printed to the console.

I think the logic of this line is wrong:
https://github.com/epics-base/epics-base/blob/00dc55b8a2f8426e531b99b66264c6eeff1e340b/modules/libcom/src/error/errlog.c#L163

It only allows printing to the console during shutdown.

I think is was supposed to mean: `if((localEcho && isOkToBlock) || atExit)` ?
or maybe ` if((localEcho || atExit) && isOkToBlock)` ?

@mdavidsaver, your comment please?

ralphlange, 2023-08-11:
I have seen a somewhat similar behavior in tests.

If IOCs are run under python, the errlog message queue doesn't get flushed.
In my case, the tests fail because the expected output does not appear. However, when the IOC shuts down, all the queued error messages are printed.

Workaround that causes the errlog queue to be flushed:
On the console, run 'eltc' or 'errlog x'.

dirk-zimoch, 2023-08-11:
Ralph, that may something else. Usually stdout is in line mode when connected to a console and ins full buffered mode otherwise (e.g. when writing to a pipe to your python script. But stderr should always be line buffered, thus flushing automatically with every `\n`. You may try to call the IOC through something like "expect" which fakes to be a tty.

ralphlange, 2023-08-11:
The change that Michael's PR introduced is double-buffering the error output.

The logic that you are pointing at is making sure that direct synchronous writes indeed may only happen during exit (because the double-buffering is being torn down).

mdavidsaver, 2023-08-11:
Part of #113 is that calls to `errlogPrintf()` no longer block for I/O except during IOC shutdown.  My goal being to reduce the impact of logging on callers.

To that end, the errlog message buffer also contains a byte with flags.  The `ERL_*` bit masks.

https://github.com/epics-base/epics-base/blob/00dc55b8a2f8426e531b99b66264c6eeff1e340b/modules/libcom/src/error/errlog.c#L48-L57

Where `ERL_LOCALECHO` marks whether a message should be printed to the local console.  This flag is captured just below the line quoted above.

https://github.com/epics-base/epics-base/blob/00dc55b8a2f8426e531b99b66264c6eeff1e340b/modules/libcom/src/error/errlog.c#L163-L170

This flag is consumed by the errlog thread

https://github.com/epics-base/epics-base/blob/00dc55b8a2f8426e531b99b66264c6eeff1e340b/modules/libcom/src/error/errlog.c#L671-L676

mdavidsaver, 2023-08-11:
@dirk-zimoch Are you testing with 7.0.7 or the latest 7.0 branch?

#304, which fixed a possible loss of buffered messages on exit, has merged after 7.0.7.

dirk-zimoch, 2023-08-11:
I am using 7.0.7 and 7.0.6. But there may be some pre-release cherry-picks applied.

I noticed it when debugging one of our drivers (on VxWorks). It did not print some errors on the console during initialization when using 7.0.7 or 7.0.6 but it did when using 3.14.12. It uses `errlogPrintf`.  So I typed `errlogPrintf "test\n"` on the VxWorks shell ans was surprised to see nothing.

dirk-zimoch, 2023-08-11:
It show on the logger. But I would really like to see it locally too, like before.

mdavidsaver, 2023-08-11:
I can't comment about (or test with) vxworks.  For Linux I can say with confidence that I see errlog printing locally as I depend heavily on this during PVXS development.

Does your test IOC have any errlog listeners?  If so, can you try omitting/disabling these?

dirk-zimoch, 2023-08-11:
Yes, I have a listener. That's what I meant with "it shows on the logger".
I have now unset `EPICS_IOC_LOG_INET` and `EPICS_IOC_LOG_PORT` and tested again. Nothing.

dirk-zimoch, 2023-08-11:
It is late now. I can test more after the weekend...

mdavidsaver, 2023-08-13:
errlog works for me on RTEMS as well.

```
qemu> dbpf tick.OUT '{foo:{}}'
dbJLinkInit: Link type 'foo' not found
DBF_STRING:         ""
```

https://github.com/epics-base/epics-base/blob/00dc55b8a2f8426e531b99b66264c6eeff1e340b/modules/database/src/ioc/db/dbJLink.c#L264-L265

dirk-zimoch, 2023-08-14:
I tried the same on my systems and I get the error message on Linux but not on VxWorks. Neither on the native VxWorks shell nor in iocsh.

I will now check if it has anything to do with what I modified in our version of  EPICS 7.0.7. I will compile a "clean" 7.0.7 installation. We'll see...

dirk-zimoch, 2023-08-15:
I checked with a "clean" installation of tag R7.0.7 and with the current 7.0 HEAD. Both show the same behavior not to print errlogs to console.
I found that the `console` used in `errlogThread` is invalid. `fileno(console)` returns `-1`. How that when it had been initialized to `stderr`? It seems that the `stderr` which was valid during `errlogInitPvt()` is no longer valid at run-time.

In VxWorks, `stderr` is a macro that expands to `(*__stderr())`, calling a function that returns the stderr of the current thread. Each thread may have a different `stderr`.

For example, in the startup script I call 
```
errlogPrintf "Starting EPICS %s\n", EPICS
```
(This is a generic part of the script called by all IOCs and I want to log the EPICS version it uses.)
This calls `errlogInitPvt()` and  initializes `pvt.console` to `0x6e79f0`, the current `stderr`. It successfully logs to both, the console and the log server.
But when I call `errlogPrintf` later on the interactive shell, `stderr` is `0x6edad0` while `console` is still `0x6e79f0` but that one is meanwhile invalid. Probably because the shell that had executed the startup script has terminated and the new (interactive) shell has a new `stderr`.

In other words: You cannot pass `stderr` from one task to another and expect it to stay valid.

I suggest to initialize `pvt.console=NULL` and print to `console=pvt.console ? pvt.console : stderr`. this still allows to use `errlogSetConsole`. I have tested that (including temporarily logging to a file instead of stderr) and it seems to work fine.

Alternatively, don't use `FILE*` but file numbers instead. Even on VxWorks, `STDERR_FILENO` stays 2 constantly (actually using fd 0, 1 or 2 are still thread specific but get automatically mapped to the "real" file numbers used by the current thread, typically connected to the console. Thread specific stdin/out/err is a VxWorks specialty we do not find in Linux. Hence the EPICS-implementation in `epicsStdioRedirect.h`.

mdavidsaver, 2023-08-15:
> In other words: You cannot pass `stderr` from one task to another and expect it to stay valid.

Well, that is unexpected...  Since b8eec29e43b2fd41b80a4336fa33200de2859f1a `stderr` is captured by whichever thread first calls `errlogInit2()`, and then used from other threads.  Initially, many other threads through direct calls from `errlogPrintf()` and friends, and recently from one other thread.  What makes `errlogThread` special in this respect vs. one of the dbScan threads?

@anjohnson Are there vxworks configuration options effecting this behavior?

> I suggest to initialize `pvt.console=NULL`

This seems workable for `stderr`.  Although I think `errlogSetConsole() would need special case handling for both `stdout` and `stderr` to prevent surprises if eg `errlogSetConsole(stdout)` were called.

Also, Google finds an old `man iocLib` with references to `ioGlobalStdSet()` and `ioTaskStdSet()`, but these seem to apply to the file descriptors (`int` not `FILE*`).  It is also stated that FDs 0, 1, and 2 are special on vxworks.  So it may be premature to focus on `FILE*`?

dirk-zimoch, 2023-08-15:
`ioGlobalStdSet()` and `ioTaskStdSet()` change the mapping between FDs 0,1 or 2 and "real" device FDs, either globally or overwrite them for the current task/thread. There are getter functions as well. For example I find on my  vxWorks system:
```
> ioTaskStdGet(2)
value = -1 = 0xffffffff
> ioGlobalStdGet(2)
value = 3 = 0x3
```
Meaning: No local (task specific) STDERR set. User the global one, which is mapped to FD 3.

dirk-zimoch, 2023-08-15:
Fix in PR #417. Checks are running...