epics-base/epics-base #419: errSymBld is not compatible with incremental dynamic loading

dirk-zimoch, 2023-08-21:
When an IOC loads libraries containing errSymbols dynamically in its startup script, these symbols may not be resolved later.
The reason is that once `errSymBld()` has been called for the first time (e.g. from `errlogInitPvt()`), it sets a flag to never run again. 
https://github.com/epics-base/epics-base/blob/00dc55b8a2f8426e531b99b66264c6eeff1e340b/modules/libcom/src/error/errSymLib.c#L97
Thus all error symbols which are added later using `errSymbolAdd()`, e.g. from a dynamically loaded library are never hashed.

mdavidsaver, 2023-08-23:
@dirk-zimoch Can you look at https://github.com/epics-base/epics-base/pull/357 to see if that change would be sufficient for your cases?

dirk-zimoch, 2023-09-01:
Confirmed. This is my simple test program:
```
#include "drvSup.h"
#include "errSymTbl.h"
#include "epicsExport.h"

#define M_drvErrorTest       ( 1234<<16 )
#define S_drvErrorTest_error ( M_drvErrorTest | 1 )

static void drvErrorTestRegistrar()
{
    errSymbolAdd( S_drvErrorTest_error,
                "Test Error");
    errSymTestPrint(S_drvErrorTest_error);
}
epicsExportRegistrar(drvErrorTestRegistrar);
```
On my EPICS 3.14.12 IOC it prints correctly
```
module 1234 number 1 message="Test Error"
```
but on my EPICS 7.0.6 and 7.0.7 IOCs it instead printed
```
module 1234 number 1 message="Error (1234,1)"
```

Now, with #357 applied, the output is correct again.

dirk-zimoch, 2023-09-01:
This was a duplicate of #268.

mdavidsaver, 2023-09-01:
Reopening until #357 is merged.  (it currently fails with appveyor and vs2012)

closed 2024-04-02 by dirk-zimoch