epics-base/epics-base #422: caget with DBR_CTRL_* format no longer returns alarm and warning limits

AlexanderWells-diamond, 2023-09-13:
When upgrading from EPICS 3.14.12.7 to 7.0.7, there seems to be a bug where `caget` with the `DBR_CTRL_*` format codes no longer returns the values for the `Lo alarm limit`, `Lo warn limit`, `Hi warn limit`, and `Hi alarm limit` values. 

To recreate this, use this simple database file:
```
record(longout, "test:longout") {
  field(HOPR, "100")
  field(HIHI, "98")
  field(HIGH, "96")
  field(DRVH, "90")
  field(DRVL, "10")
  field(LOW, "5")
  field(LOLO, "2")
  field(LOPR, "0")
  field(VAL, "42")
  field(PINI, "YES")
}
```

Using EPICS 3.14.12.7, run `softIoc`:
```
$ softIoc -d simple.db
```

And then run `caget`:
```
$ caget -d DBR_CTRL_LONG test:longout
 
test:longout
    Native data type: DBF_LONG
    Request type:     DBR_CTRL_LONG
    Element count:    1
    Value:            42
    Status:           NO_ALARM
    Severity:         NO_ALARM
    Units:            
    Lo disp limit:           0
    Hi disp limit:         100
    Lo alarm limit:          2
    Lo warn limit:           5
    Hi warn limit:          96
    Hi alarm limit:         98
    Lo ctrl limit:          10
    Hi ctrl limit:          90
```

This returns all the expected values.

Now swap the `softIoc` executable for a `7.0.7` version and re-run `caget`:
```
$ caget -d DBR_CTRL_LONG test:longout
test:longout
    Native data type: DBF_LONG
    Request type:     DBR_CTRL_LONG
    Element count:    1
    Value:            42
    Status:           NO_ALARM
    Severity:         NO_ALARM
    Units:            
    Lo disp limit:           0
    Hi disp limit:         100
    Lo alarm limit:          0
    Lo warn limit:           0
    Hi warn limit:           0
    Hi alarm limit:          0
    Lo ctrl limit:          10
    Hi ctrl limit:          90
```

And now the `alarm` and `warning` limits are no longer returned. The values are set and can be inspected on the record, it's just that this particular request format seems to no longer return the values.

mdavidsaver, 2023-09-13:
When setting range alarms, it is also necessary to set the corresponding severities.  I don't recall this being any different with 3.14, so I can't explain why this appears to be a different with >=3.15.

```
  field(HHSV, "MAJOR")
  field(HSV, "MINOR")
  field(LSV, "MINOR")
  field(LLSV, "MAJOR")
```

AlexanderWells-diamond, 2023-09-13:
Thank you @mdavidsaver, adding those fields has fixed the `caget` for an R7.0.7 `softIoc`. 

I'm happy to accept this was a misconfiguration on my part, if you think the issue should be closed. Otherwise, it can be left open to investigate what changed?

mdavidsaver, 2023-09-13:
Look [back to 3.14](https://github.com/epics-base/epics-base/blob/612d57a9b3aa651960400f651ff93d8bdc858dc8/src/rec/longoutRecord.dbd#L86-L113), longout HSV and friends defaulted to NO_ALARM.  Is it possible that your 3.14 case has some local patch changing the defaults of the severity fields?

ralphlange, 2023-09-13:
Here's the commit that changed the behavior: ca22d508312b823b3d686206ea8d16cdda2c1c11.
It was applied to fix [LaunchPad bug #1770292](https://bugs.launchpad.net/epics-base/+bug/1770292), i.e. inconsistent behavior between record types.