epics-base/epics-base #426: Problem with CA nameserver and CA_V413 protocol

mdavidsaver, 2023-09-23:
https://epics.anl.gov/tech-talk/2023/msg01394.php

> I recently had to diagnose a problem with our local CA nameserver. We use a fork of the EPICS CA nameserver, which exhibits the same problem.
> 
> The symptom was that certain clients would receive monitor updates with element counts of zero, even for scalar PVs, when the PVs were resolved through a new build of the CA nameserver. This only occurred with clients built against R3.14.12 or newer, a nameserver built against R3.14.12 or newer, and an ioc built against R3.14.11 or older. (Note that I use the term "ioc" throughout to refer to any CA server in order to avoid confusion over which server is being described.) We identified camonitor and PyEpics as two clients that were affected. Notably caget was not.
> 
> I traced the problem to a change made in CA_V413, that allows  clients to use a zero element count in a CA_PROTO_EVENT_ADD or CA_PROTO_READ_NOTIFY messages. Prior to 413 an element count of zero in a request is supposed to always trigger an error. 413 allows a request for zero elements to which the server responds with the actual number of elements available. Using the element count of zero seems to have become the default when a client is communicating with an ioc that supports V413. Everything works fine when the client and ioc handle name resolution directly. The channel falls back to the lowest common denominator of CA minor protocol version.
> 
> The problem occurs when a CA nameserver built against a newer EPICS redirects newer clients to older iocs. The nameserver always answers the client with a message that correctly gives the ioc's address, but includes the nameserver's own protocol version. When the client then opens a channel with the ioc, the client has the false belief that the ioc has a later protocol version that it really does. This never seemed to create a problem for us until CA_V413, but when a 413 client connects to a <= 412 ioc it uses the zero element count which the ioc handles by returning updates with zero elements. (This itself seems like another bug in that for V412 and earlier an element count of zero is supposed to generate an error.)
> 
> My fix was to create patches to libca and libcas and modify the nameserver to use them. The libca patch adds a ca_host_minor_protocol(chid) function to the CA client API that allows a client to request the minor protocol version of a connected server. The client side of the CA nameserver uses it to fetch and store the minor protocol version from each ioc. The libcas patch adds an overload of pvExistReturn() that allows the server to respond to a client with both a specified network address and a specified minor protocol version number. The server side of the CA nameserver uses this to respond to clients with both the stored address and the stored protocol version. With these changes in place the problem disappears.
> 
> I know that pcas is basically dead, but the CA nameserver seems to still be in use. I can provide the patches against R3.15.9 if there is any interest in that.
> 
> --Brian Bevins
> Jefferson Lab

mdavidsaver, 2023-09-23:
Some history of servers sending CA_PROTO_VERSION over TCP.

RSRV prior to 3.14.12 (ca v12) did not send CA_PROTO_VERSION when a client connected.  And if memory serves me, there was a bug in how this was added such that the CA_PROTO_VERSION was queued but not flushed.  I can't remember now when that was fixed.

In any case, CA clients before 3.14.12 only found out the "server" version from the search reply.

Since 3.14.12, a CA client may be certain to find out the server version before the reply to any other request is received.  So eg. it will be available before any GET, PUT, or EVENT_ADD can be requested.

klemenv, 2025-03-10:
Before I realized this bug was already submitted, I posted a message on Tech Talk: https://epics.anl.gov/tech-talk/2025/msg00250.php
My findings are essentially what Michael explains, I'll paste here part of the email with the Wireshark dump for completeness:

After debugging with Wireshark, it appears that the problem lies when a CA Version 13 client connects to a V13 nameserver which redirects it to V11 IOC. In particular, the ChannelAccess V13 introduced dynamic resizing of arrays, which means that the event subscribe command requests 0 elements. The V11 IOC understands such a request as ‘send me events but no data’. This doesn’t happen when V13 client talks directly to V11 IOC, but only when going through the nameserver. I believe that is because the V11 IOC ignores the VERSION request in the subscribe command, and so the V13 clients continues to assume the remote side is V13 which it got from the nameserver. This may sound confusing, so here’s a dump from Wireshark with some of my comments inline, the 7.0.8.1 and 3.14.8.2 are the EPICS base versions that client/nameserver/IOC are compiled against:

# camonitor (base 7.0.8.1) asking NameServer (base 7.0.8.1,PCAS 4.13.3) on port 5062 about the 'TEST' PV
     15 1.296358367    192.168.1.34          192.168.1.34          CA       84     39208->5062 Version(13), Search('TEST',1), 
     20 1.328361887    192.168.1.34          192.168.1.34          CA       84     39208->5062 Version(13), Search('TEST',1), 

# NameServer (base 7.0.8.1) starts the broadcast search
     25 1.339764217    192.168.1.34          192.168.1.255         CA       84     50905->5064 Version(13), Search('TEST',1), 

# IOC (base 3.14.8.2) replies
     26 1.339837224    192.168.1.34          192.168.1.34          CA       84     5064->50905 Version(11), Search Reply(1), 

# NameServer (7.0.8.1) establishes the connection and from now on knows the 'TEST' PV
     32 1.340686305    192.168.1.34          192.168.1.34          CA       156    58972->5064 Version(13), User('klemen'), Host('homebox'), Create Request('TEST', cid=1), 
     34 1.340729527    192.168.1.34          192.168.1.34          CA       100    5064->58972 Rights(cid=1, RW), Create Reply(cid=1, sid=16), 

# camonitor (7.0.8.1) asking NameServer about the 'TEST' PV again
     42 1.392397425    192.168.1.34          192.168.1.34          CA       84     39208->5062 Version(13), Search('TEST',1), 

# NameServer (7.0.8.1) replies back pointing it to the IOC
     51 1.412974029    192.168.1.34          192.168.1.34          CA       84     5062->39208 Version(13), Search Reply(1), 

# camonitor (7.0.8.1) requests 'TEST' PV from IOC (3.14.8.2)
     56 1.413850566    192.168.1.34          192.168.1.34          CA       156    58980->5064 Version(13), User('klemen'), Host('homebox'), Create Request('TEST', cid=1), 

# IOC (3.14.8.2) replies but doesn't include a Version response
     58 1.413913529    192.168.1.34          192.168.1.34          CA       100    5064->58980 Rights(cid=1, RW), Create Reply(cid=1, sid=17), 

Linux cooked capture v1
Internet Protocol Version 4, Src: 192.168.1.34, Dst: 192.168.1.34
Transmission Control Protocol, Src Port: 5064, Dst Port: 58980, Seq: 1, Ack: 89, Len: 32
Channel Access
    Command: Rights (0x0016)
    Payload Size: 0
    Client Channel ID: 1
    Rights: RW (0x00000003)
Channel Access
    Command: Create Channel (0x0012)
    Payload Size: 0
    DBR Type: DOUBLE (6)
    Data Count: 1
    Client Channel ID: 1
    Server Channel ID: 17

# camonitor 7.0.8.1 assumes Version(13) from the search request packet #51, and specifies Data Count: 0
     61 1.414229433    192.168.1.34          192.168.1.34          CA       100    58980->5064 Event Add(sid=17, sub=1, mask=5), 

Frame 61: 100 bytes on wire (800 bits), 100 bytes captured (800 bits) on interface any, id 0
Linux cooked capture v1
Internet Protocol Version 4, Src: 192.168.1.34, Dst: 192.168.1.34
Transmission Control Protocol, Src Port: 58980, Dst Port: 5064, Seq: 89, Ack: 33, Len: 32
Channel Access
    Command: Event (0x0001)
    Payload Size: 16
    DBR Type: TIME_DOUBLE (20)
    Data Count: 0
    Subscription ID: 1
    Server Channel ID: 17
    Event Mask: 0x0005

No.     Time           Source                Destination           Protocol Length Info

# IOC (3.14.8.2) doesn't recognize Data Count: 0 from Version(13) as dynamic array, and hence sends an event with no elements
     62 1.414314283    192.168.1.34          192.168.1.34          CA       108    5064->58980 Event(sub=1, value=[]), 

Frame 62: 108 bytes on wire (864 bits), 108 bytes captured (864 bits) on interface any, id 0
Linux cooked capture v1
Internet Protocol Version 4, Src: 192.168.1.34, Dst: 192.168.1.34
Transmission Control Protocol, Src Port: 5064, Dst Port: 58980, Seq: 33, Ack: 121, Len: 40
Channel Access
    Command: Event (0x0001)
    Payload Size: 24
    DBR Type: TIME_DOUBLE (20)
    Data Count: 0
    Subscription ID: 1
    Status: ECA_NORMAL (0x00000001)
    Status: 0
    Severity: 0
    Timestamp: Sat 08 Mar 2025 10:40:26 AM EST
    Padding: 17

No.     Time           Source                Destination           Protocol Length Info

ralphlange, 2025-03-10:
One quick workaround is to put a CA Gateway in front of the old IOC(s).

In that case, the client talks to a modern server for both the name resolution and the data connection. Of course, a CA Gateway adds complexity, latency, management issues etc. etc.

anjohnson, 2025-03-10:
Since it isn't obvious, there is was a PR posted that fixes this issue on the 3.15 branch, see #425. We haven't rejected that, but were reluctant to merge it because of its need to extend the APIs, and it would be hard to test the result.

We might be persuaded to merge it into the 3.15 branch without pulling the fix into the 7.0 branch.

mdavidsaver, 2025-03-11:
I would indeed by reluctant to see #425 merged into any series.  I see allowing a name server to misrepresent server versions as a violation of the CA protocol version negotiation.  (which, yes, was broken prior to 3.14.12)

To my mind, a compliant nameserver would need to act more like a gateway by querying (and caching) the actual/downstream server versions (plural).

mdavidsaver, 2025-03-11:
> And if memory serves me, there was a bug in how this was added ([23612a7](https://github.com/epics-base/epics-base/commit/23612a7afe1c6e0a208bf4a0acecd2a5e2468380)) such that the CA_PROTO_VERSION

Looking back, my memory did in fact fail me.  23612a7afe1c6e0a208bf4a0acecd2a5e2468380 includes the fix.  Sending the version message with `rsrv_version_reply()` from `create_tcp_client()`.

The bug was that originally the version message was queued on entry to `camsgtask()`, but not flushed (`cas_commit_msg()`).  So it would not be sent until the response to some other event triggered a flush.

Circa 3.14.8.2:

https://github.com/epics-base/epics-base/blob/e9b49d233bc5b508065e54b9490a3e72f09ea60a/src/rsrv/camsgtask.c#L52-L53

Comparing with the current `rsrv_version_reply()`, I suspect that there was missing `SEND_LOCK()` as well.

klemenv, 2025-03-11:
@mdavidsaver We have some 3.14.8.2 and some 3.14.7. We also have some LabView CA IOcs that claim they're V(11).

I'm looking into patching 7.0.8.1 CA client code, to detect if server doesnt respond with a version on a newly established tcp socket, and in this case THE CLIENT SENDS version+search command first which would force old servers to respond.

klemenv, 2025-03-11:
Well that didnt go as planned. Turns out rsrv in 3.14.8.2 (maybe later too) doesn't react to the Version command, its supposed to send it automatically but as Michael notes it doesn't flush the buffer. So there's no way to force sending the version from the client side. Aside from upgrading all our IOCs, the only other option seems what @bsbevins suggests.

ralphlange, 2025-03-12:
> ... Aside from upgrading all our IOCs, the only other option seems what [@bsbevins](https://github.com/bsbevins) suggests.

Have you tried putting a CA Gateway in front?

klemenv, 2025-03-12:
I have not. But yes, that's an option too.

ralphlange, 2025-03-13:
My 2 cts...

I see the root problem being on the old servers, which don't send their version on the TCP circuit as they should.

I very much do not like the idea of adding complexity and new APIs to the new server and client, for everyone, forever, just to be able to work around that problem with the old servers in a specific setup (using the nameserver) that could be solved by putting a zero-config Gateway in front of the old servers and no code changes whatsoever.

Feels like fixing the problem on the wrong end, exposing internals through public APIs and promoting what should be a temporary fix to be a permanent part of the spec.

klemenv, 2025-03-13:
> Hence my suggestion that we merge #425
> <https://github.com/epics-base/epics-base/pull/425> onto the 3.15 branch,
> but no further (and we don't document the API changes). Gateways don't
> perform well when some PVs are large arrays, because the server side of the
> PCAS is single-threaded, so that isn't a universal solution.

To me this behavior is a bug. Workarounds like using old epics to build
against or using gateway in front of it are adding unnecessary complexity
and difficulty to maintain. But I also don't think that nameserver should
be faking its version, although I would prefer this over the other two
options.

When I'm back from vacation next week, I'm planning to try another idea to
detect a server prior to ca_v413 by not receiving version response from it.
It may be a bit ugly solution, but if we can make it to work reliebly with
or without nameserver and older servers, it would be no impact to the API.

bsbevins, 2025-03-13:
The whole idea of the nameserver is that it impersonates the real servers in response to PV search requests. In a sense, everything the nameserver returns is fake. It fakes having the PV and it fakes its address so the client will connect to the real server. This bug shows that faking all that isn't enough to guarantee CA will work correctly in a backwards compatible way. The nameserver also needs to fake its protocol version so that it matches the real server. 

The APIs already allow the nameserver to fake its address. As far as I can tell, that's purely to allow the existence of the nameserver. #425 makes the minimal addition to the APIs needed for it to also "fake" its protocol version so that it can return the actual protocol version of the real server to the client. I believe this support is necessary for the nameserver to fully function as intended. 

I understand the reluctance to add to longstanding APIs. We are very dependent on our nameserver, and for us having a nameserver that works as advertised was the cleanest solution.

mdavidsaver, 2025-03-13:
> Hence my suggestion that we merge [#425](https://github.com/epics-base/epics-base/pull/425) onto the 3.15 branch, but no further (and we don't document the API changes).

I know I'm a [broken record](https://github.com/epics-base/epics-base/pull/425#issuecomment-1732359736) sometimes, but in this case:  Why does this change need to involve epics-base at all?  PCAS is a separate module now (https://github.com/epics-modules/pcas).  This is where development of it should happen.  (and would seem to me an easier place to carry a site-specific patch)

mdavidsaver, 2025-03-13:
@klemenv

> To me this behavior is a bug. Workarounds like using old epics to build against or using gateway in front of it are adding unnecessary complexity and difficulty to maintain

Hiding ancient IOCs behind an old gateway has long been our answer to eg. site still using Base <= 3.13.  While I would like to put it off for as long as reasonably possible, I think it likely that the same will be true of 3.14 at some point.

However, I don't think we have reached that point.  I think your immediate "worst case" is having to run your nameserver(s) with a patched version of Base or the separate PCAS module.

And your real worst case is having to upgrade/replace those old IOCs.

mdavidsaver, 2025-03-13:
> ... impersonates the real servers in response to PV search requests. In a sense, everything the nameserver returns is fake ...
> ... I believe this support is necessary for the nameserver to fully function as intended.

@bsbevins I know it is tangential, but I smiled when reading this.  Software is all hacky, until it does what I need.  :)

bsbevins, 2025-03-13:
> I know I'm a [broken record](https://github.com/epics-base/epics-base/pull/425#issuecomment-1732359736) sometimes, but in this case: Why does this change need to involve epics-base at all? PCAS is a [...]

Does the PCAS module include its own version of the CA client library? The client library has to be patched too, to allow the nameserver to determine the protocol versions of real servers. The PCAS change allows the nameserver to specify a protocol version in its search replies. The PCAS part can certainly go into the PCAS module, but you need both to make the nameserver work.

anjohnson, 2025-03-14:
> Why does this change need to involve epics-base at all?

Because both PCAS and libCA are present in the 3.15 branch of Base, and changes to both are required for the name-server to be able to properly "fake" its replies?

@bsbevins No, the unbundled PCAS only contains CA server code. Question: Won't you also have to make changes to the name-server to have it call the new APIs? I don't see a PR [here](https://github.com/epics-extensions/ca-nameserver/pulls) for that, I guess you were waiting for Base to get modified first? A PR for that (marked as draft) could be useful to see what the changes would be there, assuming you have them already.

When this problem was unique to just one or two labs I was okay with requiring a fork, but since then @klemenv has spent quite a few hours (days?) investigating the problem and coming up with the same suggestions for what needs to be done to work around it. It's much easier to tell any other lab that discovers they need this to build the name-server against the tip of the 3.15 branch (or against R3.15.10 if we decided to release the result) than it would be to say they need to apply one patch to Base and another patch to PCAS and then build the name-server against that combination. That would also require those patches to be maintained.

We're unlikely to make many more changes to the 3.15 branch in the future, we've talked about dropping support for it in the Core Group meetings. Applying these changes to it signals that we still value fixes that the community develops for older code because it needs them; refusing them completely doesn't support our users very well.

ralphlange, 2025-03-14:
FWIW:
In the last 30 years, I have never worked at or for a facility that uses the CA name-server. Total PV number between 15k and 1.1M.
I know what it does. But why are some facilities forced to use it and many others aren't? Network design/architecture issues?

bsbevins, 2025-03-20:
> [@bsbevins](https://github.com/bsbevins) Question: Won't you also have to make changes to the name-server to have it call the new APIs? I don't see a PR [here](https://github.com/epics-extensions/ca-nameserver/pulls) for that, I guess you were waiting for Base to get modified first? A PR for that (marked as draft) could be useful to see what the changes would be there, assuming you have them already.

Correct. Our local nameserver forked from the standard one a while ago and has site-specific customizations. I planned to backport the changes to the standard nameserver, but I was waiting to see if the API changes needed to support it got adopted.

klemenv, 2025-03-25:
Could someone please review the proposed patch and check if it makes sense? 

Patch has been tested to work as described with IOCs based on base 3.14.8.2, 3.14.12.8 and 7.0.8.1. Further testing is in progress in bigger SNS environment.

mdavidsaver, 2025-03-31:
I expect that this issue and related PRs will be discussed at the core developer meeting this week.

For my part, I think there is a way forward by splitting #425 in two pieces.  The addition of `ca_host_minor_protocol()` to libca (on 7.0) seems reasonable to me.  The changes to PCAS should be moved to https://github.com/epics-modules/pcas .