epics-base/epics-base #428: Undocumented OUTLINK behavior change in IOCs

daykin, 2023-10-03:
The Application Developers' guide has always stated:
>Only CA is appropriate for an output link. The write to a field over channel access causes processing as specified in Channel Access Puts to Passive Scanned Records.

In older versions of base (e.g. 3.15.9), a developer who would like their output link to process the destination record, but inelegantly creates an output link with CPP instead of CA, is still somewhat bailed out, as "PP" matches. 
https://github.com/epics-base/epics-base/blob/R3.15.9/src/ioc/dbStatic/dbStaticLib.c#L2177

But, in versions with the newer dbLinkInfo struct (since commit https://github.com/epics-base/epics-base/commit/200355bc0030267431161018547180c292378fff on 7.0 branch), links are filtered according to the logic in dbParseLink, which silently negates CPP to NPP. 

I am wondering if it might be a good idea to add some verbiage in the documentation and/or a runtime warning.

mdavidsaver, 2024-06-12:
> I am wondering if it might be a good idea to add some verbiage in the documentation and/or a runtime warning.

This sounds like a good project.  Will you take it on?

ralphlange, 2024-06-12:
Well spotted! Make sure to mention that this applies only to local target links,

If the target of the OUT link is remote, it will always use CA, the `PP` property of the target field decides the processing policy and the PP/NPP link property of the OUT link is not relevant.

daykin, 2024-06-12:
I am happy to put something together. Currently on the Phoebus track at codeathon, so maybe not this week.