epics-base/epics-base #431: 3.14 make runtests does not (alway?) propagate failures

mdavidsaver, 2023-10-10:
I have observed a `make runtests` with a failing test ending with `exit(0)`, which GHA interprets as a success.  In my case, the failure is entirely expected as the test should not have been run for 3.14.  However, I am surprised that this failure was not flagged.

[image]

anjohnson, 2023-10-15:
We made quite a few changes on the 3.15 branch to significantly improve test handling, including ensuring that any failed tests result in the build failing overall — I think the best commit was c60056d4d63894c0f06e0f51114390c73b370ffc where the testFailures.pl script does the necessary work at the end of the runtests target (called from RULES_TOP).

Can you replicate that failure on the 3.15 branch?

mdavidsaver, 2023-10-16:
> Can you replicate that failure on the 3.15 branch?

No.  I see no lack of failure with 3.15.

To be clear.  While I wouldn't say no to a simple fix, I mostly want to log this issue in the hope that I won't be surprised by it again (although I likely will be...).

closed 2024-06-19 by mdavidsaver