epics-base/epics-base #441: CONFIG.gnuCommon contains questionable STATIC_LDFLAGS_* settings

ralphlange, 2023-11-20:
`CONFIG.gnuCommon` contains the following settings for static vs. shared loading:
```
STATIC_LDFLAGS_YES = -static
STATIC_LDFLAGS_NO =
```

These setting do not work as expected for GNU compilers/linkers.

Instead, the setting in `CONFIG.Common.linuxCommon` is correct.
```
STATIC_LDFLAGS_YES= -Wl,-Bstatic
STATIC_LDFLAGS_NO=
STATIC_LDLIBS_YES= -Wl,-Bdynamic
```

This was breaking a partially-static Windows build under MSYS2/MinGW64, which includes the former file but should use the configuration of the latter.

ralphlange, 2023-11-20:
> Which targets would be effected by this change?

That is a good question.

> > These settings do not work as expected for GNU compilers/linkers.
> 
> Is your expectation that `-static` would produce a fully static executable while a pair of `-Wl,-Bstatic` and `-Wl,-Bdynamic` will produce a partially static executable? (preferring a static library when resolving any `-l...` while `-Bstatic` is active)

That's exactly my use case: Embedding the low-level client library in a Device Support shared lib.
Switching forth and back on the linker line works perfectly with the two definitions used on Linux. The original definition for MinGW does not allow that technique, while using the Linux configuration works fine.

> On Linux, the recommendation against `-static` is (as I understand it) primarily deriving from details about how GLibc is implemented. Is there an equivalent recommendation for the libc used in mingw builds?

Thanks, I was not aware of that. I'm mainly interested in making my use case work: Is there another way to allow switching on the linker line? The `gnuCommon` configuration seems to not have a way to switch back to shared mode.

So: no urgency, more a request for clarification. I'm overwriting the definitions in my application anyway, to make things work for all Base versions.

mdavidsaver, 2023-11-20:
> On Linux, the recommendation against `-static` is (as I understand it) primarily deriving from details about how GLibc is implemented.  ...

fyi. from the glibc [FAQ](https://sourceware.org/glibc/wiki/FAQ#Even_statically_linked_programs_need_some_shared_libraries_which_is_not_acceptable_for_me.__What_can_I_do.3F).

> In glibc 2.27, released in 2018, support for statically linked programs that call dlopen was deprecated.  ...  Internally glibc continues to use dlopen for several major subsystems including NSS, gconv, IDN, and thread cancellation.  ...

ericonr, 2023-12-05:
> Instead, the setting in CONFIG.Common.linuxCommon is correct.

However, it being correct isn't actually enough for it to apply to Linux targets, because `gnuCommon` overrides it, not the other way around. What ends up happening is that the duplicated definitions in **some** (not all) OS+arch specific files override `gnuCommon` to the correct value: https://github.com/search?q=repo%3Aepics-base%2Fepics-base+STATIC_LDFLAGS_YES&type=code

mdavidsaver, 2024-06-12:
As I see it, the choice of `-static` vs. `-Wl,-Bstatic` is effectively toolchain specific decision.  Any "default" is historical.

What, if any, change would resolve this ticket?

ralphlange, 2024-06-12:
Well...

Either the snippet order could be changed, so that first the toolchain-specific `gnuCommon` is included, then the architecture-specific `linuxCommon`. Not sure about side effects, though. There is IMHO no valid answer to which definition is more generic and should come first.

Or the `gnuCommon` could use the definitions from `linuxCommon` (and say goodbye to `-static`). That would need verification if it works for our supported GNU toolchains on all architectures.