epics-base/epics-base #455: OS clock < EPICS epoch causes CA clients to spin

ralphlange, 2024-01-08:
_Moving Launchpad bug https://bugs.launchpad.net/epics-base/+bug/1815177 into the spotlight..._

Since at least EPICS 7.0.2:
A host system that is unsynchronized and has a system clock before 1990 (i.e., older than the EPICS epoch) sets a simple CA client, e.g. `caget` into a seemingly endless loop of printing warnings:
```
$ caget BLA
WARNING: OS Clock pre-dates the EPICS epoch!
Using 1990-01-02 00:00:00.000000 UTC
WARNING: OS Clock pre-dates the EPICS epoch!
Using 1990-01-02 00:00:00.000000 UTC
WARNING: OS Clock pre-dates the EPICS epoch!
Using 1990-01-02 00:00:00.000000 UTC
WARNING: OS Clock pre-dates the EPICS epoch!
Using 1990-01-02 00:00:00.000000 UTC
...
...
```

Running an IOC on such a system produces a similar loop of warning printouts, but the IOC seems to be running fine. Using CA command line tools from a different host (with sync'ed clock) shows that the records are alive, processing, etc.

Easily reproducible on any Linux box by disabling the time daemon, setting the system clock (e.g., `sudo date 0101000080`) and running `caget` with any PV name.

ralphlange, 2024-01-08:
On the Launchpad bug, Michael was asking...

> On 2/8/19 1:48 AM, Ralph Lange wrote:
> > Public bug reported:
> >
> > Using EPICS 7.0.2 on a Linux box with a system clock that was set to the
> > Posix epoch (NULL) by an irresponsible time synchronization daemon, use
> > of a simple CA client sends it into a fast infinite loop printing
> 
> Can you get a stack trace of where ClockTimeGetCurrent()
> is being called from (via. epicsTimeGetCurrent() )?
> 
> I would be curious to know if use of the monotonic clock would help.
> 
> https://code.launchpad.net/~epics-core/epics-base/+git/Com/+merge/361380

The monotonic clock code has been merged a long time ago. The behaviour is unchanged.

ralphlange, 2024-01-08:
Looking at the source code... https://github.com/epics-base/epics-base/blob/8998341588112e317f4f8cf9ae3930e958b323e5/modules/libcom/src/osi/osiClockTime.c#L249

The minimal invasive thing would be mimicking the VxWorks/RTEMS behavior: only print the warning once.

The loop character doesn't sound good, though.

mdavidsaver, 2024-06-11:
> The minimal invasive thing ...

Is there a minimally invasive change here?

If epicsTimeGetCurrent() is always returning 0.0, then epicsTimer is not advancing.  Or if time is underflowing, then `epicsTimeGetCurrent()` is returning some far future date, and eventually the jump backwards in time will also throw things off.

The best I can think of, for an IOC, would be to stall the whole process early until time catches up.  Or for Linux, to depend on an init system to wait for time sync (eg. [systemd-time-wait-sync.service](https://www.freedesktop.org/software/systemd/man/latest/systemd-time-wait-sync.service.html)).

mdavidsaver, 2024-06-11:
Actually, as I read more carefully...

https://github.com/epics-base/epics-base/blob/6ca716a77f24247e96952d1778bb1673600bb420/modules/libcom/src/osi/osiClockTime.c#L239-L242

I was mis-reading this as one big `if()`.  A case where I would put that `{` on the next line.

```c
    if (!ClockTimePvt.synchronized &&
        clockNow.tv_sec < POSIX_TIME_AT_EPICS_EPOCH) 
    {
        clockNow.tv_sec = POSIX_TIME_AT_EPICS_EPOCH + 86400;
        clockNow.tv_nsec = 0;
```

Maybe a workaround would be to `+=`

```c
        clockNow.tv_sec += POSIX_TIME_AT_EPICS_EPOCH + 86400;
```

This would handle 1970 as if it were 1990, and more importantly, allow time to advance until an NTP client can sync up.

Presumably the log spam is triggered when `clock_settime()` is omitted or a no-op.  Maybe the solution there is a `static int` flag to only print one the edges.

ralphlange, 2024-06-12:
I was only planning to add the static flag to print the warning once, but the `+=` also looks reasonable.

Note that this happens for _every client_ - IOCs were not really my main concern here.