epics-base/epics-base #456: dry-run make deletes some files

minijackson, 2024-01-19:
This issue was quite a bit fun to debug!

Turns out VSCode recommends installing the [Makefile tools](https://github.com/Microsoft/vscode-makefile-tools/) extension, which runs this command-line to get some info on the project:

```bash
make --dry-run --always-make --keep-going --print-directory
```

This command-line on an EPICS top removes all `CONFIGS` files during the installation phase. From my debugging, the files are removed by the `installEpics.pl`. Auditd says this command removed the `configure/CONFIG` file:

```bash
perl -CSD /nix/store/3jdcplbpdj2177csm7nbw3prjlaq1bdn-epics-base-7.0.7/bin/linux-x86_64/installEpics.pl -d -m 444 ../CONFIG ../.
```

And `installEpics.pl` says:

```
installEpics.pl: Copy failed: No such file or directory
installEpics.pl:        ../CONFIG -> ../../configure/TEMP.CONFIG.1267659
```

<details>
<summary>Full make logs</summary>

```
make: Entering directory '/tmp/tmp.69OuGuGda6'
make -C ./configure install
make[1]: Entering directory '/tmp/tmp.69OuGuGda6/configure'
perl -CSD /nix/store/3jdcplbpdj2177csm7nbw3prjlaq1bdn-epics-base-7.0.7/bin/linux-x86_64/makeMakefile.pl O.linux-x86_64 ../..
mkdir -p O.Common
make -C O.linux-x86_64 -f ../Makefile TOP=../.. \
    T_A=linux-x86_64 install
make[2]: Entering directory '/tmp/tmp.69OuGuGda6/configure/O.linux-x86_64'
Installing config file ../../configure/CONFIG
installEpics.pl: Copy failed: No such file or directory
installEpics.pl:        ../CONFIG -> ../../configure/TEMP.CONFIG.1267659
make[2]: *** [/nix/store/3jdcplbpdj2177csm7nbw3prjlaq1bdn-epics-base-7.0.7/configure/RULES_BUILD:539: ../../configure/CONFIG] Error 2
Installing config file ../../configure/RELEASE
installEpics.pl: Copy failed: No such file or directory
installEpics.pl:        ../RELEASE -> ../../configure/TEMP.RELEASE.1267661
make[2]: *** [/nix/store/3jdcplbpdj2177csm7nbw3prjlaq1bdn-epics-base-7.0.7/configure/RULES_BUILD:539: ../../configure/RELEASE] Error 2
Installing config file ../../configure/CONFIG_SITE
installEpics.pl: Copy failed: No such file or directory
installEpics.pl:        ../CONFIG_SITE -> ../../configure/TEMP.CONFIG_SITE.1267663
make[2]: *** [/nix/store/3jdcplbpdj2177csm7nbw3prjlaq1bdn-epics-base-7.0.7/configure/RULES_BUILD:539: ../../configure/CONFIG_SITE] Error 2
Installing config file ../../configure/RULES
installEpics.pl: Copy failed: No such file or directory
installEpics.pl:        ../RULES -> ../../configure/TEMP.RULES.1267665
make[2]: *** [/nix/store/3jdcplbpdj2177csm7nbw3prjlaq1bdn-epics-base-7.0.7/configure/RULES_BUILD:539: ../../configure/RULES] Error 2
../Makefile:3: Failed to remake makefile '../../configure/CONFIG'.
make[2]: Leaving directory '/tmp/tmp.69OuGuGda6/configure/O.linux-x86_64'
../../configure/CONFIG:9: Failed to remake makefile '../../configure/RELEASE'.
../../configure/CONFIG:33: Failed to remake makefile '../../configure/CONFIG_SITE'.
../Makefile:8: Failed to remake makefile '../../configure/RULES'.
make[1]: *** [/nix/store/3jdcplbpdj2177csm7nbw3prjlaq1bdn-epics-base-7.0.7/configure/RULES_ARCHS:58: install.linux-x86_64] Error 2
make[1]: Leaving directory '/tmp/tmp.69OuGuGda6/configure'
make: *** [/nix/store/3jdcplbpdj2177csm7nbw3prjlaq1bdn-epics-base-7.0.7/configure/RULES_DIRS:85: configure.install] Error 2
make: Leaving directory '/tmp/tmp.69OuGuGda6'
```

</details>

minijackson, 2024-01-19:
Thanks to @agaget for finding out that Makefile tools was the culprit!

ralphlange, 2024-01-19:
OMG!

agaget, 2024-01-19:
> Thanks to @agaget for finding out that Makefile tools was the culprit!

My pleasure, it's only been 3 years since we don't understand why this problem occurs😭

minijackson, 2024-01-19:
Turns out, it's probably the `--always-make` that leads to the removal of config files, just running `make --always-make` deletes my `configure/CONFIG` file.

anjohnson, 2024-01-19:
Confirmed, in my case `--always-make` deleted `RULES_COMMON`:

```
ctldev% make --always-make
perl -CSD ../src/tools/makeMakefile.pl O.linux-x86_64 ../..
mkdir -p O.Common
make -C O.linux-x86_64 -f ../Makefile TOP=../.. \
    T_A=linux-x86_64 install
make[1]: Entering directory '/home/phoebus4/ANJ/epics/base/7.0/configure/O.linux-x86_64'
Installing config file ../../configure/RULES_COMMON
installEpics.pl: Copy failed: No such file or directory
installEpics.pl:	../RULES_COMMON -> ../../configure/TEMP.RULES_COMMON.3366234
make[1]: *** [../../configure/RULES_BUILD:539: ../../configure/RULES_COMMON] Error 2
make[1]: Leaving directory '/home/phoebus4/ANJ/epics/base/7.0/configure/O.linux-x86_64'
make: *** [../configure/RULES_ARCHS:58: install.linux-x86_64] Error 2
ctldev% git st .
On branch 7.0
Your branch is up to date with 'upstream/7.0'.

Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	deleted:    RULES_COMMON

no changes added to commit (use "git add" and/or "git commit -a")
```

Interesting, I suspect this may be related to the code that copies the `configure/` tree when you have `INSTALL_LOCATION` set. I don't think it's likely to get fixed any time soon...

minijackson, 2024-01-19:
My guess is that when `INSTALL_LOCATION` is `$(TOP)`, the build system generates a rule like this:

```make
../configure/CONFIG: CONFIG
	@perl installEpics.pl ...
```

and because this is the same file, `../configure/CONFIG` is always up-to-date.

When overridden by `--always-make`, the `installEpics.pl` still gets run, and tries to:

- remove the destination file,
- copy the source file to a temp file,
- move the temp file to the destination

but because the source and destination are the same file, the source gets deleted.

One fix that could be simple to implement in `installEpics.pl`, is to check whether the source and destination are the same file, and do nothing if so (or just set the permissions).

I'm not sure why this script is still called on `--dry-run`, though…

simon-ess, 2024-02-07:
This seems to be relevant: https://stackoverflow.com/questions/67828011/automake-autotools-and-using-dry-run-and-always-make-for-make

It seems that included makefiles will always be remade (https://www.gnu.org/software/make/manual/html_node/Remaking-Makefiles.html). The solution could be, as above, to have a check in `installEpics.pl`, but we could also mark the RULES/CONFIG files as `.PHONY` targets as well, perhaps?

PR #461 (2024-02-21): Fix issue where VSCode makefile extension can delete files
The problem is that VSCode's make extension, in order to determine some information about the project, runs
```
make --dry-run --always-make
```
which despite its name will actually try to remake the configure/* files. Running `installEpics.pl` on these will delete them first, then try copy them, resulting in an error.

Fixes #456
closed 2024-02-21 by anjohnson