epics-base/epics-base #469: iocshLoad sets IOCSH_STARTUP_SCRIPT but doesn't undo that afterwards

anjohnson, 2024-03-16:
The `iocshLoad` command takes the path to a script and a list of macro parameters. By pushing the macLib scope before setting them, those macro values only exist while that script is running, and get discarded afterwards.

However the `iocshLoad()` code currently calls `epicsEnvSet()` to write the script name to the `IOCSH_STARTUP_SCRIPT` variable before it passes the script name and macro list to `iocshBody()` to execute it. There is code in `epicsEnvSet()` to allow scripts that are run using `iocshLoad` to set variables in the global environment using the `epicsEnvSet` command, so no matter how many `iocshLoad` operations are in progress the top-level environment variable by that name will always be overwritten. This means that the variable doesn't get restored when the scope is popped, and at the end of the original st.cmd file the `IOCSH_STARTUP_SCRIPT` variable is left set to the last script that was executed by `iocshLoad` at any level.

One solution might be to move the code which sets the environment variable into `iocshBody` after the new macLib scope has been pushed. However the original startup script path should be left set after the top-level script has finished running, so this might not be the best answer.

mdavidsaver, 2024-03-18:
> ... at the end of the original st.cmd file the `IOCSH_STARTUP_SCRIPT` ...

I find this a slightly surprising, I would probably have unset() afterwards.  What was your motivation?  (c11a31f5499f54bc471693a799f691f69b41da55)

mdavidsaver, 2024-03-18:
> ... macro parameters. By pushing the macLib scope ...

This idea may run into the same 256 character limit in macLib which has stalled https://github.com/epics-base/epics-base/pull/102#issuecomment-796958504 .

anjohnson, 2024-03-18:
> I find this a slightly surprising, I would probably have unset() afterwards.

Okay that bit is wrong, `IOCSH_STARTUP_SCRIPT` _should_ be unset after the script has finished executing.

However if one script calls another using an `iocshLoad` command, `IOCSH_STARTUP_SCRIPT` should be reverted back to the first script after that `iocshLoad` exits, which it doesn't now. That's the real bug.

anjohnson, 2024-03-18:
> 256 character limit in macLib

That sounds like another good Codeathon project. This comment appears in the `macLibNOTES` file in the source directory:

> Macro values can reference other macros in an arbitrarily complex way.
The only current restrictions are that a macro name or value cannot
exceed 256 characters in length. This restriction could fairly easily
be lifted.

The `MAC_SIZE` macro is only used in macCore.c and in the macLibTest.c code, so I think the above comment should still be accurate.

mdavidsaver, 2024-03-29:
> ... `IOCSH_STARTUP_SCRIPT` _should_ be unset after the script has finished executing.

Agreed.

> > ... This restriction could fairly easily be lifted.

Perhaps ;)

oksanagit, 2024-06-10:
I am looking into this

anjohnson, 2024-06-11:
As described in #487, I'm suggesting that `IOCSH_STARTUP_SCRIPT` should instead become a local variable that is stored in the current macLib scope. That means it will be replaced in iocsh command strings as before; it wouldn't be accessible as an environment variable or appear in the output from `epicsEnvShow`, but it would appear in the output from the new `show` (or `epicsShow`, `iocshShow` or `localShow` as we decide) command.

Making it local will be a user-visible change, and might break IOCs at any site which uses `getenv()` to read it, but I don't see an easy way to keep the environment variable value in sync with the local value after a `macPopScope()` operation.

anjohnson, 2024-06-11:
One issue I realize is that iocStats and other similar modules are reading and publishing the environment variable's value, although I'm not sure exactly when. However this will see the same value as my original bug report above, and when fixing that I think that the environment variable should probably only ever be set to the original st.cmd or whatever filename was given when the IOC was started.

That functionality can be added to the environment with a call to `epicsEnvSet()` from the `iocsh()` function in iocsh.cpp since that's what all our `main()` programs call. It should only set it the very first time `iocsh()` is called, and if the pathname given is NULL the user evidently didn't provide a startup script; should that set it to an empty string, or not set it at all?

mdavidsaver, 2024-06-11:
> One issue I realize is that iocStats ...

I see this as an issue _with_ iocStats, which a `field(SCAN, "10 seconds")` would address.  I don't think this effects Base one way or the other.

That said ...

> ... I think that the environment variable should probably only ever be set to the original st.cmd or whatever filename was given when the IOC was started.

Agreed.

So the logic could then become?

```c
    if (pathname && !getenv("IOCSH_STARTUP_SCRIPT"))
        epicsEnvSet("IOCSH_STARTUP_SCRIPT", pathname);
```

anjohnson, 2024-06-12:
> So the logic could then become?

Yeah, that looks right.