epics-base/epics-base #471: Interaction between `iocshLoad` and `on error`

exzombie, 2024-03-25:
I was surprised to find that `iocshLoad` does not appear to propagate errors to the parent scope, and I'm wondering whether I got something wrong, or is this by design, or by accident.

My goal is to stop executing scripts and get back the shell prompt when something goes wrong. In other words, `on error halt` is not what I'm looking for, so I used `on error break`. Using this statement both in `main.cmd` and `script.cmd` (which is called from the former using `iocshLoad`), I observe

- when an error is returned by a command in `script.cmd`, that script stops executing, but `main.cmd` continues execution;
- when an error is returned by a command in `main.cmd`, executions stops and I get the prompt back ...
- ... unless the command that failed was `iocshLoad` itself, in which case, execution of `main.cmd` continues.

Both files are quoted below, as is the output.

`main.cmd`:
```
on error break
iocshLoad  "nonexisting.cmd"
iocshLoad  "script.cmd"
dbLoadRecords "anothernonexisting.db"
iocInit
```

`script.cmd`:
```
on error break
dbLoadRecords("nonexisting.db")
```

Output:
```
on error break
iocshLoad  "nonexisting.cmd"
Can't open nonexisting.cmd: No such file or directory
iocshLoad  "script.cmd"
on error break
dbLoadRecords("nonexisting.db")
filename="../dbStatic/dbLexRoutines.c" line number=277dbRead opening file nonexisting.db
ERROR failed to load 'nonexisting.db'
iocsh Error: Break
dbLoadRecords "anothernonexisting.db"
filename="../dbStatic/dbLexRoutines.c" line number=277dbRead opening file anothernonexisting.db
ERROR failed to load 'anothernonexisting.db'
iocsh Error: Break
epics>
```

mdavidsaver, 2024-03-26:
> ... I'm wondering whether I got something wrong, or is this by design, or by accident.

I think: no, no, and yes.

mdavidsaver, 2024-06-11:
6ca716a77f24247e96952d1778bb1673600bb420 ensures that the `iocshLoad()` wrapper communicates failure.

exzombie, 2024-11-06:
I have cherry-picked that commit to our build and it appears to work well. Thank you!

closed 2024-11-06 by exzombie