epics-base/epics-base #472: Add a Q&A section to the Discussions page

wigging, 2024-04-01:
Can someone please add a Q&A section to the [Discussions](https://github.com/epics-base/epics-base/discussions) page?

anjohnson, 2024-04-01:
Hi Gavin,

Why? Is there something specific that GitHub Discussions provides, or are you just looking for somewhere to ask general EPICS questions?

We recommend using the [tech-talk mailing list](https://epics.anl.gov/tech-talk/index.php) for EPICS Q&A conversations, there are around a thousand people who read that list, but probably only a few tens of people who follow the Discussions area here. The tech-talk list archives for the last 30 years are [searchable](https://epics.anl.gov/tech-talk/search.php), and you don't have to subscribe to post there (just wait for the moderators to approve your messages) or to read (we provide RSS feeds).

- Andrew

wigging, 2024-04-01:
It's just more convenient (and modern) to ask programming related questions in GitHub Discussions than email. Code syntax highlighting and formatting is supported in GitHub Discussions and I can link directly to issues/code in the repository from within the discussion text. It's just a much better experience than email.

anjohnson, 2024-04-03:
Core Developers Meeting: We discussed this today, we don't have the time or effort to maintain or monitor yet another communications channel ourselves, if there would be enough people who would follow such a discussion and that could answer those kinds of questions we'd be fine with creating it, but I think you need to demonstrate that there are enough people who could answer those questions would be willing to monitor such a channel.

anjohnson, 2024-04-03:
@wigging Many of the roughly 1000 members of the tech-talk mailing list are employed as physicists, engineers and managers, and aren't programmers at all. Do you think they would all be willing to sign up for a GitHub account? Tech-talk is the heart of the EPICS community, which encompasses a lot more than just the software.

If you have questions that are not bug reports or suggested improvements against a specific EPICS module, the mechanism we offer at the moment to help you is the tech-talk list; please try it.

closed 2024-04-03 by anjohnson