epics-base/epics-base #474: Issues with asTrapWrite

mdavidsaver, 2024-04-15:
As mentioned in https://github.com/epics-modules/caPutLog/pull/23#issuecomment-2032499953 I think that the `asTrapWrite*()` API has a race condition introduced when `dbChannel::pfield` became mutable.

Beyond this bug, there a couple of API design decisions which with hindsight would be done differently.

1. `asTrapWriteRegisterListener()` accepts a function pointer, but no context `void*` pointer.  This limits flexibility.

https://github.com/epics-base/epics-base/blob/cb1571783b4793cb9c545e5736659ee5ed7d03bb/modules/libcom/src/as/asTrapWrite.h#L82-L83

2. `asTrapWriteMessage::serverSpecific` is untyped.  This `void*` was originally a `dbAddr*`, then in >= 3.15 changed to `dbChannel*`.

https://github.com/epics-base/epics-base/blob/cb1571783b4793cb9c545e5736659ee5ed7d03bb/modules/libcom/src/as/asTrapWrite.h#L42

This was effectively an incompatible API change which slipped through.  The (Marty's?) original version of `dbChannel*` started with an embedded `dbAddr`, so casting as safe (68dfeb76c0c59b21e45f2f57819d412e8c6c9507).  However, this changed early (eee5a2898368005d9ce5c12a343cff5a0355c188).

So user code is forced to `#if`.

https://github.com/epics-modules/caPutLog/blob/3df5287558d4a7da4ac5ac4f333bf7861eecd39e/caPutLogApp/caPutLogAs.c#L115-L121

(A note for anyone doing `git blame` on the early revision adding dbChannel.  27 April 2012 was the last day of a collaboration meeting.  I was furiously working to rebase the enormous `server-side-plugins` branch, and messed up the authorship of some of the revisions.  I believe that work originated with Marty at an earlier codeathon)

mdavidsaver, 2024-04-15:
When I wrote (https://github.com/epics-modules/caPutLog/pull/23#issuecomment-2032499953)

> ... So I think asLib.h needs to change so that asTrapWriteBeforeWithData() makes a "defensive" copy of the dbChannel* prior to queuing. ....

I wasn't thinking.  `dbChannel*` can't be cloned at present, and I am not sure how to do so without (at least) extending the server-side-filter API.

mdavidsaver, 2024-04-15:
Maybe the least horrible change would be to queue a copy of `dbAddr`, then make the asTrapWrite callback with an internally created temporary `dbChannel` with no filters?

mdavidsaver, 2024-04-15:
Is it worthwhile at this late date to rearrange `dbChannel` so that `dbAddr addr;` is the first member?

mdavidsaver, 2024-04-15:
> ... has a race condition ...

Ok, not a race condition in epics-base proper, actually a potential state inconsistency issue.

mdavidsaver, 2024-04-15:
sigh... `asTrapWriteUnregisterListener()` is in libCom, so it doesn't know about `dbChannel*` or `dbAddr*`, and can't easily take the simple action to save/restore `dbAddr::pfield`.

mdavidsaver, 2024-04-15:
Looking for modules which might be effected by changes to asTrapWrite API...

- caPutLog, the obvious
- PVXS, from [test code](https://github.com/mdavidsaver/pvxs/blob/4bd884719ef6e38cc5569b49ac22a8d9be0a833d/test/testqsingle.cpp#L609) only.  Callback calls `dbChannelGetField()`.  Base 3.14 not supported.
- DLS epics_device, from (maybe?) [test code](https://github.com/dls-controls/epics_device/blob/5ac267df169e2522f44be6022fe65b4ab0dc3c00/src/pvlogging.c#L127).  (@Araneidae fyi.)
- caputRecorder [does quite a bit](https://github.com/epics-modules/caputRecorder/blob/a82bcddbb640c61eaf3cabd945c71a074c295d91/caputRecorderApp/src/caputRecorder.c#L400).  Would be broken by making `asTrapWriteMessage::data` const.

Any others?

mdavidsaver, 2024-04-15:
> ... `asTrapWriteUnregisterListener()` is in libCom ...

One way to deal with this would be to move all the `asTrap*` from Com to dbCore.  ca-gateway doesn't actually call any `asTrap*()`.  The other callers I find are already linking to dbCore.  So maybe no compatibility issue?

anjohnson, 2024-04-16:
I don't think Marty would have come up with the name `dbChannel`. I imported the YAJL parser to libCom and designed and wrote the original code for parsing JSON into server-side channel filters, then gave it to Ralph write the server-queue filtering parts. Ralph added both our names as Authors to the dbChannel.h file a little later in 2012.

I am fine with moving the `asTrap*` implementation to the dbCore library, it makes more sense there anyway, especially if that would that simplify your changes in #475.

ralphlange, 2024-04-16:
I support the move to dbCore.

I remember (vaguely) working with Andrew on getting the server-side filters into Base, with me focusing on the filters and Andrew on the Base side of things.

mdavidsaver, 2024-04-18:
> I don't think Marty would have come up with the name `dbChannel`.  ...

Apologies.  Perhaps I was confusing this change with another.  (dbNotify?)

> I am fine with moving the `asTrap*` ...

> I support the move to dbCore.

Ok, so this is a direction.

Also, I added a workaround to QSRV2 mdavidsaver/pvxs@57f79ce747f062a32eede1a630d33a5234e0808f which saves/restores `dbAddr::pfield` around the asTrapWrite callback points.

mdavidsaver, 2025-02-19:
I think #501 would fix to misbehavior originally observed with caPutLog (which was separately patched with a workaround, which remains necessary for older Base).

closed 2025-02-19 by mdavidsaver