epics-base/epics-base #476: Possibility of including limit severities

fonty422, 2024-04-21:
We have a use case that requires us to be aware of the severity attached to each limit. Unfortunately we can't assume that if there is a high limit that the severity will be MAJOR, otherwise we would default to this.

We can get these values directly and store them as some local state, but if they are modified at any point we will be unaware unless we specifically make a call to get these values again.

It would be nice if they can be included in the dbr such that when using pyepics we can monitor those values for updates dynamically when monitoring a PV.

I'm ok if this is a large ask and won't be done, I can create a workaround if need be.

ralphlange, 2024-04-21:
Client can indeed set up subscriptions to these metadata properties.
Subscribe to VAL, use a DBR_* type that includes the metadata you're interested in and use the event flag DBE_PROPERTY.

Or consider moving to pvAccess, which is a lot more flexble.

fonty422, 2024-04-21:
Thanks @ralphlange.
We are already setting up our PV to use DBE_PROPERTY but it seemingly does not include the HHSV through LLSV fields.
For example:

```python
pv = PV("TEST:MONITOR", form="ctrl", auto_monitor=(dbr.DBE_VALUE | dbr.DBE_PROPERTY | dbr.DBE_ALARMY))
pv.get_with_metadata() // does include the limits, but not the limit severities
pv.get_ctrlvars() // As above, does include the limits, but not the limit severities
```
The only option I think seems to be to subscribe directly with `TEST.MONITOR.HHSV`

Please correct me if I'm wrong and have completely misunderstood.

ralphlange, 2024-04-21:
You are right.
I'm on my phone and didn't check the details. Yes, the levels are sent, but not the severities. 

Setting up separate monitors to the severity fields is probably your best option.

fonty422, 2024-04-21:
Thanks. Will do.

mdavidsaver, 2024-04-21:
fyi. The NTScalar definition includes eg. `valueAlarm.lowAlarmSeverity`, however the record support DBR_ types provide only the alarm range, not severity.  So there isn't much which QSRV can do.  The first step for someone to add this would be to extend that API as was done in #71.

https://github.com/epics-base/epics-base/blob/cb1571783b4793cb9c545e5736659ee5ed7d03bb/modules/database/src/ioc/db/dbAccessDefs.h#L144-L148

anjohnson, 2024-05-15:
Core: This might be a suitable codeathon project, but it is quite advanced. Requires RSET changes (possibly a new routine), and dbAccessDefs.h as @mdavidsaver notes above.