epics-base/epics-base #483: Incompatible change proposal: install path namespace

kenhys, 2024-05-14:
# Problem

header files of epics-base is installed under /usr/include/ directly, 
it occupies too comprehensive namespace there.
I'm not observed that the actual conflicted use case yet, but it may be better to install own namespace
such as /usr/include/epics. 

# Actual result

header files were installed under:

* /usr/include/
* /usr/include/os/
* /usr/include/compiler
* /usr/include/pv
* /usr/include/pva
* /usr/include/valgrind

# Expected result

Install header files under own namespace.

* /usr/include/epics/
*  /usr/include/epics/os/
* /usr/include/epics/compiler
* /usr/include/epics/pv
* /usr/include/epics/pva
* /usr/include/epics/valgrind

Not to conflict with other software.

Yes, it is apparently incompatible changes, but it avoids collisions with other software in the future.

kenhys, 2024-05-14:
FYI: epics-dev which is  packaged in debian adopt to install /usr/include/epics/.

https://packages.debian.org/sid/amd64/epics-dev/filelist

anjohnson, 2024-05-15:
Hi, welcome to the EPICS community!

If your build resulted in the EPICS Base header files being installed into `/usr/include` you must have set `INSTALL_LOCATION=/usr` when building it. EPICS was not designed to be installed there, our `INSTALL_LOCATION` setting is not equivalent to the prefix value used to build many other applications. We don't recommend building or installing EPICS Base or its support modules as root or using `sudo` either; I suggest making the directory that you install EPICS modules into be owned by the person who builds them, or by a shared account if that could be several different people.

@mdavidsaver explained above that epicsdeb picked `/usr/lib/epics` as their standard location for EPICS Base. Other sites have traditionally used locations such as `/usr/local/epics` or `/opt/epics/base`. You will usually need to install several other EPICS support modules such as Asyn as well, which might be installed alongside EPICS Base at `/opt/epics/asyn`. As you use EPICS more and need to upgrade modules you often want to include version numbers in the install paths too such as `/opt/epics/base-7.0.8` and `/opt/epics/asyn-4.42`.

kenhys, 2024-05-18:
I hope that there is no conflict and safe to install by default even though user install in system wide.

But it turned out that it is not recommend way. so I'll close this issue.

Thanks.

closed 2024-05-18 by kenhys