epics-base/epics-base #485: IOC Segmentation fault when SIZV is larger than 32767

Insomnia1437, 2024-05-17:
Found on R7.0.8, and can repeat on latest commit d8b5616772df78c50e94fb642c4903d757232e38
Test db:
```
record(lsi, lsi){
  field(SIZV, "32767")
  field(INP,  {const: aaa})
  field(PINI, "YES")
}
record(lsi, lsi-bad){
  field(SIZV, "32768")
  field(INP,  {const: aaa})
  field(PINI, "YES")
}
```
When using  `caget lsi-bad`, ioc crashs.
```shell
# iocsh
$ softIocPVA -d bug.db
Starting iocInit
############################################################################
## EPICS R7.0.8
## Rev. 2024-03-27T10:23+0100
## Rev. Date build date/time:
############################################################################
iocRun: All initialization complete
epics> [1]    1485 segmentation fault  softIocPVA -d bug.db

# another terminal
$ caget lsi
lsi                            aaa
$ caget lsi-bad
CA.Client.Exception...............................................
    Warning: "Virtual circuit disconnect"
    Context: "172.29.165.14:5064"
    Source File: ../cac.cpp line 1238
    Current Time: Fri May 17 2024 22:15:48.322332782
..................................................................
CA.Client.Exception...............................................
    Warning: "Virtual circuit disconnect"
    Context: "op=0, channel=lsi-bad, type=DBR_TIME_STRING, count=1, ctx="172.29.165.14:5064""
    Source File: ../getCopy.cpp line 91
    Current Time: Fri May 17 2024 22:15:48.322409017
..................................................................
Read operation timed out: some PV data was not read.
lsi-bad
```
stack trace:
```
Thread 35 "CAS-client" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffaf6126c0 (LWP 1803)]
__strncpy_avx2 () at ../sysdeps/x86_64/multiarch/strcpy-avx2.S:845
845     ../sysdeps/x86_64/multiarch/strcpy-avx2.S: No such file or directory.
(gdb) bt
#0  __strncpy_avx2 () at ../sysdeps/x86_64/multiarch/strcpy-avx2.S:845
#1  0x00007ffff7e7beb8 in getStringString (paddr=0x7fff94005370, pto=<optimized out>, nRequest=<optimized out>, no_elements=1, offset=0)
    at ../db/dbConvert.c:146
#2  0x00007ffff7e73201 in dbGet (paddr=paddr@entry=0x7fff94005370, dbrType=dbrType@entry=0, pbuffer=<optimized out>,
    options=options@entry=0x7fffaf611740, nRequest=nRequest@entry=0x7fffaf611c70, pflin=pflin@entry=0x0) at ../db/dbAccess.c:1015
#3  0x00007ffff7e75f0c in dbChannelGet (chan=chan@entry=0x7fff94005368, type=type@entry=0, pbuffer=<optimized out>,
    options=options@entry=0x7fffaf611740, nRequest=nRequest@entry=0x7fffaf611c70, pfl=pfl@entry=0x0) at ../db/dbChannel.c:630
#4  0x00007ffff7e8d07d in dbChannel_get_count (chan=chan@entry=0x7fff94005368, buffer_type=<optimized out>, pbuffer=0x7ffff40ae0a0,
    nRequest=nRequest@entry=0x7fffaf611c70, pfl=pfl@entry=0x0) at ../db/db_access.c:196
#5  0x00007ffff7eb3cf2 in read_reply (pArg=pArg@entry=0x7fffaf611cc0, dbch=0x7fff94005368, eventsRemaining=eventsRemaining@entry=1,
    pfl=pfl@entry=0x0) at ../rsrv/camessage.c:548
#6  0x00007ffff7eb4d74 in read_notify_action (mp=0x7fffaf611d60, pPayload=0x7ffff40aa098, client=0x7fffb4001238) at ../rsrv/camessage.c:745
#7  0x00007ffff7eb5cfd in camessage (client=client@entry=0x7fffb4001238) at ../rsrv/camessage.c:2546
#8  0x00007ffff7eb2f48 in camsgtask (pParm=0x7fffb4001238) at ../rsrv/camsgtask.c:116
#9  0x00007ffff7e147a5 in start_routine (arg=0x7fffb00011c0) at ../osi/os/posix/osdThread.c:442
#10 0x00007ffff7a34134 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#11 0x00007ffff7ab47dc in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) frame 1
#1  0x00007ffff7e7beb8 in getStringString (paddr=0x7fff94005370, pto=<optimized out>, nRequest=<optimized out>, no_elements=1, offset=0)
    at ../db/dbConvert.c:146
146             strncpy(pdst, psrc, sizeto);
(gdb) info locals
psrc = <optimized out>
pdst = <optimized out>
size = -32768
sizeto = -32768
```
Overflow occurred. The type of `SIZV` field is USHORT, but short type is used here. 
https://github.com/epics-base/epics-base/blob/d8b5616772df78c50e94fb642c4903d757232e38/modules/database/src/ioc/db/dbConvert.c#L132-L146

Is changing the type of `SIZV` to short a good idea to fix? I don't know...

mdavidsaver, 2024-05-19:
Confirmed.  `sizeto` overflows.  Since it is signed (sigh...) it wraps negative.

```
(gdb) f 1
#1  0x00007ffff7ec6e77 in getStringString (paddr=(dbAddr*)(0x7fffa8005f58, "lsi-bad.VAL"), pto=0x7ffff55300ac, 
    nRequest=1, no_elements=1, offset=0) at ../db/dbConvert.c:146
146             strncpy(pdst, psrc, sizeto);
(gdb) p pdst
$1 = 0x7ffff55300ac "aaa"
(gdb) p psrc
$2 = 0x555555649ea0 "aaa"
(gdb) p sizeto
$3 = -32768
```

mdavidsaver, 2024-05-19:
There is an implicit limitation on field size set by `short dbAddr::field_size`.

https://github.com/epics-base/epics-base/blob/d8b5616772df78c50e94fb642c4903d757232e38/modules/database/src/ioc/db/dbAddr.h#L23

In the short term, this limit needs to be enforced.  In the longer term, it should be lifted.

mdavidsaver, 2024-05-19:
The offending cast happens in lsoRecord.c `cvt_dbaddr()` since `SIZV` is (imo. reasonably) `DBF_USHORT`.

https://github.com/epics-base/epics-base/blob/d8b5616772df78c50e94fb642c4903d757232e38/modules/database/src/std/rec/lsiRecord.c#L141

mdavidsaver, 2024-05-19:
> In the short term, this limit needs to be enforced.  ...

`SIZV` capped at 32767 by e5b4829074b296634233dea756d77379749c1bd6

Insomnia1437, 2024-05-19:
@mdavidsaver I added a patch also for `printf` record

closed 2024-05-29 by anjohnson