epics-base/epics-base #487: Local variables in iocsh scripts

exzombie, 2024-05-29:
Being able to define local variables would make it easier to avoid repetition in iocsh scripts loaded by `iocshLoad`. Currently, the only way to define a variable is using `epicsEnvSet`, which creates a globally visible environment variable.

To be specific, I would like something like the following, with emphasis on the `$(PORT)`; the other stuff is there only to illustrate the use case (hopefully not obscuring the goal too much)

`script1.cmd`:
```
localSet("PORT", "A_COMPLICATED__AND_DYNAMIC_PORT_NAME")
drvAsynIPPortConfigure("$(PORT)", "$(ADDRESS)", 0, 0, 0)
dbLoadRecords("database.db", "PREFIX=$(DEVICE),PORT=$(PORT)")
...
iocshLoad("script2.cmd", "DEVICE=$(DEVICE),PARAM=SOMETHING")
...
```

`script2.cmd`:
```
localSet("PORT", "SOMETHING_COMPLETELY_DIFFERENT")
drvCustomDriver("$(PORT)", "ARG=$(PARAM)")
dbLoadRecords("other.db", "PREFIX=$(DEVICE),PORT=$(PORT)")
```

It would be nice if `script2.cmd`, which could come from another module, didn't need to know that it shouldn't use the name `PORT`. This cannot be achieved with `epicsEnvSet`. A possible workaround is to "namespace" variable names by prefixing them with something unique to each script, which is less than ideal.

Given that `iocshLoad` already creates a scope for the purpose of scoping its macro arguments, I'm wondering if this issue could be solved in a straightforward manner by exposing `macPutValue()` through an iocsh command.

anjohnson, 2024-06-11:
I like this, but would personally prefer the commands `set` and `show` — are these names better than `epicsSet` and `epicsShow` (to match `epicsEnvSet` etc.) or `iocshSet` and `iocshShow`? Implement with calls to `macPutValue(context->handle, name, value)` and `macReportMacros(context->handle)`. The implementation should go in iocsh.cpp.

@oksanagit has shown that using `macPutValue` with the iocsh `context->handle` does not put the name/value in the global environment, which is what @exzombie suggested.

This is related to #469; that issue would be modified to make `IOCSH_STARTUP_SCRIPT` a local variable; that would remove it from the environment and make it only visible to other iocsh commands, is that likely to be a problem?

mdavidsaver, 2024-06-11:
I like this idea.

One potential complication (or at least confusion) is that different parse of Base currently use separate `MAC_HANDLE`.

```
modules/database/src/ioc/dbStatic/dbLexRoutines.c:    if(macCreateHandle(&macHandle,NULL)) {
modules/database/src/ioc/dbtemplate/msi.cpp:    macCreateHandle(&macPvt, 0);
modules/database/test/ioc/dbtemplate/dbltExpand.c:    if (macCreateHandle(&macHandle, NULL)) {
modules/libcom/src/as/asLibRoutines.c:        if((status = macCreateHandle(&macHandle,NULL))) {
modules/libcom/src/iocsh/iocsh.cpp:        if (!context || macCreateHandle(&context->handle, pairs)) {
```

exzombie, 2024-06-11:
IIUC, these files all have separate purposes, and so, it's appropriate that they have separate contexts/handles. This issue is about iocsh and should not affect other files. The new iocsh commands would clearly affect only the context of `iocsh.cpp`, so I don't think the reader of the code would be confused as to which macros are affected.

mdavidsaver, 2024-06-11:
> ... I don't think the reader of the code ...

I was thinking more about someone reading an iocsh script.

I guess it is reasonable as long as we are clear that IOCsh locals are really local only to IOCsh, and not used by eg. the .db parser.

exzombie, 2024-06-12:
I started looking at #222 again and noticed that `set` is used there, albeit in the context of db files, e.g., `set("link:scope", "INT")`. While the context is different, I'm not sure whether having different user-facing commands/directives named "set" would be confusing or not. Something to think about. Perhaps naming the commands `iocshSet`/`iocshShow` would also deal with Michael's concern above?