epics-base/epics-base #488: CA (and PVA) peers do DNS lookups once on startup, if the IP changes (but DNS does not) camonitor/cagateway  disconnect forever

amichelotti, 2024-06-04:
This issue has been detected on a kubernates setup where it's used EPICS_CA_ADDR_LIST to list the CA peers by using DNS entries after restarting IOC/POD. If the peer change its IP (very common in k8s environments) or maybe for hw replacement maintaining  its DNS name camonitor/cagateway must be restarted. It would be very nice if a check IP/DNS could be done on disconnection/timeout and if not match a channel be recreated.

ralphlange, 2024-06-05:
Conceptionally, I would say that allowing names in ADDR_LIST and silently assuming their numerical representation never changes is a design flaw.
On the other hand, such changes rarely happen in traditional control systems and IOCs coming up with a different IP at every reboot are pretty much on the opposite side of the spectrum.

Because of that, I would think it's a legitimate request with low priority, which realistically makes it - as Andrew says - a do-it-yourself (or fund-it-yourself) project.

amichelotti, 2024-06-06:
Hi @anjohnson,

I completely agree with @ralphlange about the "design flaw." This issue can create problems in modern workflows and is not easily understood by 'expert newcomers to EPICS' like myself (I spent a week trying to figure out CAGateway reconnection problems before realizing the issue with the help of Ralph, Michael, and Giles). UDP multicast depends on the Kubernetes CNI that your IT group has deployed, or you can use the host network as Giles does. I currently don't have this option in my OpenShift Kubernetes deployment.

Since I'm relatively new to the EPICS world, I'm not very familiar with epics UDP discovery, broadcast, and CA repeater. These concepts seem to me well-designed but a bit dated, so in my new k8s workflow, I try to use approaches that involve just dns names and TCP connections that are well supported across different k8s distributions. Now that I understand the problem, I'll implement a workaround but even if k8s projects can work around this issue I strongly suggest to fix this 'bug' to make epics more robust and opened to new incoming workflows.

amichelotti, 2025-07-21:
> [@amichelotti](https://github.com/amichelotti) I think you posted recently that you were not seeing this issue with PVA any more, just with CA? I just updated the demo so that pvagw is also using DNS names. The test seems to show PVA also suffering from the same issue - but I have not spent much time to verify this.

Hi @gilesknap,
I'm not 100% sure the issue with PVA is fully resolved, as I haven’t had a chance to retest it recently. As a workaround in my current setups, I force the IOC and services pods to retain the same IP address across restarts. This seems to work for both protocols, although I’m not very happy with this solution—it’s quite a workaround in Kubernetes and could become problematic at scale.

I also recall that with PVA specifically, I ran into an additional issue related to Kubernetes NAT (which didn’t occur with CA Gateway). However, by using EPICS_PVA_NAME_SERVERS instead of EPICS_PVA_ADDR_LIST, as you suggested, I was able to overcome that issue.

I’m really glad that @EmilioPeJu  is looking into this bug, as it seems to impact in particular modern deployments like ours (and I hope many others in the future).

mdavidsaver, 2025-07-25:
> I'm not 100% sure the issue with PVA is fully resolved

wrt. address lists, the situation with PVA is the same.  wrt. ACF files, the situation is different.

Be aware that any proposed "fix" needs to address the question of when DNS resolution should be repeated.  Also, it is essential to avoid blocking DNS lookups (forward or reverse) on socket handling threads.  DNS query timeouts are frequently longer than PVA/CA timeouts.