epics-base/epics-base #494: `pushd`, `popd` and `dirs` commands for iocsh

JJL772, 2024-06-10:
These commands tend to be quite useful in both interactive shell sessions and in scripts. Providing these for iocsh might be helpful.

This page describes their behavior in bash: https://www.gnu.org/software/bash/manual/html_node/Directory-Stack-Builtins.html

Of course, the iocsh equivalents of these don't need to support all of those features.

This might be a good quick codeathon task.

mdavidsaver, 2024-06-11:
I can see the usefulness.  Potential complications for you to consider.

How would this directory stack interact with `iocshLoad()`?

```sh
pushd("$PWD")
iocshLoad("some.cmd")
popd()
```

What if `some.cmd` makes unbalanced `pushd` / `popd` calls?

JJL772, 2024-06-11:
Thanks for the feedback. Should there also be a limit on the directory stack size?

mdavidsaver, 2024-06-11:
> ... Potential complications for you to consider.

Ah, right Looking at #497, we need to remember that iocsh() can be called from any thread at any time.  Not just main().

What should happen if some stringout record calls `iocshCmd("pushd /home")`?

Should these new commands be limited to interactive sessions?  Tied to the readline context?

JJL772, 2024-06-12:
> Should these new commands be limited to the interactive session? Tied to the readline context?

Are there any other commands that are only available in the interactive session? Might be a bit inconsistent to only register these there. Also, do startup scripts (i.e. st.cmd) count as part of the interactive session?

anjohnson, 2024-06-12:
I don't know of a way to register different command-sets with different sessions at the moment, how would the registration indicate the session? I guess @mdavidsaver might be thinking that the iocsh-provided commands could do that. I'm not aware of anything preventing there from being more than one concurrent interactive session other than the coding to accept a connection from another pty.

mdavidsaver, 2026-10-14:
> ... I'm not aware of anything preventing there from being more than one concurrent interactive session other than the coding to accept a connection from another pty.

The readline API is not reentrant.  Lots of global variables.