epics-base/epics-base #495: posix: data race on `pthreadInfo->osiPriority`

mdavidsaver, 2024-06-11:
There is no locking of `pthreadInfo->osiPriority` when accessed by `epicsThreadSetPriority()` and `epicsThreadGetPriority()`.

```
valgrind --tool=helgrind ./modules/libcom/test/O.linux-x86_64-debug/epicsThreadTest
...
==521448== Possible data race during read of size 4 at 0x4ED7698 by thread #12
==521448== Locks held: none
==521448==    at 0x48A6BA0: epicsThreadGetPriority (osdThread.c:769)
==521448==    by 0x489E71F: epicsThread::getPriority() const (epicsThread.cpp:275)
==521448==    by 0x10A5F9: (anonymous namespace)::myThread::run() (epicsThreadTest.cpp:64)
==521448==    by 0x489DD1D: epicsThreadCallEntryPoint (epicsThread.cpp:95)
==521448==    by 0x48A5D43: start_routine (osdThread.c:442)
==521448==    by 0x484B7D6: mythread_wrapper (hg_intercepts.c:406)
==521448==    by 0x4BD8133: start_thread (pthread_create.c:442)
==521448==    by 0x4C57A3F: clone (clone.S:100)
==521448== 
==521448== This conflicts with a previous write of size 4 by thread #1
==521448== Locks held: none
==521448==    at 0x48A6C66: epicsThreadSetPriority (osdThread.c:793)
==521448==    by 0x489E745: epicsThread::setPriority(unsigned int) (epicsThread.cpp:280)
==521448==    by 0x10A6F8: (anonymous namespace)::testMyThread() (epicsThreadTest.cpp:80)
==521448==    by 0x10AED1: main (epicsThreadTest.cpp:254)
```
```

mdavidsaver, 2024-06-11:
This seems unlikely to cause a practical issue, and I hesitate to either add a mutex to `epicsThreadOSD` or make use of `epicsAtomic*`.  Out of concern for resource usage on RTEMS, or recursion on targets which fallback to epicsMutex for atomic ops.

mdavidsaver, 2025-02-19:
It seems like other mutable members of `epicsThreadOSD` will have similar races (eg. `isSuspended` or `schedParam`).

hjunkes, 2026-10-19:
The race is real but harmless: epicsThreadSetPriority() stores an aligned unsigned int with no lock, so a concurrent epicsThreadGetPriority() sees either the old or the new priority, never a torn value. The writer still applies the change correctly through pthread_setschedparam().
In epicsThreadTest the write even stores the value the field already holds (50+i), so nothing observable changes. The helgrind/TSan report comes from the test calling setPriority() after start().
Proposed fix: start each thread only after setPriority() in the test, which leaves TSan with no osiPriority reports. osdThread.c stays
unchanged, so no per-thread mutex or atomics are added.
TSan also still reports a race on pthreadInfo->tid. That one is intentional and benign: both threads write the same value, and the source comments already document it.

hjunkes, 2026-10-22:
PR #1019 avoids the race in epicsThreadTest by setting the priority before start().

A separate PR (osdThreadHardening) will address the unlocked accesses in
osdThread.c itself (osiPriority, isSuspended, schedParam, and the
concurrent writes to tid).