epics-base/epics-base #499: iocsh allows recursive iocshLoad

mdavidsaver, 2024-06-11:
Currently this "works".

```
$ cat recursive.cmd 
echo "Oh No!"
iocshLoad("recursive.cmd")
```

```
$ softIoc recursive.cmd
...
echo "Oh No!"
Oh No!
iocshLoad("recursive.cmd")
echo "Oh No!"
Oh No!
iocshLoad("recursive.cmd")
echo "Oh No!"
Oh No!
iocshLoad("recursive.cmd")
echo "Oh No!"
Oh No!
iocshLoad("recursive.cmd")
echo "Oh No!"
Oh No!
iocshLoad("recursive.cmd")
ERROR Too many open files (24) loading '.iocsh_history'
echo "Oh No!"
Oh No!
iocshLoad("recursive.cmd")
Can't open recursive.cmd: Too many open files
epics> 
```

mdavidsaver, 2024-06-11:
Any attempt to detect recursion would need to take into account that the same file might legitimately be loaded several times with different macro values, which could only determined after expansion.  So maybe a simple bound on call depth (~10)?  Or maybe "won't fix"?

exzombie, 2024-06-13:
> Can't open recursive.cmd: Too many open files

Sounds like there already is a recursion limit, albeit incidental. Running out of fds is not very nice, but I don't see it as any worse than imposing yet another arbitrary limit. The important thing is that the recursion is limited *somehow* in order not to bring the machine down by running out of memory.