epics-base/epics-base #506: Preventing conflicts from Release Notes entries

anjohnson, 2024-06-12:
This issue provides some ideas on how to get around the problem that commits to the `documentation/RELEASE_NOTES.md` file generate merge conflicts on GitHub and require manual resolution when merging.

Here are some of the requirements that I have in mind for any replacement:

* Easy to add new entries to the notes for a future release, preferably using GitHub Flavored Markdown, in such a way that new entries don't cause merge conflicts.
* Provide a template for new entries, and/or a `make` command to generate a new entry.
* PR merges may be delayed or held back, so new entries cannot name the version or branch they expect to be included in.
* A make command can be run (in the `documentation` directory?) to combine all new entries into a single Markdown file which describes everything new since the last release.
* At release time, all new entries may be converted into a single file, or moved into a release-specific directory. However see the next item:
* Multiple active and supported Git branches may exist for different major versions, any of which can be targeted by GitHub pull requests. Changes applied to earlier branches (e.g. 7.0) will be merged up into later ones (e.g. 7.1), so release entries must also merge up correctly and appear in releases for all branches they were applied to.

In #464 @ericonr wrote:
> I think the best way to handle this that I've seen is how CPython does it. Each PR should include a blurb in the relevant https://github.com/python/cpython/tree/main/Misc/NEWS.d/next directory, and when there is a release, the release notes are automatically generated by pooling all changes together.

I was thinking similarly, my suggestion is to add a `documentation/notes` directory with a subdirectory named `pending`. New entries for the release notes would go in new files added to `pending`, but could be combined into a single file (that would be added to git) at the next release.

Running `make` in the `documentation` directory would combine all the `.md` files from the `pending` subdirectory into one file installed into `$(INSTALL_DOC)` (i.e. `$(TOP)/doc`). This might need a Perl script that would also add titles and other boiler plate.

Discussion welcomed.

JJL772, 2024-06-12:
I suggest enforcing a naming scheme such as `issuenumber-this-is-my-title.md`. For example: `464-allow-deleting-record.md` or `544-my-cool-feature.md`. Then, in the generated release notes, you can provide a link to that issue or pull request. Adding the issue number prefix would also ensure that the files are unique.

mdavidsaver, 2024-06-19:
I like the idea of splitting out pending release notes entries into separate files.

> * A make command can be run ...

While I hope it would be no more complicated than a `$(CAT)`, I don't see automation as a "requirement".