epics-base/epics-base #513: Investigate uses of Windows.h

JJL772, 2024-06-14:
`epicsAtomicOSD.h` for Windows includes `windows.h` for the `InterlockedXXX` declarations. This leads to some pollution due to the pile of other headers `windows.h` includes. User code including `epicsAtomic.h` will eventually include `epicsAtomicOSD.h` and thus `windows.h` on Windows.

#510 improves on this behavior somewhat by replacing `VC_EXTRALEAN` with `WIN32_LEAN_AND_MEAN` and adding `NOMINMAX`. `VC_EXTRALEAN` is only relevant to MFC sources.

More cleanup is required though. There are some options for tweaking what is included or declared, as seen at the top of windows.h:
```
/*  If defined, the following flags inhibit definition
 *     of the indicated items.
 *
 *  NOGDICAPMASKS     - CC_*, LC_*, PC_*, CP_*, TC_*, RC_
 *  NOVIRTUALKEYCODES - VK_*
 *  NOWINMESSAGES     - WM_*, EM_*, LB_*, CB_*
 *  NOWINSTYLES       - WS_*, CS_*, ES_*, LBS_*, SBS_*, CBS_*
 *  NOSYSMETRICS      - SM_*
 *  NOMENUS           - MF_*
 *  NOICONS           - IDI_*
 *  NOKEYSTATES       - MK_*
 *  NOSYSCOMMANDS     - SC_*
 *  NORASTEROPS       - Binary and Tertiary raster ops
 *  NOSHOWWINDOW      - SW_*
 *  OEMRESOURCE       - OEM Resource values
 *  NOATOM            - Atom Manager routines
 *  NOCLIPBOARD       - Clipboard routines
 *  NOCOLOR           - Screen colors
 *  NOCTLMGR          - Control and Dialog routines
 *  NODRAWTEXT        - DrawText() and DT_*
 *  NOGDI             - All GDI defines and routines
 *  NOKERNEL          - All KERNEL defines and routines
 *  NOUSER            - All USER defines and routines
 *  NONLS             - All NLS defines and routines
 *  NOMB              - MB_* and MessageBox()
 *  NOMEMMGR          - GMEM_*, LMEM_*, GHND, LHND, associated routines
 *  NOMETAFILE        - typedef METAFILEPICT
 *  NOMINMAX          - Macros min(a,b) and max(a,b)
 *  NOMSG             - typedef MSG and associated routines
 *  NOOPENFILE        - OpenFile(), OemToAnsi, AnsiToOem, and OF_*
 *  NOSCROLL          - SB_* and scrolling routines
 *  NOSERVICE         - All Service Controller routines, SERVICE_ equates, etc.
 *  NOSOUND           - Sound driver routines
 *  NOTEXTMETRIC      - typedef TEXTMETRIC and associated routines
 *  NOWH              - SetWindowsHook and WH_*
 *  NOWINOFFSETS      - GWL_*, GCL_*, associated routines
 *  NOCOMM            - COMM driver routines
 *  NOKANJI           - Kanji support stuff.
 *  NOHELP            - Help engine interface.
 *  NOPROFILER        - Profiler interface.
 *  NODEFERWINDOWPOS  - DeferWindowPos routines
 *  NOMCX             - Modem Configuration Extensions
 */
```

Another option might be to provide our own `InterlockedXXX` declarations in `epicsAtomic.h` instead of including `winnt.h`/`windows.h`

Some discussion was made in #510

FreddieAkeroyd, 2024-06-14:
`epicsAtomicOSD.h` contains things like 
```
#define MS_InterlockedIncrement InterlockedIncrement
```
so it would be relatively easy to move these to a separate file and have  `MS_InterlockedIncrement` as real functions, however there would be a performance penalty but i don't know how big a worry that is. These functions are often implemented as compiler intrinsics  via  https://learn.microsoft.com/en-us/cpp/intrinsics/interlockedexchange-intrinsic-functions?view=msvc-170 it may be possible to keep this optimisation and remove windows.h though it may be after defining all of the above `NO*` we get what we want pretty much with `windows.h`.

mdavidsaver, 2024-06-26:
> ... These functions are often implemented as compiler intrinsics ...

While I also think that intrinsic/builtin functions should be investigated first, I also note that the docs for the functions like [`InterlockedExchange()`](https://learn.microsoft.com/en-us/windows/win32/api/winnt/nf-winnt-interlockedexchange) state:

```
Header 	winnt.h (include Windows.h)
```

Is including `winnt.h` an option?  Would it actually give any benefit vs. `windows.h`?

JJL772, 2024-06-26:
Including `winnt.h` directly is not possible because `windows.h` has some required defines at the top of it. 
This is what you'll see if you do that: `Z:/compilers/windows-kits-10/include/10.0.22621.0/um\winnt.h(169): fatal error C1189: #error:  "No Target Architecture"`
https://godbolt.org/z/o9Kcs491o

We *could* also copy those definitions into `epicsAtomicOSD.h` and include `winnt.h` directly.