epics-base/epics-base #514: Issue with `-D_FORTIFY_SOURCE=3` in `dbAllocRecord()`

mdavidsaver, 2024-06-14:
I (and others) have been successfully building/running epics-base with `-D_FORTIFY_SOURCE=2` for some time now.  However, enabling `-D_FORTIFY_SOURCE=3` triggers (at least) a fault in `dbAllocRecord()`. see #502

The extra checks enabled by level 3 are [described as](https://www.gnu.org/software/libc/manual/html_node/Source-Fortification.html):

> 3: This enables buffer bounds checking using the value returned by the __builtin_dynamic_object_size compiler builtin function. If the function returns (size_t) -1, the function call is left untouched. Fortification at this level may have a impact on program performance if the function call that is fortified is frequently encountered and the size expression returned by __builtin_dynamic_object_size is complex. 

The second argument of [`__builtin_dynamic_object_size(var, type)`](https://gcc.gnu.org/onlinedocs/gcc/Object-Size-Checking.html#index-_005f_005fbuiltin_005fdynamic_005fobject_005fsize) is relevant.  This is a bit mask with (as of gcc 12) two bits defined.

- bit 0
  - clear - `objects are whole variables`
  - set - `closest surrounding subobject`
- bit 1
  - `second bit determines if maximum or minimum of remaining bytes is computed. `   (I guess 0 is max. and 1 is min. ???)

#502 proposed to change:

```diff
-                    strcpy(pfield,pflddes->initial);
+                    memcpy(pfield, pflddes->initial, initial_len);
+                    pfield[initial_len] = '\0';
```

In glibc 2.36, `bits/string_fortified.h` defines

```c
__fortify_function char *
__NTH (strcpy (char *__restrict __dest, const char *__restrict __src))
{
  return __builtin___strcpy_chk (__dest, __src, __glibc_objsize (__dest));
}
__fortify_function void *
__NTH (memcpy (void *__restrict __dest, const void *__restrict __src,
               size_t __len))
{
  return __builtin___memcpy_chk (__dest, __src, __len,
                                 __glibc_objsize0 (__dest));
}
```

One difference being `__glibc_objsize()` vs. `__glibc_objsize0()`.  These macros expand to `__builtin_dynamic_object_size()` with type `1` and `0` respectively.

Adding some debug prints shows that in `dbAllocRecord()` that `__glibc_objsize(pfield) ==0` while `__glibc_objsize0(pfield) ==size_t(-1)`.

Why?

Tracing back, GCC has no idea how large `ppvt` is because `callocMustSucceed()` has no `alloc_size()` annotation.  (simply adding this changes nothing)

```c
    ppvt = dbCalloc(1, offsetof(dbCommonPvt, common) + pdbRecordType->rec_size);
// __glibc_objsize(ppvt) == __glibc_objsize0(ppvt) == size_t(-1)
```

```c
    precord = &ppvt->common;
// __glibc_objsize0(ppvt) == size_t(-1)
// __glibc_objsize(ppvt) == sizeof(dbCommon)
```

Ah, here we are.  GCC doesn't understand that `ppvt->common` is the head of a variable size struct.

```c
    pfield = (char*)precord + pflddes->offset;
```

So when `pflddes->offset >= sizeof(dbCommon)`, the tracked size of `pfield` becomes zero.

I can trick GCC by replacing with

```c
precord = (dbCommon*)(offsetof(dbCommonPvt, common) + (char*)ppvt);
```

But I don't see this as a great solution, and there may well be other places where struct casting will cause problems.

mdavidsaver, 2024-06-14:
I _think_ size tracking issue will be limited to casting `dbCommon` -> `*Record`.  Casting `*Record` -> `dbCommon` should be fine.

mdavidsaver, 2024-06-14:
Also, we should add a GHA job with `-D_FORTIFY_SOURCE=3`.  eg. I am currently working with:

```
$ cat configure/os/CONFIG_SITE.Common.linux-x86_64
...
OP_SYS_CFLAGS += -fstack-protector-strong -Wformat -Werror=format-security
OP_SYS_CXXFLAGS += -fstack-protector-strong -Wformat -Werror=format-security
OP_SYS_CPPFLAGS += -Wdate-time -D_FORTIFY_SOURCE=3
```

mdavidsaver, 2024-06-14:
... llvm/clang also implements this builtins, hopefully in the same way.

physwkim, 2024-06-20:
I would like to report another Fortify issue. When starting `simDetectorIOC`, it works fine on its own. However, when attempting channel access via `medm` with the command `medm -x -macro "P=13SIM:,R=cam1:" simDetector.adl`, an error occurs immediately. Below is the gdb backtrace information:

```bash
(gdb) backtrace
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff784526e in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff78288ff in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff78297b6 in __libc_message_impl (fmt=fmt@entry=0x7ffff79ce765 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:132
#6  0x00007ffff7936c19 in __GI___fortify_fail (msg=msg@entry=0x7ffff79ce74c "buffer overflow detected") at ./debug/fortify_fail.c:24
#7  0x00007ffff79365d4 in __GI___chk_fail () at ./debug/chk_fail.c:28
#8  0x00007ffff79381b9 in __GI___strncpy_chk (s1=<optimized out>, s2=<optimized out>, n=n@entry=40, s1len=s1len@entry=0) at ./debug/strncpy_chk.c:26
#9  0x00007ffff7b62b8e in strncpy (__len=40, __src=<optimized out>, __dest=<optimized out>) at /usr/include/x86_64-linux-gnu/bits/string_fortified.h:95
#10 get_enum_strs (options=0x7fffbe9ff680, prset=0x7ffff7d88e20 <asynRSET>, ppbuffer=<synthetic pointer>, paddr=0x7ffe5c081f70) at ../db/dbAccess.c:192
#11 getOptions (pflin=0x0, options=0x7fffbe9ff680, poriginal=<synthetic pointer>, paddr=0x7ffe5c081f70) at ../db/dbAccess.c:415
#12 dbGet (paddr=paddr@entry=0x7ffe5c081f70, dbrType=dbrType@entry=11, pbuffer=pbuffer@entry=0x7fffbe9ff6d0, options=options@entry=0x7fffbe9ff680, nRequest=nRequest@entry=0x7fffbe9ff688, pflin=pflin@entry=0x0)
    at ../db/dbAccess.c:917
#13 0x00007ffff7b65860 in dbChannelGet (chan=chan@entry=0x7ffe5c081f68, type=type@entry=11, pbuffer=pbuffer@entry=0x7fffbe9ff6d0, options=options@entry=0x7fffbe9ff680, nRequest=nRequest@entry=0x7fffbe9ff688,
    pfl=pfl@entry=0x0) at ../db/dbChannel.c:630
#14 0x00007ffff7b7e458 in dbChannel_get_count (chan=chan@entry=0x7ffe5c081f68, buffer_type=<optimized out>, pbuffer=0x7ffe8c03ce30, nRequest=nRequest@entry=0x7fffbe9ffbf8, pfl=pfl@entry=0x0) at ../db/db_access.c:644
#15 0x00007ffff7ba7574 in read_reply (pArg=pArg@entry=0x7fffbe9ffc50, dbch=0x7ffe5c081f68, eventsRemaining=eventsRemaining@entry=1, pfl=pfl@entry=0x0) at ../rsrv/camessage.c:548
#16 0x00007ffff7ba86d8 in read_notify_action (mp=0x7fffbe9ffcf0, pPayload=0x7ffe8c038e28, client=0x7ffe84001238) at ../rsrv/camessage.c:745
#17 0x00007ffff7ba96fd in camessage (client=client@entry=0x7ffe84001238) at ../rsrv/camessage.c:2546
#18 0x00007ffff7ba6748 in camsgtask (pParm=0x7ffe84001238) at ../rsrv/camsgtask.c:116
#19 0x00007ffff7afea11 in start_routine (arg=0x7ffe8c041120) at ../osi/os/posix/osdThread.c:442
#20 0x00007ffff789ca94 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:447
#21 0x00007ffff7929c3c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

mdavidsaver, 2024-06-20:
Another fault reported [on tech-talk](https://epics.anl.gov/tech-talk/2024/msg00752.php) and https://github.com/epics-base/epics-base/pull/517#issuecomment-2179950539 with `get_enum_strs()` in `dbAccess.c`.

#517 updated with a prospective fix.

mdavidsaver, 2024-10-14:
fyi. github actions `ubuntu-latest` now points to `ubuntu-24.04`.  My builds against older Base versions abort on fortify source assertions by default.  I think I will start by setting `CMD_CPPFLAGS="-U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2"` for these jobs.

mdavidsaver, 2024-12-31:
```
*** buffer overflow detected ***: terminated
```

#517 fixes the known false positive.  It has been merged, but not yet released.  Those encountering fortify source aborts can either **update to at least 052a0c7e19580c18077a6fc5cf9f67ce75e3082b on the 7.0 branch**, or disable the fortify source checks for Base and all EPICS modules with:

```sh
cat << EOF >> epics-base/configure/CONFIG_SITE.local
OP_SYS_CPPFLAGS += -U_FORTIFY_SOURCE
EOF
```