epics-base/epics-base #521: mbbi / mbbo shoud thread RVAL as index when no *VL is defined

dirk-zimoch, 2024-06-19:
The mbbi and mbbo records have a field `SDEF` to "to save time if no states are defined".
In addition to "save time", it also changes the behavior (in an undocumented way): In the case that "states are defined", the record copies `RVAL` directly to `VAL` (or vice versa on case of the mbbo), treating it as an index instead of searching a matching `*VL` value. This behavior is greatly appreciated but should be documented.

But "no states are defined" means: All `*VL` are 0 (the default) AND all of the `*ST` state strings are empty. This is useful to "save time" when converting from and to strings.

But even in the case that state strings are defined, it would be useful to still treat `RVAL` as an index when none of the `*VL` values are defined. At the moment, either the `read()` (and `init_record()`) device support function has to know about this (for example by examining the `*VL` fields and then set `VAL` and return 2) or the user has to set `ZRVL=0`, `ONVL=1`, `TWVL=2`, etc. which is tedious.

This change would not break anything because when used with device support that sets `RVAL` and not `VAL`, having state strings defined but none of the `*VL` fields does not have any useful behavior right now.

It should be discussed if the meaning of `SDEF` can be changed to implement this (e.g. by defining a third state or using it as a bit mask for "strings defined" an "values defined") or if this is likely to break anything and the behavior change needs to be implemented differently.

ralphlange, 2024-06-19:
> treat `RVAL` as an index when none of the `*VL` values are defined

That sounds very reasonable. I can't think of a use case where not setting `*VL` should express something else.

anjohnson, 2024-06-24:
Looking at the current mbbi/mbbo sources, I'm not expressing my opinion on this yet:
* The `SDEF` = "States Defined" field of both records gets set when at least one of the string and/or value fields has been set. It's declared as a `DBF_SHORT`.
* The `mbbi::process()` and `mbbo::init_record()` routines currently skip comparing `RVAL` against the `..VL` fields and just copy `RVAL` to `VAL` when no states are defined.
* When there are states defined but `RVAL` doesn't match any of the `..VL` fields, both `mbbi::process()` and `mbbo::init_record()` set the `VAL` field to 65535, which is a kind of (likely undocumented) NaN value for ENUMs, but isn't always handled well.
* In both records' `put_enum_str()` methods, `SDEF` determines whether the code compares the new string value against the `..ST` fields at all.
* In `mbbo::cvt_dbaddr()` the `VAL` field's data type gets changed to `DBF_USHORT` if no states are defined. Then in `mbbo::convert()`, `SDEF` controls whether `VAL` is copied to `RVAL` or used as an index as @dirk-zimoch mentioned above. This lets clients write a `uint16` raw value directly to an mbbo that has no states defined without any `DBF_ENUM` handling getting in the way.
* There is no `mbbi::cvt_dbaddr()` method, so an mbbi `VAL` field is always a `DBF_ENUM` even when no states are defined.

How about making `SDEF` a new type of `DBF_MENU` with choices `None`, `Strings defined`, `Values defined` and `States defined`? The `put_enum_str()` methods need to know whether strings are defined, the `process()`/`init_record()` methods want to know about values, so they can examine bits 0 and 1 of `SDEF` respectively.

Should mbbi copy the mbbo approach and make `VAL` a `DBF_USHORT` if no states are defined?

There also appears to be a disagreement between the mbbo alarms generated when `val > 15` in `convert()` vs. in `checkAlarms()`; `convert()` checks `SDEF` and raises a `SOFT/INVALID` alarm while `checkAlarms()` doesn't and raises a `STATE/(UNSV)` alarm. It's too late for me to work out what should be happening there though.