epics-base/epics-base #522: New iocsh command: casgrep

anjohnson, 2024-06-21:
Here's an iocsh command I'd like to be able to run from the IOC console to help track down rogue CA clients:

`casgrep name-pattern level`

The existing `casr` command provides lots of useful information from the remote server (rsrv), but when you have ten or more clients connected to an IOC you usually want to filter the output, even more so when it's 100 clients. It's possible to use `casr 2 > /tmp/iocname-casr.log` and search through the log-file, but navigating that output can still be painful so something quicker would be nice.

I'm looking for a way to list all the clients currently connected to all PV names matching a glob pattern. The existing `casr` code already scans all the PV names connected to each client in turn, so the implementation of this should mostly involve copying code and adding the part to pattern-match the individual record names (using `epicsStrGlobMatch()` from libCom, see the dbgrep command code too for reference). There will need to be some reorganization of how/where things get printed since a client's info should only be displayed if that client is connected to at least one matching PV.

The level argument might not be needed, it would control how much information is displayed about each channel, but there might not be enough to make it worth controlling.

ralphlange, 2024-06-22:
+1

I always find `casr` with higher levels pretty useless on real-world IOCs.

To avoid code duplication: 
Looks like `casr` could simply call `casgrep *`.

mdavidsaver, 2024-06-27:
Could this be added through the `dbServer` API?

anjohnson, 2024-07-05:
> Could this be added through the `dbServer` API?

Maybe, it's definitely worth looking at that possibility if other IOC servers are likely implement it too (which would be a separate project).

grasswang, 2026-08-20:
I would like to work on this during the Codeathon in October 2026 Oct.

mdavidsaver, 2026-10-20:
> casgrep

> `epicsStrGlobMatch`

To save @simon-ess from a rant.  Since the pattern is name a regex, should the command name be `casmatch` (or `casm`)?

simon-ess, 2026-10-21:
To be consistent with my previous complaint, it could be `casglob`, but `casmatch` would be fine as well.

(Thank you for your consideration! 😄 )

closed 2026-11-12 by anjohnson