epics-base/epics-base #524: GHA: centos:7 repositories (re)moved

mdavidsaver, 2024-07-08:
centos:7 jobs are now failing with:

```
Could not retrieve mirrorlist http://mirrorlist.centos.org/?release=7&arch=x86_64&repo=os&infra=container error was
14: curl#6 - "Could not resolve host: mirrorlist.centos.org; Unknown error"
```

https://blog.centos.org/2023/04/end-dates-are-coming-for-centos-stream-8-and-centos-linux-7/

states:

> The packages will be archived on [vault.centos.org](https://vault.centos.org/) after June 30, 2024.

I find [some](https://stackoverflow.com/questions/78692851/could-not-retrieve-mirrorlist-http-mirrorlist-centos-org-release-7arch-x86-6) [hints](https://groups.google.com/g/cloudlab-users/c/Re6Jg7oya68) that files under `/etc/yum.repos.d/` can be edited to replace:

```diff
- mirrorlist=http://mirrorlist.centos.org/?release=$stream&arch=$basearch&repo=AppStream&infra=$infra
+ baseurl=http://vault.centos.org/$contentdir/$stream/AppStream/$basearch/os/
```

```sh
sed -i 's/mirrorlist/#mirrorlist/g' /etc/yum.repos.d/CentOS-*
sed -i 's|#baseurl=http://mirror.centos.org|baseurl=http://vault.centos.org|g' /etc/yum.repos.d/CentOS-*
```

I have not tested this.

tynanford, 2024-07-09:
> I have not tested this.

This fixed our centos 7 docker builds at LBL (used a slightly different [link](https://serverfault.com/a/1161847))

ralphlange, 2024-09-09:
Fixed it the same way as @tynanford for my OPCUA-Binary-Builder containers.
https://github.com/ralphlange/opcua-binary-builder/commit/031be11ab8b1ba972020185ea7647f8352c20d2f