epics-base/epics-base #529: msi cannot handle relative paths together with include paths

simon-ess, 2024-07-26:
If you have a substitution file like the following:
```
$ cat foo.substitutions
file some/relative/path/bar.template {
# some contents
}
```
then this can only be expanded by msi if the file `bar.template` is sitting in the given relative path to the current working directory, regardless of what include paths you pass. That is, you cannot do something like
```
$ msi -I /some/other/path foo.substitutions
```
(where `bar.template` would be sitting at `/some/other/path/some/relative/path`).

In other words, the include paths are only used if your substitutions file lists only the filename, such as
```
$ cat foo-2.substitutions
file bar.template {

}
```

This is due to the following: https://github.com/epics-base/epics-base/blob/91e1d0ab8022d610bb12b5536a7a6a21644d47f7/modules/database/src/ioc/dbtemplate/msi.cpp#L502 where we simply check if there exists a `/` in the path we are trying to load instead of what I presume was intended, that the _first_ character is a `/`.

simon-ess, 2024-07-26:
It is possible there is some reason for this behaviour, but it is certainly surprising.

ralphlange, 2024-07-26:
I would be all for allowing the relative paths to be added to include path elements.

(For ITER, I've been enforcing the use of only-file-names in substitution files because of this.)

anjohnson, 2024-07-28:
The line `file "H:/db/foo.db" { … }` works fine with the existing code on Windows, but wouldn't if you only tested the first character — `inputAddPath()` always prepends one of the search path entries when it sees a relative path, so it would become `./H:/db/foo.db` or worse.

Thus I don't agree with your guess about Marty's original intention, the `strchr()` test has been included in the `inputOpenFile()` routine in every version we have public sources for (msi 1.2, 1997). I'm not saying we couldn't change this behavior, but it would have to be done carefully, I doubt if our Windows tests use drive spec's but I'm sure some of our users do.

Handling filenames and paths can be tricky on Windows. MSI does use `OSI_PATH_LIST_SEPARATOR` correctly in `inputAddPath()` when parsing the `-I` options — on Windows the OS path separator is `;` to allow the paths to include drive letters, e.g. `-I D:/db;E:/db`.

Here are some of the other messy parts for Windows that the code doesn't currently detect or handle well at all:
1. `\db\foo.db`, `H:\db\foo.db` and `\\server\db\foo.d` don't have any `/` characters, so probably don't work with the existing code.
2. Is `H:foo.db` a relative path? It doesn't have either `/` or `\` characters in it.
3. A relative `file` parameter can't contain a drive spec, but an absolute one must be allowed to.

simon-ess, 2024-07-29:
@anjohnson That shows how long it has been since I have used Windows that I don't even really remember how paths work there... 😄 

Anyhow, am I to understand that this probably don't work anyhow in these cases for msi and windows? Would it be possible to add an `#ifdef _WIN32` (or whatever the correct conditional is) to allow for posix-ish systems to have better suppport for relative paths in MSI?

FreddieAkeroyd, 2024-08-08:
Hi, i'm now back from a few weeks A/L so can join the discussion when i've caught up on things that have happened at ISIS while I have been away.  Andrew is correct about the interpretation of `H:db/foo.db` and `H:/db/foo.db` - there are a list of such windows things at  e.g. https://learn.microsoft.com/en-us/dotnet/standard/io/file-path-formats    From a cmd window there is certainly a different current directory per drive letter, I am less certain how this behaves in a process but  https://learn.microsoft.com/en-us/windows/win32/fileio/changing-the-current-directory  suggests there is a per volume directory remembered when switching just drive letter of the process working directory, but i'd need to dig more or try an example to confirm how this behaves if you don't change directory but just use a relative path on a different volume, though i guess it may mirror the cmd window behaviour.

FreddieAkeroyd, 2024-08-09:
comment on `PathIsRelative`  https://stackoverflow.com/questions/26099361/is-it-a-winapi-bug-with-pathisrelative-function

It looks like each drive having its own current directory is now just a CMD.EXE thing https://devblogs.microsoft.com/oldnewthing/20101011-00

found some interesting path stuff on https://googleprojectzero.blogspot.com/2016/02/the-definitive-guide-on-win32-to-nt.html

FreddieAkeroyd, 2024-08-09:
I think a reasonable approach would be to define a new `isRelativePath()` function which on linux checks that the first character is not `/` and on windows calls `PathIsRelative()`. Though  `PathIsRelative` reports some of the edge cases like `H:db/foo.db` are absolute, i'm not sure we want to handle these as relative anyway e.g. is you pass an include path of `c:\somewhere` to msi and there is a relative path like `H:db/foo.db` in the template i'm not sure you should be looking on `c:\somewhere` for `foo.db` as `h:` was specified. So it would seem that `PathIsRelative` only reports the subset of relative paths we are most interested in as actual relative paths.

FreddieAkeroyd, 2024-08-09:
Or given our interest in only handling   `a\b` type relative paths, we could define our own `isRelative` for windows: as does not start with wither `/` or `\` and does not contain `:`