epics-base/epics-base #532: Duplicate targets for the same make rule when building application examples on Windows

levismith, 2024-08-09:
## Description

On Windows, building application skeletons generated with `makeBaseApp.pl` results in a number of warnings. For example, building the result of `makeBaseApp.pl -t example example`:

```
...
configure/RULES_DIRS:84: target 'src' given more than once in the same rule
configure/RULES_DIRS:84: target 'Db' given more than once in the same rule
configure/RULES_DIRS:84: target 'src.inc' given more than once in the same rule
configure/RULES_DIRS:84: target 'src.build' given more than once in the same rule
...
configure/RULES_DIRS:84: target 'Db.inc' given more than once in the same rule
configure/RULES_DIRS:84: target 'Db.build' given more than once in the same rule
...
```

Adding some debug output revealed `DIRS` was receiving a value of `src src Db Db`.

Wildcards in the template Makefiles for `exampleApp`, `iocApp` and `supportApp` are used to populate `DIRS` with values like "src", "fooSrc", "Db", and "FooDb":

```
DIRS += $(wildcard src* *Src*)
DIRS += $(wildcard db* *Db*)
```
in files
- `modules/database/src/template/top/exampleApp/Makefile`
- `modules/database/src/template/top/iocApp/Makefile`
- `modules/database/src/template/top/supportApp/Makefile`

On Windows, the patterns given to make's `$(wildcard ..)` command seem to be interpreted with case insensitivity. This results in a duplicate match for the the `src/` and `Db/` subdirectories.

## Workaround

I've fixed this on my end by renaming template directories to a lowercase `db/`, and modifying the patterns to include a `?` wildcard to prevent the duplicate match:
```
DIRS += $(wildcard src* ?*Src*)
DIRS += $(wildcard db* ?*Db*)
```

ralphlange, 2024-08-12:
I can't reproduce this.
------------------------

In my 'Developer Command Prompt' on 'Windows 10 Enterprise Version 22H2'
```
C:\Users\langer\Work Folders\Work\EPICS\test-532>perl ..\base\7.0\bin\windows-x64\makeBaseApp.pl -t example example

C:\Users\langer\Work Folders\Work\EPICS\test-532>dir
...
 Directory of C:\Users\langer\Work Folders\Work\EPICS\test-532

12/08/2024  12:13    <DIR>          .
12/08/2024  12:13    <DIR>          ..
12/08/2024  12:13               400 .gitignore
12/08/2024  12:13    <DIR>          configure
12/08/2024  12:13    <DIR>          exampleApp
12/08/2024  12:13               931 Makefile
               2 File(s)          1,331 bytes
               4 Dir(s)  46,044,672,000 bytes free
C:\Users\langer\Work Folders\Work\EPICS\test-532>"C:\Users\langer\bin\make.exe"
C:/Users/langer/bin/make -C ./configure install
make[1]: Entering directory 'C:/Users/langer/Work Folders/Work/EPICS/test-532/configure'
...
```
does not produce such errors.

Notes
------
* GNU Make 4.2.1
* Microsoft (R) C/C++ Optimizing Compiler Version 19.29.30153 for x64
* This is perl 5, version 26, subversion 1 (v5.26.1) built for MSWin32-x64-multi-thread

ralphlange, 2024-08-12:
Updated perl to
* This is perl 5, version 38, subversion 2 (v5.38.2) built for MSWin32-x64-multi-thread

Still can't reproduce.

ralphlange, 2024-08-12:
Using a newer make (from an MSYS2 installation, but in the same Command prompt)
* GNU Make 4.4.1

Still can't reproduce.

ralphlange, 2024-08-12:
What shell are you using?

anjohnson, 2024-08-12:
GNUmake's `$(wildcard ..)` function is implemented internally and I'm pretty sure it doesn't rely on the shell, but it might give different results on file-systems that are case-sensitive vs. case-preserving — examples [here](https://en.wikipedia.org/wiki/Case_preservation). Apparently NTFS file-systems can either be case-sensitive or case-preserving. It might be that the libc used to build GNUmake might have an effect too, so a MinGW-built make might give different results than a MSVC one.

levismith, 2024-08-12:
@ralphlange VS2022 Developer PowerShell for sure, and I think also with the VS2022 Developer Command Prompt.

@anjohnson This was with the GNUmake distributed with Strawberry Perl 5.38.2.2.

levismith, 2024-08-12:
Same errors using make from MSYS2 packages. It's the same, regardless of whether I run the build from Developer PowerShell, or Developer Command Prompt.

make from `UCRT64` environment, package `mingw-w64-ucrt-x86_64-make`
- mingw32-make: GNU Make 4.4.1; Built for x86_64-w64-mingw32

make from `MINGW64` environment, package `mingw-w64-x86_64-make`
- mingw32-make: GNU Make 4.4.1; Built for x86_64-w64-mingw32

anjohnson, 2024-08-12:
Okay so it's probably not the Make version then, what filesystem are these applications on, and can you try a different one?

levismith, 2024-08-13:
Sanity check...
- passed?

## Build environment

AWS EC2 Image
- instance type `t2.large` (*suggestion: use one with more cores, next time*)
- architecture `64-bit (x86)`
- AMI Name "Microsoft Windows Server 2022 Base"
- AMI ID `ami-0cdbb1e6297270d80`
- source `amazon/Windows_Server-2022-English-Full-Base-2024.07.10`

Install Build Tools for Visual Studio 2022, 17.10.5
- https://aka.ms/vs/17/release/vs_BuildTools.exe
- option "Desktop development with C++"

Install Strawberry Perl
- `strawberry-perl-5.40.0.10-64bit.msi`

Install Git for Windows
- `Git-2.46.0-64-bit.exe`

## Get epics-base sources
```
mkdir c:\epics
cd c:\epics
git clone https://github.com/epics-base/epics-base.git --recurse-submodules
```

## Build epics-base
- Using "x64 Native Tools Command Prompt for VS 2022"
   - %comspec% /k "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\Build\vcvars64.bat"
```
cd c:\epics\epics-base
set "EPICS_HOST_ARCH=windows-x64"
gmake
```
*build log for `epics-base` omitted due to length*

## Create and build example
```
mkdir c:\epics\example
cd c:\epics\example
perl c:\epics\epics-base\bin\windows-x64\makeBaseApp.pl -t example example
gmake
```

See attached file [532_cmd-log.txt](https://github.com/user-attachments/files/16592867/532_cmd-log.txt)

ralphlange, 2024-08-13:
Weird.
Same good result (no errors) occurs for me between Developer PowerShell, Developer Command Prompt and MSYS2, no matter which `make` I use.

```
PS C:\Users\langer\Work Folders\Work\EPICS\test-532> fsutil file queryCaseSensitiveInfo .
Case sensitive attribute on directory C:\Users\langer\Work Folders\Work\EPICS\test-532 is disabled.
```

anjohnson, 2024-08-21:
Core Group: @FreddieAkeroyd can you think what might be causing this problem? We can't reproduce it, and we think it has to be something to do with the file-system but we're stumped.

mdavidsaver, 2024-08-21:
fyi. Windows [`diskpart`](https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/diskpart) seems a rough super-set of *nix `mount` to query type and [info](https://www.minitool.com/news/see-drive-file-system-win-10.html) of active filesystems.

FreddieAkeroyd, 2024-08-21:
Just finishing for day, but I have seen something like this before. It seemed to me that it was make rather than the file system being case insensitive, NTFS is (usually) case preserving but not case sensitive. For me (make 4.3, windows cmd) 

`DIRS += $(wildcard src* Src*)` no errors 
`DIRS += $(wildcard src* src*)`  errors 
`DIRS += src Src` no errors 
`DIRS += src src` errors 

i had interpreted this as being a make "feature" but i hadn't looked into it any further, but can do a bit more checking.

it is mentioned above that `$(wildcard src* *Src*)` seems to give `src src` but i'm not sure offhand what option might do this as it seems for me `$(wildcard src* *Src*)` is just giving just `src`

FreddieAkeroyd, 2024-08-21:
Its an old article but may be relevant https://lists.nongnu.org/archive/html/help-gnu-utils/2005-02/msg00008.html

By default, GNU make retains its case sensitivity when comparing
target names and existing files or directories. It can be
configured, however, into a case preserving and case insensitive
mode by adding a define for HAVE_CASE_INSENSITIVE_FS to
config.h.W32.

FreddieAkeroyd, 2024-08-21:
no problem for me with strawberry perl
```
GNU Make 4.2.1
Built for x86_64-w64-mingw32
```
so it feels more like file system as you sugges, or some environment variable/other setting changing make behaviour?

konosubakonoakua, 2024-10-31:
Same here after set shell environment variable `export EPICS_HOST_ARCH=linux-arm`
```bash
make[2]: Entering directory '/epics/3.15/base/src/std/filters/test'
make[2]: Entering directory '/epics/3.15/base/src/std/rec/test'
../../../../configure/RULES_ARCHS:57: target 'inc.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'build.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'install.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'buildInstall.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'runtests.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'tapfiles.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'clean-tests.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'test-results.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'junitfiles.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:61: target 'linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:66: target 'O.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:82: target 'clean.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'inc.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'build.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'install.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'buildInstall.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'runtests.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'tapfiles.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'clean-tests.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'test-results.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:57: target 'junitfiles.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:61: target 'linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:66: target 'O.linux-arm' given more than once in the same rule
../../../../configure/RULES_ARCHS:82: target 'clean.linux-arm' given more than once in the same rule
perl -CSD /epics/3.15/base/bin/linux-arm/makeMakefile.pl O.linux-arm ../../../../..

```

anjohnson, 2024-10-31:
@konosubakonoakua I think that's likely to be a different problem, does your `configure/CONFIG_SITE` file also set `CROSS_COMPILER_TARGET_ARCHS=linux-arm`? The host architecture must not be added as a cross-target.

If you want to be able to cross-build for linux-arm on say a linux-x86_64 host but also build is as a native host architecture you should set `CROSS_COMPILER_TARGET_ARCHS=linux-arm` in your `configure/os/CONFIG_SITE.linux-x86_64.Common` file and not in the main `CONFIG_SITE`.