epics-base/epics-base #538: Missing function attribute EPICS_NORETURN

dirk-zimoch, 2024-08-23:
The function `testAbort()` does not return. It should be decorated accordingly. Therefore, a macro `EPICS_NORETURN` or similar needs to be defined in compilerSpecific.h for each compiler type. The gcc version would be `__attribute__((noreturn))` but I don't know what other compilers use.

dirk-zimoch, 2024-08-23:
The MSVC version seems to be `__declspec(noreturn)`.

mdavidsaver, 2024-08-24:
`EPICS_NORETURN` seems a reasonable a addition to `compilerDependencies.h` / `compilerSpecific.h`.

Another candidate application would be to `cantProceed()`.

#155

dirk-zimoch, 2024-08-27:
I have let gcc find out which functions don't return using `-Wsuggest-attribute=noreturn`. It found:
- `acctst()`
- `flexfatal()`
- `lerrif()`
- `lerrsf()`
- `cantProceed()`
- `epicsExit()`
- `testAbort()`
- `terminal_start()`
- `cast_server()`
- `rsrv_online_notify_task()`

The yacc and flex files already use a macro `NORETURN` defined for gcc as `__attribute__((noreturn))` for some functions that do not return but not for all (ending by calling other `NORETURN` functions).

To cover Windows complicates the task slightly because `__declspec(noreturn)` must be specified *before* the function and `__attribute__((noreturn))` *after* the function. Thus I ended up for gcc and clang with
```
#define EPICS_NORETURN(f) f __attribute__((noreturn))
```
and for MSVC with
```
#define EPICS_NORETURN(f) __declspec(noreturn) f
```
decorating the functions like this:
```
LIBCOM_API void EPICS_NORETURN(epicsExit(int status));
```
or this:
```
LIBCOM_API EPICS_NORETURN(void epicsExit(int status));
```
Both works.

dirk-zimoch, 2024-08-27:
See PR #540.

ericonr, 2025-04-11:
I believe this issue can be closed, right?

closed 2025-04-11 by dirk-zimoch