epics-base/epics-base #545: Build messages with `$(ECHO)` are suppressed on Windows due to spurious detection of `make-s` flag

levismith, 2024-08-30:
Build messages that should be output using `$(ECHO) <message>` are suppressed.

The file `configure/CONFIG_COMMON` sets a value for `ECHO` based on the result of single-letter flag detection. The actual value of `makeflags` is being set to `T_A=windows-x64` instead of an expected word containing all single-letter flags, and the subsequent use of `$(findstring ...)` results in a spurious detection of the `make-s` due to the presence of "s" within "T_A=windows-x64".

```
DEBUG $(MAKEFLAGS) =  -- T_A=windows-x64 TOP=../..
DEBUG $(makeflags) = T_A=windows-x64
DEBUG $(make-s) = s
DEBUG $(ECHO) = @:
```

See the attached build log.

## Notes

- GNU Make 4.4.41, x86_64-w64-mingw32

levismith, 2024-08-30:
Updating `checkFlags` to use the example shown at https://www.gnu.org/software/make/manual/html_node/Testing-Flags.html seems to improve detection:
```
# Just test MAKEFLAGS directly.
define checkflags
  make-$1 := $(findstring $1,$(firstword -$(MAKEFLAGS)))
endef
```

This works in the presence of long arguments. `$(ECHO)` messages are now output.
```
DEBUG $(MAKEFLAGS) =  -- T_A=windows-x64 TOP=../..
DEBUG $(make-s) =
DEBUG $(ECHO) = @echo
...
"Installing CFG file C:/epics/cfg/TOOLCHAIN.windows-x64.windows-x64"
```

Correct detection of `-s`:
```
DEBUG $(MAKEFLAGS) = s -- T_A=windows-x64 TOP=../../..
DEBUG $(make-s) = s
DEBUG $(ECHO) = @:
```

anjohnson, 2024-08-30:
Hi Levi, which version of Base are you building?

Please read the full discussion in PR #395 which was merged into the 7.0 branch last June — if you're building an older branch that change probably hasn't been applied there. That PR was tested on older GNU Make versions as well as 4.4.1 but I haven't tried anything newer.

Note that macOS still ships with GNU Make 3.81 (Apple won't upgrade since later versions came under the GPLv3) and we don't want to break builds using that, although that's the earliest version we do support.

anjohnson, 2024-09-03:
Okay, with GNUmake 4.4.1 I'm seeing that `makeflags` incorrectly contains the `T_A=…` setting when no flags are given inside an `O.<target>` build directory, but that doesn't happen in the other source dir's. Our previous manual tests of #395 were evidently not checked properly on Windows, thanks for the heads-up!

Different versions of GNU Make that we wanted to support behaved differently before 4.0, which is why this has been so tricky to get right. I'm currently leaning towards using Perl to filter the `MAKEFLAGS` variable, which requires moving all the checkflags stuff into `configure/CONFIG_BASE`. Expect to see a PR fixing this very soon.

commit f47e1d94 (2024-10-30): Use Perl for filtering MAKEFLAGS properly

Requires moving the checkflags code into CONFIG_BASE
as that's where FIND_TOOL gets defined.

Fixes #545
closed 2024-10-30 by anjohnson