epics-base/epics-base #548: Setting a (short) string in the VAL field of an lso record fails

ralphlange, 2024-09-10:
The IOC fails loading the lso record in the following test database

```
record(lso, "lso:short") {
 field(VAL, "this is a short string")
}

record(lso, "lso:long") {
 field(SIZV, 100)
 field(VAL, "this is a very long string that would NEVER fit into 40 characters")
}

record(stringout, "so") {
   field(VAL, "this is a short string")
}
```

with 

```
ralph@debian-12:~/work/EPICS$ ./3.15.9/base/bin/linux-x86_64/softIoc -d test.db 
Can't set "lso:short.VAL" to "this is a short string"
Error at or before ")" in path "."  file "test.db" line 2
Can't set "lso:long.VAL" to "this is a very long string that would NEVER fit into 40 characters"
ErrordbLoadRecords: failed to load test.db

ralph@debian-12:~/work/EPICS$ ./7.0/base/bin/linux-x86_64/softIoc -d test.db 
Can't set "lso:short.VAL" to "this is a short string"  : Error (512,11)
ERROR:  at or before ')' in path "."  file "test.db" line 2

 2 |   field(VAL, "this is a short string")

Can't set "lso:long.VAL" to "this is a very long string that would NEVER fit into 40 characters"  : Error (512,11)
ERROR: 
 7 |   field(VAL, "this is a very long string that would NEVER fit into 40 characters")

ERROR failed to load 'test.db'
ERROR: Failed to load: test.db
```

On the other hand, the stringout record can be set using the VAL field.

See tech-talk conversation _add link when the tech-talk archive is up again_

hjunkes, 2024-09-10:
If I read the Record reference manual correctly, then the VAL field of the lso cannot be set via DCT?

```
Field | Summary | Type | DCT | Default | Read | Write | CA PP
-- | -- | -- | -- | -- | -- | -- | --
VAL | Current Value | STRING or CHAR[SIZV] | No |   | Yes | Yes | Yes
```

mdavidsaver, 2024-09-10:
> field(VAL, "this is a short string")

This is expected.  lso/lsi are similar to waveformRecord in that VAL is an array.  Array fields can't be initialized prior to `iocInit()` as this is when memory for that array is allocated.

ralphlange, 2024-09-10:
Ok, then the Record Reference Manual is wrong and needs to be updated (in the pod file/s).

mdavidsaver, 2024-09-10:
> Can't set "lso:short.VAL" to "this is a short string"  : Error (512,11)

This `Error (512,11)` aka. `S_dbLib_badField` is coming from the default case in `dbPutString()` since the field type of `VAL` is `DBF_NOACCESS`.

https://github.com/epics-base/epics-base/blob/804577075120218c2a38d457906e6a3182659e82/modules/database/src/ioc/dbStatic/dbStaticLib.c#L2642

mdavidsaver, 2024-09-10:
```diff
--- a/modules/database/src/ioc/dbStatic/dbStaticLib.c
+++ b/modules/database/src/ioc/dbStatic/dbStaticLib.c
@@ -2778,7 +2778,9 @@ long dbPutString(DBENTRY *pdbentry,const char *pstring)
             }
         }
         break;
-
+    case DBF_NOACCESS:
+        dbMsgPrint(pdbentry, "Too early to set array field");
+        /* fall through */
     default:
         return S_dbLib_badField;
     }
```

Changes the error line to:

```
Can't set "lso:short.VAL" to "this is a short string" Too early to set array field : Error (512,11)
```

"dbPutString() can't set array field" would be strictly correct, but seems to me practically less useful.

@anjohnson @ralphlange Any thoughts on wording?

ralphlange, 2024-09-10:
Mmmmh... "too early" sounds a lot like wallclock time.

Maybe: "Can't set array field before iocInit()"?

anjohnson, 2024-09-10:
> "To initialize a string output record it is simplest to set the VAL field directly;" Nah.

Oops, that phrase was evidently copied from the stringout record type, where it was correct. The wording of this whole lso record description could use some rewriting, fixing it properly isn't a 5-minute job.

Ralph's wording above for the message looks good to me.

Why are the error messages including `Error (512,11)`? That's a fairly recent regression, older IOCs knew that error code and used to output this instead:
```
Can't set "lso.VAL" to "This is a string"  : Bad Field value
```

mdavidsaver, 2024-10-16:
Core Meeting:

> Maybe: "Can't set array field before iocInit()"?

This seems better.

> Can't set array field before iocInit(). ... Consider using a '["..."]' input link instead.