epics-base/epics-base #549: DB parsing errors cause "Failed to load" message to appear twice

ralphlange, 2024-09-10:
Using EPICS 7 with a test database

```
record(lso, "lso1") {
  field(VAL, "this is a short string")
}
record(lso, "lso2") {
  field(VAL, "this is a short string")
}
record(lso, "lso3") {
  field(VAL, "this is a short string")
}
```

the IOC shows

```
ralph@debian-12:~/work/EPICS$ ./7.0/base/bin/linux-x86_64/softIoc -d test.db 
Can't set "lso1.VAL" to "this is a short string"  : Error (512,11)
ERROR:  at or before ')' in path "."  file "test.db" line 2

 2 |   field(VAL, "this is a short string")

Can't set "lso2.VAL" to "this is a short string"  : Error (512,11)
ERROR: 
 5 |   field(VAL, "this is a short string")

Can't set "lso3.VAL" to "this is a short string"  : Error (512,11)
ERROR: 
 8 |   field(VAL, "this is a short string")

ERROR failed to load 'test.db'
ERROR: Failed to load: test.db
```

I.e., repeating the final error message.

mdavidsaver, 2024-09-10:
> ... maybe `softMain.cpp` should stop printing anything?

Removing redundant output if fine with me, so long as the on-error behavior remains (`errIf()` throws).

ericonr, 2025-01-07:
```
    }catch(std::exception& e){
        errlogFlush();
        std::cerr<<ERL_ERROR ": "<<e.what()<<"\n";
        epicsExit(2);
        return 2;
    }
```

`errIf` throws, and its exception has to include *some* message, so that catch block can print it. If that message shouldn't be `"Failed to load:"`, what should it be? `"Failed to parse databases"`?

mdavidsaver, 2025-01-28:
> `"Failed to parse databases"`?

Fine by me.

fyi. another pattern I have seen is to default a special/local exception sub-class like eg. `handled_exception` which is then caught and calls `epicsExit(2)` without printing.