epics-base/epics-base #553: RT Linux and default priorities

mdavidsaver, 2024-09-26:
Remembering @kasemir s [talk](https://conference.sns.gov/event/448/contributions/667/) about [PREEMPT_RT Linux](https://lwn.net/Articles/990985/) and some reading I am doing today, I wanted to highlight what I think is an easy "gotcha" for someone configuring IOCs on such a system.

One of the changes with PREEMPT_RT is the introduction of [threaded interrupt handlers](https://wiki.linuxfoundation.org/realtime/documentation/technical_details/threadirq) which puts hardware interrupt handling into a kernel thread, which then has a priority.  The current default seems to be `50`.

The current mapping for OSI to OSS priorities on Linux __puts most IOC threads at a higher priority than interrupts handling__ theads.  Which seems to me likely to surprise.

eg. running an IOC on my laptop finds NIC IRQ handling at a lower priority than the low priority callback thread.  Which to me seems backwards.

```
$ ps -eLo rtprio,pid,lwp,comm,cmd|sort -n
...
    50     882     882 irq/184-iwlwifi [irq/184-iwlwifi:queue_12]
    50     883     883 irq/185-iwlwifi [irq/185-iwlwifi:exception]
    50     944     944 irq/186-AudioDS [irq/186-AudioDSP]
    51  795513  795582 timerQueue      ../../bin/linux-x86_64/nasaAcq ./node01.cmd
    51  795513  795587 CAC-event       ../../bin/linux-x86_64/nasaAcq ./node01.cmd
    53  795513  795583 CAC-UDP         ../../bin/linux-x86_64/nasaAcq ./node01.cmd
    58  795513  795527 cbLow           ../../bin/linux-x86_64/nasaAcq ./node01.cmd
    59  795513  795568 scan-10         ../../bin/linux-x86_64/nasaAcq ./node01.cmd
    60  795513  795569 scan-5          ../../bin/linux-x86_64/nasaAcq ./node01.cmd
    61  795513  795570 scan-2          ../../bin/linux-x86_64/nasaAcq ./node01.cmd
    62  795513  795571 scan-1          ../../bin/linux-x86_64/nasaAcq ./node01.cmd
 ...
```

mdavidsaver, 2024-09-26:
Questions...

- Am I correct in seeing this as a priority inversion?
- Should this be addressed by default in epics-base?
- If yes, then how?
- If no, then what?  Mandate (effectively require) mcoreutils and/or admins needing to change TISR priorities?

mdavidsaver, 2024-09-26:
I also notice a couple of common daemons which get top priority.

```
...
...
    99  FF  793704  793704 ntpd            /usr/sbin/ntpd -p /run/ntpd.pid -c /etc/ntpsec/ntp.conf -g -N -u ntpsec:ntpsec
    99  RR    1293    1295 rtkit-daemon    /usr/libexec/rtkit-daemon
```

kasemir, 2024-09-26:
I see interrupt threads at 50 as well, and one "migration" thread per CPU core at the very top.

Changing the default priorities used by EPICS, or tweaking the mapping from the epics thread priorities to the posix priorities might upset existing systems.

Maybe it's best to stress the need to review your thread settings, not just those created by the IOC but also system threads, in some piece of documentation (where??).

Would you adjust the threads created by the IOC, for example using `mcoreutils`? Or would you leave the IOC alone and move the interrupt threads?

anjohnson, 2024-09-26:
These are the RT threads from one of our Libera Brilliance+ IOCs which run with RT enabled, on a Linux Kernel version 5.4.0:
```
RTPRIO   PID   LWP COMMAND         CMD
    10  1286  1289 errlog          liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    10  1286  1290 logRestart      liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    10  1286  1298 taskwd          liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    12  1286  1392 CAS-UDP         liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    14  1286  1393 CAS-beacon      liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    16  1286  1391 CAS-TCP         liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    18  1286 10018 CAS-event       liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    18  1286 10859 CAS-event       liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    20  1286  1394 save_restore    liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    20  1286  1436 CAS-client      liberaIOC iocBoot/ioc2s40bpm3/st.cmd
...
    20  1286  9367 CAS-client      liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    22  1286  1406 CAC-event       liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    34  1286  1415 CAS-event       liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    34  1286  6535 CAS-event       liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    36  1286  1416 CAS-client      liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    36  1286  6536 CAS-client      liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    50   110   110 irq/123-aerdrv  [irq/123-aerdrv]
    50   111   111 irq/124-aerdrv  [irq/124-aerdrv]
    50   112   112 irq/125-aerdrv  [irq/125-aerdrv]
    50    12    12 idle_inject/0   [idle_inject/0]
    50  1286  1303 dbCaLink        liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    50    16    16 idle_inject/1   [idle_inject/1]
    50    22    22 idle_inject/2   [idle_inject/2]
    50    28    28 idle_inject/3   [idle_inject/3]
    50   328   328 irq/141-mei_me  [irq/141-mei_me]
    51  1286  1395 CAC-event       liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    57  1286  1304 timerQueue      liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    58  1286  1300 cbLow           liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    59  1286  1291 timerQueue      liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    59  1286  1292 timerQueue      liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    59  1286  1346 scan-10         liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    60  1286  1348 scan-5          liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    61  1286  1356 scan-2          liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    62  1286  1357 scan-1          liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    63  1286  1301 cbMedium        liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    63  1286  1360 scan-0.5        liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    64  1286  1361 scan-0.2        liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    65  1286  1362 scan-0.1        liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    66  1286  1305 scanOnce        liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    69  1286  1299 timerQueue      liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    70  1286  1302 cbHigh          liberaIOC iocBoot/ioc2s40bpm3/st.cmd
    99    11    11 migration/0     [migration/0]
    99    17    17 migration/1     [migration/1]
    99    23    23 migration/2     [migration/2]
    99    29    29 migration/3     [migration/3]
    99    99    99 watchdogd       [watchdogd]
```

I agree that the priority which Linux chooses for the interrupt threads seems unfortunate for us, although I'm not sure where I would want to see them within our thread priorities.

Our Posix osdThread.c implementation has code that queries the libc routines `sched_get_priority_min()` and `sched_get_priority_max()` at initialization time to map EPICS priorities to the OS ones, so we might have some flexibility to adjust levels there. We do have to remember that RTEMS also uses the Posix implementation of osdThread too, so any changes we make will have to work with that too.

anjohnson, 2024-09-26:
I don't think I'd want to touch the priorities of any Linux OS threads, especially of the kernel ones.

mdavidsaver, 2024-09-29:
> ... code that queries the libc routines `sched_get_priority_min()` ...

Thank you Andrew, I was forgetting about this.  In the context of `systemd.exec` `LimitRTPRIO=50` does have the desired effect of keeping the `rtprio` of all of the IOC threads below 50.

```
epics> epicsThreadShow
            NAME       EPICS ID   LWP ID   OSIPRI  OSSPRI  STATE
          _main_ 0x556726681070   963319      0       0       OK
          PVXTCP 0x5567266a43a0   963322     18       9       OK
          PVXUDP 0x55672668e0d0   963323     16       8       OK
          errlog 0x5567267489d0   963324     10       5       OK
          taskwd 0x556726748e80   963325     10       5       OK
      timerQueue 0x5567267495d0   963326     70      35       OK
...
```

hjunkes, 2026-10-19:
Is this just a documentation issue?
Can we sort this out together with 
https://github.com/epics-base/epics-base/issues/984

mdavidsaver, 2026-10-20:
> 7\. Can we sort this out together with
>         [#984](https://github.com/epics-base/epics-base/issues/984)

It seems reasonable to try.