epics-base/epics-base #555: printfRecord cannot handle long string (CHAR array) input

dirk-zimoch, 2024-10-14:
```
record(lsi, "LongStringIn") {
  field(SIZV, "200")
  field(INP, ["This is a test"])
}

record(printf, "Printf") {
  field(SIZV, "200")
  field(FMT,  "%s")
  field(INP0, "LongStringIn.VAL$ CP")
}
```
The content of the "Printf" record is not "This is a test" but "84" (the ASCII code of 'T')

anjohnson, 2024-10-14:
I believe you need to use `"%ls"` in the format string if your link provides a long string (array of char). Please try that and confirm. If it works we might want to add something to the documentation in the .dbd file to make that clearer.

dirk-zimoch, 2024-10-15:
Yes, `"%ls"` works. Thanks.
I still do not find it very intuitive that `"%s"` does not work.

anjohnson, 2024-10-20:
It's needed for the same reason that `caget` needs a `-S` flag to print long strings, which just aren't native to the dbAccess or Channel Access APIs. This is a price we continue to pay for keeping compatibility with older EPICS releases when we introduced them.

anjohnson, 2024-10-30:
Core group: No better solution yet.

closed 2024-10-30 by anjohnson