epics-base/epics-base #557: Inconsistency with CP link and asynchronous record

mdavidsaver, 2024-10-20:
The following contrived DB example ends with `a:done.VAL == 0`.  I expect that the final result should be `3` (`1|2`).  The initial update of the `a:calc.INPB` CA link is dropped since that record is still processing as a result of `a:one.PINI`.

To my mind this is a bug since it violates the guarantee of eventual consistency made for EPICS subscriptions.

```
record(longout, "a:one") {
    field(PINI, "YES")
    field(VAL, "1")
    field(FLNK, "a:calc")
    field(TPRO, "1")
}
record(longout, "a:two") {
    field(VAL, "2")
    field(TPRO, "1")
}
record(calcout, "a:calc") {
    field(INPA, "a:one MS")
    field(INPB, "a:two CP MS")
    field(CALC, "A|B")
    field(ODLY, "1")
    field(OUT , "a:done PP MS")
    field(TPRO, "1")
}
record(longin, "a:done") {
    field(TPRO, "1")
}
```

```
$ softIoc -d async-calink.db 
Starting iocInit
############################################################################
## EPICS R7.0.8.2-DEV
## Rev. R7.0.8.1-146-g5fa58673ab437f74fd36-dirty
## Rev. Date Git: 2024-10-20 09:34:51 -0700
############################################################################
_main_: dbProcess of 'a:one'
_main_: dbProcess of 'a:calc'
iocRun: All initialization complete
scanOnce: dbProcess of Active 'a:calc' with RPRO=0
epics> cbLow: dbProcess of 'a:done'
epics> dbgf a:done
DBF_LONG:           0 = 0x0   
epics>
```

mdavidsaver, 2024-10-20:
I see this behavior as related to what I addressed with `dbPutField()` and `PUTF / RPRO` (https://code.launchpad.net/~epics-core/epics-base/+git/prop-putf/+merge/336468) to allow chaining of async. records following a remote PUT.

In short, I think that the `scanOnceCallback()` calls made from `dbCa.c` should somehow set `PUTF` (or `RPRO`) to trigger the re-processing behavior if the target record is busy.

This would in effect generalize `PUTF` to mean that processing was initiated by an event external to the local database.

anjohnson, 2024-10-20:
Why are you setting `a:calc.ODLY`?

`a:calc.VAL` is zero at the end of processing your database, while `a:calc.A` is 1. The reading of the INP links and the CALC expression evaluation doesn't get delayed by the `ODLY`, that only delays writing the output, and that's what's causing the problem — `VAL` gets calculated before the delay is started. Not setting the `ODLY` field does result in `a:calc` being reprocessed and `a:done` being set to 3:

```
Cox$ softIoc -d issue-557.db 
Starting iocInit
############################################################################
## EPICS R7.0.8.2-DEV
## Rev. R7.0.8.1-94-g152d306ad87dfd6fc899-dirty
## Rev. Date Git: 2024-10-02 09:22:00 -0500
############################################################################
_main_: dbProcess of 'a:one'
_main_: dbProcess of 'a:calc'
_main_: dbProcess of 'a:done'
iocRun: All initialization complete
scanOnce: dbProcess of 'a:calc'
scanOnce: dbProcess of 'a:done'
epics> dbgf a:done
DBF_LONG:           3 = 0x3   
```

mdavidsaver, 2024-10-20:
> Why are you setting `a:calc.ODLY`?

> The following contrived DB example ...

Emphasis on _contrived_.  The actual case where I encounter this is a longoutRecord with async device support which is quite a bit more complex.

anjohnson, 2024-10-22:
Removing `PINI` from `a:one` also results in both `a:calc` and `a:done` processing and getting the value 3. Doing much during PINI can be tricky, the [menuPini documentation](https://epics.anl.gov/base/R7-0/8-docs/menuPini.html) says about the `YES` choice:

> `initHookAfterScanInit` — All records and links have been initialized but the scan threads and CA server are not running yet, nor have CA links been connected up. The initHook `initHookAfterInitialProcess` immediately follows this processing.

Is setting `PINI` a vital part of your original complex example? Many APS IOCs trigger processing or enable functionality after `iocInit()` using `dbpf` commands in the st.cmd file, which may have been a workaround to the complexity.

I'm not sure that eventual consistency has ever been guaranteed from a single CA monitor event (I doubt there's any documentation which says that). There is a comment in dbCa.c saying that the dbCaTask must not lock the record as that would violate lock order, so setting `PUTF` or `RPRO` looks tricky to do. Do you have specific ideas for fixing this issue yet?

mdavidsaver, 2024-10-23:
> Is setting `PINI` a vital part of your original complex example?

No.  In the example PINI and a fixed delay serve to setup a race condition which always causes bad behavior.  I will see if I can come up with a different example which avoids these.

The real database where I think I am seeing the same effect has 32x pairs of CP links (so 64x links in total) triggering recompute of `ASLO` and `AOFF` for 32x aoRecord.  There are also several hundred other CA links to complicate the timing.  (it is a big database...)  What I see is that sometimes a couple of the pairs will not compute correctly, leaving ALSO or AOFF with an incorrect value.  Seems something like a race condition.

In the short term I will rework my database to avoid having CP links in a processing chain upstream of an asynchronous record.

In the longer term, I think this is a bug worth fixing.

> I'm not sure that eventual consistency has ever been guaranteed from a single CA monitor event ...

Not per event, per subscription.  Another way of stating this guarantee for an OPI is that.  While a subscription buffer may overflow with a burst of activity, when the dust settles the display will show the last value published (aka. the current value).

This is not what my example shows.

mdavidsaver, 2024-12-27:
I think this behavior is the same as @anjohnson described in https://bugs.launchpad.net/epics-base/+bug/1841634 .  Indeed [this comment](https://bugs.launchpad.net/epics-base/+bug/1841634/comments/1) sounds quite familiar.  I'm still not thrilled about the potential fixes wrt. wasted work, but years later and I can think of no better solution.  With CA, there may not be one.  Maybe with PVA links something could be arranged with per-subscription flow control.

anjohnson, 2025-06-11:
Fixed by #559 at a4bc0db6e

closed 2025-06-11 by anjohnson