epics-base/epics-base #562: [7.0.8.1] failure during Debian build process

picca, 2024-11-06:
Hello, I just got this bug from Debian

  https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1084828

the full build log is here

  https://people.debian.org/~sanvila/unshare/unstable-20241008/epics-base_7.0.8.1+dfsg1-2_amd64-20241008T170205.030Z

I instrumented the code and there is a failure with this network configuration

```
make[1]: Entering directory '/<<PKGBUILDDIR>>'
# display the network configuration
ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo 
       valid_lft forever preferred_lft forever
```

It works on the buildd array  with this configuration

```
# display the network configuration
ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq state UP group default qlen 1000
    link/ether aa:00:00:71:0c:2b brd ff:ff:ff:ff:ff:ff
    inet 128.31.0.50/24 brd 128.31.0.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 2603:400a:ffff:bb8::801f:32/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::a800:ff:fe71:c2b/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
```

So my question is: is it normal ?

picca, 2024-11-06:
The code of the test

```
static void test()
{
    PVDatabasePtr master = PVDatabase::getMaster();
    ChannelProviderLocalPtr channelProvider = getChannelProviderLocal();
    StandardPVFieldPtr standardPVField = getStandardPVField();
    string properties;
    ScalarType scalarType;
    string recordName;
    properties = "alarm,timeStamp";
    scalarType = pvDouble;
    recordName = "exampleDouble";
    PVStructurePtr pvStructure;
    pvStructure = standardPVField->scalar(scalarType,properties);
    PVRecordPtr pvRecord = PVRecord::create(recordName,pvStructure);
    master->addRecord(pvRecord);
    pvRecord = master->findRecord("exampleDouble");
    {
     	pvRecord->lock();
	pvRecord->process();
	pvRecord->unlock();
    }
    if(debug) {cout << "processed exampleDouble "  << endl; }
    ServerContext::shared_pointer ctx =
	startPVAServer("local",0,true,true);
    testOk1(ctx.get()!=0);
}

MAIN(testPVAServer)
{
    testPlan(1);
    test();
    return 0;
}
```

So I imagine thaht the failure is in `startPVAServer`

ralphlange, 2024-11-06:
So the part of the test that creates a network server fails if there is no network.

Somewhat understandable, but the test might better be skipped in that case.

picca, 2024-11-06:
> So the part of the test that creates a network server fails if there is no
> network.

and what about the lo interface ?

> Somewhat understandable, but the test might better be skipped in that case.

the last solution :))

ralphlange, 2024-11-06:
> and what about the lo interface ?

By default, EPICS doesn't use localhost/loopback.
EPICS name resolution depends on unique replies, and using localhost would always create two replies from any server/IOC running on the same host (assuming standard setup with one network interface plus localhost).

picca, 2024-11-06:
> By default, EPICS doesn&#39;t use localhost/loopback.
> EPICS name resolution depends on unique replies, and using localhost would
> always create two replies from any server/IOC running on the same host
> (assuming standard setup with one network interface plus localhost).

ok this is clear

thanks

Frederic

picca, 2024-11-06:
> By default, EPICS doesn&#39;t use localhost/loopback.
> EPICS name resolution depends on unique replies, and using localhost would
> always create two replies from any server/IOC running on the same host
> (assuming standard setup with one network interface plus localhost).

Is it possible to skip a test via the command line ?

This way I should skip this test  during the build but not in the Debian CI system.

ralphlange, 2024-11-06:
You could try if explicitly binding the server to localhost via environment setting (`EPICS_PVAS_INTF_ADDR_LIST=localhost`) avoids the failure during the build.

mdavidsaver, 2024-11-06:
fyi. while these tests can't use the loopback interface, each test process only talks to itself over some broadcast capable non-loopback interface.  For my test environments I add a dummy software bridge.

mdavidsaver, 2024-11-06:
> `EPICS_PVAS_INTF_ADDR_LIST=localhost`

@ralphlange This will not work for PVA.

picca, 2024-11-07:
I tried to add the usuals

testTodoBegin
testTodoEnd

but it does not help,

terminate called after throwing an instance of 'epics::pvData::detail::ExceptionMixed<epics::pvData::BaseException>'
  what():  No (specified) network interface(s) available.
On line 206 of ../../src/server/serverContext.cpp
/<<PKGBUILDDIR>>/lib/linux-x86_64/libpvAccess.so.7.1.7(_ZN5epics6pvData6detail14ExceptionMixedINS0_13BaseExceptionEEC2IPKcEET_S7_i+0xcc) [0x7f5d96ead98c]
/<<PKGBUILDDIR>>/lib/linux-x86_64/libpvAccess.so.7.1.7(+0x86f7a) [0x7f5d96e6ef7a]
/<<PKGBUILDDIR>>/lib/linux-x86_64/libpvAccess.so.7.1.7(_ZN5epics8pvAccess13ServerContext6createERKNS1_6ConfigE+0x234) [0x7f5d96f0a9c4]
/<<PKGBUILDDIR>>/lib/linux-x86_64/libpvAccess.so.7.1.7(_ZN5epics8pvAccess14startPVAServerERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEibb+0x6fe) [0x7f5d96f0be5e]
./testPVAServer(+0x2569) [0x55ad58b67569]
./testPVAServer(main+0x1f) [0x55ad58b6724f]
/lib/x86_64-linux-gnu/libc.so.6(+0x29d68) [0x7f5d967a2d68]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f5d967a2e25]
./testPVAServer(_start+0x21) [0x55ad58b67281]

testPVAServer.t ...... 
No subtests run 

Test Summary Report
-------------------
testPVAServer.t    (Wstat: 0 Tests: 0 Failed: 0)
  Parse errors: No plan found in TAP output
Files=6, Tests=129,  0 wallclock secs ( 0.03 usr  0.01 sys +  0.11 cusr  0.03 csys =  0.18 CPU)
Result: FAIL
-------------------

the TODO does not guard against premature failure :).

for now I will comment the test.

Do you have a function which test if there is network like the TODO couple? a sort of skipIfNoNetwork.

or maybe the test should just be in a try except ?

Fred

picca, 2024-11-07:
Just for info, the Official Debian build array does not allow network during the build in order to avoid access to internet and compromission of the build artefacts.

this is why these unusual failure.

thanks

Fred

mdavidsaver, 2024-11-07:
> but it does not help,

The test process is being terminated by an uncaught c++ exception.  The C `testTodo*()` functions will not have effect.

Also, this sort of failure isn't what "todo" is about.  A more direct comparison would be the `export EPICS_TEST_IMPRECISE_TIMING=YES` to extend some timeouts when the test environment has very relax scheduling common to the shared hosting of the free CI services.

I think you would do better to either disable the offending test executable entire (patch that Makefile), or if permitted add a dummy software bridge to the test host.  (If unprivileged network namespace creation is allowed, then this could also be accomplished without `sudo`)

```sh
sudo brctl addbr brX
sudo ifconfig brX 10.254.254.1/24 up
# run tests
# later cleanup if desired
sudo ifconfig brX down
sudo brctl delbr brX
```

picca, 2024-11-07:
Hello

I unactivated these tests in order to PASS al other tests during the build process with the network configuration of Debian

-TESTPROD_HOST += testServerContext
-testServerContext_SRCS += testServerContext.cpp
-TESTS += testServerContext
-

-TESTPROD_HOST += testPVAServer
-testPVAServer_SRCS += testPVAServer.cpp
-testHarness_SRCS += testPVAServer.cpp
-TESTS += testPVAServer

the best in my case is to avoid requesting root privileges during the build process, it cause other issues.

the patch is here

https://salsa.debian.org/science-team/epics-base/-/blob/master/debian/patches/0002-skip-expected-failing-test.patch?ref_type=heads

Cheers

Fred

mdavidsaver, 2024-11-07:
> I unactivated these tests ...

This is probably for the best.

> If unprivileged network namespace creation is allowed, then this could also be accomplished without `sudo`

fyi. I you do want to try:

```sh
cat << EOF > isotest.sh
#!/bin/sh
set -e -x
ifconfig lo 127.0.0.1/8 up
brctl addbr br0
ifconfig br0 10.1.1.1/24 up
ip addr
exec make tapfiles
EOF
unshare -rnU sh isotest.sh
ret=$?
make -s test-results
exit $ret
```

anjohnson, 2024-11-08:
If anyone wants to work on this, having tests check for the environment variable `EPICS_TEST_NO_NETWORK=YES` might be a good way to go in the long run (continuing @mdavidsaver's suggestion from yesterday with a suggested variable name) as it's easier to have users set that than help them create a fake network interface. We probably have quite a few tests that assume there will be a network available, @picca's patch linked above would be a good place to start when identifying them.

I see this as a potential Codeathon issue.