epics-base/epics-base #564: Improve handling of negative numbers with DBR_ULONG

alesk22, 2024-11-12:
When attempting to assign negative values to PV with DBR_ULONG, operation succeeds and value is interpreted as positive. This is not something that all users might expect:

```
$ cainfo MYIOC:application:rdwr:v:ULong 
MYIOC:application:rdwr:v:ULong
    State:            connected
    Host:             dev:5064
    Access:           read, write
    Native data type: DBF_DOUBLE
    Request type:     DBR_DOUBLE
    Element count:    1

$ caput MYIOC:application:rdwr:v:ULong -1
Old : MYIOC:application:rdwr:v:ULong 0
New : MYIOC:application:rdwr:v:ULong 4.29497e+09

caput MYIOC:application:rdwr:v:ULong -6
Old : MYIOC:application:rdwr:v:ULong 4.29497e+09
New : MYIOC:application:rdwr:v:ULong 4.29497e+09

```

It turns out that this is caused by the way glibc / `strtoul[l]` functions parses numerical strings (used by `epicsParse*`).

First it should be decided if this is, in fact, a bug, not a feature.

anjohnson, 2024-11-12:
The `U` in `DBR_ULONG` stands for unsigned, so all values stored in `DBF_ULONG` fields are by definition positive. The term "long" is an unfortunate misnomer nowadays, EPICS dates back to the time when an `int` could be 16 bits and `long` was 32. We don't actually have an IOC record type that has a `DBF_ULONG` VAL field, so I don't think you can be using a built-in record type in an EPICS IOC for the above examples.

Note that the channel appears in your `cainfo` output as holding `DOUBLE` values. That's because Channel Access doesn't actually support unsigned types, and the decision was made that fields that inside the IOC that are declared as `DBF_ULONG` would be presented to CA clients as being 64-bit floating point so the full range of uint32 values could be transported without wrapping values that have bit 31 set to negative numbers.

What outcome do you want when you put -1 to a ULong field? The above is what I expect to happen.

alesk22, 2024-11-13:
Expanding on  first comment:

**Suggested change**

Put operation to fail with value unchanged. Current behavior could be confusing to CS operators that do not know about two's complement or are unfamiliar with value's underlying (unsigned) data type.

**More information**

* on the PV / IOC from examples: record has custom type with `DBR_` set to `DBR_ULONG` at runtime. 
* on analysis: at least using `caput`, value is transported via Channel Access as string and is parsed in the IOC, which uses appropriate `epicsParseULong`.

The risk is, of course, that there already are control systems or operators that use this as a feature. This should be assessed and issue handled accordingly.

mdavidsaver, 2024-11-13:
> That's because Channel Access doesn't actually support unsigned types

The PVA protocol does support unsigned types natively.

> ... there already are control systems or operators that use this as a feature ...

This reality is unfortunately not in question.  Past attempts to make parsing stricter or to remove implicit casting have resulted in complaints, and left us with eg. the [`dbConvertStrict`](https://github.com/epics-base/epics-base/blob/1f8851a5dc511846dc41a93b128e737479215a2c/modules/database/src/ioc/dbStatic/dbStaticRun.c#L411-L413) flag.  Which defaults to "lazy".

anjohnson, 2024-11-13:
My mistake, `caput` does write the value as a string, so this code is currently used:
```C
    const char *from = (const char *) f;
    epicsUInt32 *to = (epicsUInt32 *) t;
    char *end;
    long status;

    if (*from == 0) {
       *to = 0;
       return 0;
    }
    status = epicsParseUInt32(from, to, 10, &end);
    if (status == S_stdlib_noConversion ||
        (!status && (*end == '.' || *end == 'e' || *end == 'E'))) {
        /*
         * Convert via double so numbers like 1.0e3 convert properly.
         * db_access pretends unsigned long is double.
         */
        double dval;

        status = epicsParseFloat64(from, &dval, &end);
        if (!status &&
            dval >=0 &&
            dval <= UINT_MAX)
            *to = dval;
    }
```
There's code there to even accept strings like `1.0e3`; evidently users were entering integer values in scientific notation and expected that to work, so the IOC was modified to support that.

Other clients may attempt to convert the user's value to the native type reported by CA and put the result. They will succeed in converting `-1` to a double so on the IOC the `cvt_d_ul()` routine will be called, which uses a cast similar to the `*to = dval` assignment above with the same result.

If you need a field that can hold integers up to 2^32 but you want it to detect and reject negative values, could you make the field a `DBF_INT64` and have the database or record type reject values entered that are out of your valid range?

alesk22, 2024-11-14:
> If you need a field that can hold integers up to 2^32 but you want it to detect and reject negative values, could you make > the field a DBF_INT64 and have the database or record type reject values entered that are out of your valid range?

This is precisely the idea I got when thinking about alternative approach. It should even work with `DBF_UINT64` for underlying  `ULONG` to match value's signedness info. Thanks for the confirmation.

Based on  the comments, I shal close this issue as invalid.

closed 2024-11-14 by alesk22