epics-base/epics-base #567: Propagating AMSG through links

exzombie, 2024-11-14:
I'd like to attach an error message to alarm status of records whose values come from an aSub. I treat aSub records as "private", in the sense that users have no business looking at or touching their fields. They only interact with records that are linked from an aSub and represent its inputs and outputs. When the subroutine encounters an error, I set its alarm status and use the MSS attribute on output links to make the problem visible to clients.

In cases where a situation that a message needs to be shown to the user arises often, we have always used a dedicated message record, with the aSub writing to it through one of its outputs. However, for "this should (almost) never happen" kind of situations, we just write to the IOC log and tell people to go look there. They shouldn't need to do it too often. However, it would be *really* convenient to be able to use AMSG for this purpose and let people look at that field instead of having to look at the IOC console.

But, as noted above, the person seeing the alarm will see it on the interface records, and is probably not even aware that there is a subroutine generating that alarm. For this reason, it would be good to be able to propagate AMSG through a link. I propose to extend the behavior of the MSS link attribute to do that. I don't think it's necessary to introduce another attribute: if you are overwriting STAT, you either also like overwriting AMSG, or you just don't care about AMSG. That's my view, at least.

And that's why I'm opening this issue instead of a pull request: looking at the code, extending `recGblInheritSevr()` doesn't look difficult, but I would first like to know whether others share my view of using the MSS attribute for this.

anjohnson, 2024-11-14:
Seems reasonable, how would the link severity flags control whether the message gets copied? We currently support `NMS`, `MS`, `MSI` and `MSS`, do we need more choices (hopefully not)? I’m delegating the thinking about this to you and/or other commenters…

exzombie, 2024-11-14:
As I wrote, it makes sense to me that MSS should propagate AMSG, while MS should not, because if you care about propagating STAT, you also care about propagating AMSG.

mdavidsaver, 2024-11-15:
> it would be _really_ convenient to be able to use AMSG ... instead of having to look at the IOC console.

This was one of my goals when adding AMSG...

> ... it would be good to be able to propagate AMSG through a link. I propose to extend the behavior of the MSS ...

I think this is a good idea.

> ... extending `recGblInheritSevr()` doesn't look difficult ...

Since this function does appear in the public `recGbl.h` header, the question of backwards compatibility must be asked.
 
My quick search on github.com doesn't show any calls to `recGblInheritSevr()` outside of epics-base (unlike `recGblSetSevr()`).  @anjohnson @ralphlange Do you know of any external usage?

ralphlange, 2024-11-15:
I don't know of any usage of `recGblInheritSevr()`.

exzombie, 2024-11-15:
Would it make sense to add `recGblInheritSevrMsg()` and mark `recGblInheritSevr()` as deprecated, without removing it immediately?

anjohnson, 2024-11-15:
> As I wrote, it makes sense to me that MSS should propagate AMSG, while MS should not, because if you care about propagating STAT, you also care about propagating AMSG.

Agreed. What about `MSI` - [Maximize Severity if Invalid](https://docs.epics-controls.org/en/latest/process-database/EPICS_Process_Database_Concepts.html#maximize-severity-attribute)? That doesn't propagate STAT but just sets it to `LINK_ALARM` like `MS`. How useful would it be to able to propagate AMSG (with or without also copying STAT) only when SEVR is raised to `INVALID_ALARM`?

> Would it make sense to add `recGblInheritSevrMsg()` and mark `recGblInheritSevr()` as deprecated, without removing it immediately?

We didn't flag the original `recGblSetSevr()` as deprecated when we added `recGblSetSevrMsg()` and I don't think we should do that here either. I'm not aware of any external users of `recGblInheritSevr()` but it's an API which might be used by a private link support implementation, so IMHO that's an acceptable API inflation.

exzombie, 2024-11-19:
As I understand it, MSI is just like MS (except that it only acts on `INVALID_ALARM`), so, just like MS, it shouldn't propagate AMSG.

PR #568 (2024-12-11): Propagate AMSG through MSS links
Fixes #567

MS and MSI links do not propagate STAT and therefore do not propagate AMSG, either. CA, CP and CPP links also do not propagate AMSG, but the reason is technical: the message is not available over Channel Access. Apparently, it is available over PVA (`pvmonitor` shows it without doing anything special) so it should be possible to propagate it over PVA links, but unless I'm mistaken that belongs into the pvxs module.

The new `recGblInheritSevrMsg()` function is not a formatting function (unlike `recGblSetSevrMsg()`) because it is supposed to be used  [... cut]
closed 2024-12-11 by anjohnson