epics-base/epics-base #576: EPICS 3.16 introduces monitoring issue

spreiter, 2024-12-18:
Dear all,

I am suffering currently a very random problem and am a bit puzzled. The problem first showed itself after upgrading from EPICS 3.14. Unfortunately I can barely give a test scenario, because the underlying problem does not seem to manifest each time.

### Setup

One system provides a heartbeat PV (bi). The PV is flipped by another application with 2Hz.

The second system hosts a soft IOC (no dbd file loaded), which creates a copy of the heartbeat PV (CPP on INP) plus PVs counting the time in each state. This way I make sure that the original system is up by checking whether both counters are below a certain threshold. If this is true, I interpret this as an ALIVE system, otherwise DEAD. In the .db file attached this final PV is named `$(IOC):hb_check:S:cur`.

[heartbeat_receiver.db.txt](https://github.com/user-attachments/files/18181467/heartbeat_receiver.db.txt)

The db file is loaded for 40 external systems.

### Observation

The first system is running on at all times. When starting the IOC on the second system, it looks like in some cases – 1 of 100 – the copy PV does connect to the external system, but it does not process on each monitoring update. I can put on the PROC field and the copy updates accordingly.

In my tests I started the second IOC around 500 times.

In one case where on of the heartbeats was reported dead I killed the IOC with SIGILL and found a single thread being in the `epicsSocketAccept` function. 

### Narrowing it down

I tested multiple EPICS versions. All 3.15 versions seem to be fine.

Starting with EPICS 3.16.0.1 the problem arises. I tested even single commits and found that `92641e916` is the last working version, `c9d889ef3` does not connect at all and `6ac1e050b` is the first commit of 3.16 branch, that shows the mentioned behavior.

I also checked EPICS 7. Here I even had the impression that compiling EPICS again results in different better or worse results. In the following I plotted MTBF. Each plot is created after new compilation of EPICS 7.0.7.1 and running the same test 300 times. Without entries no error happened. This behavior is also not clear to me at all..

[image]

----

I know that I can give no nice example to reproduce. I also started the the heartbeat PVs and the second system for testing and the problem never shows up.

But maybe someone has an idea, what might be the underlying problem. I am happy to follow up any ideas.

What I also tried:
- change PV to calc
- set PINI to NO
- set `callbackParallelThreads 8` in start script
- set `callbackSetQueueSize 10000` in start script

Hope to hear from you :)

Best,
Simon

mdavidsaver, 2024-12-24:
I think what is described sounds like an issue with CA links.  The only relevant change to dbCa in the range 92641e916 ... c9d889ef3 was my change to prevent a queue overflow by bounding the number of re-scans from `CP` links.
  This overflow would be of the Once queue, which is separate from the callback queues, so changes to the two callback tunables you mention would not effect anything.

Still, I have a hard time imagining that a set of 2Hz counters could trigger this limiting.

Could you test with https://github.com/epics-base/epics-base/pull/559 ?

Try setting `copy:$(IOC):heartbeat:S:cur.TPRO=1` and watch the debug logs to differences.

mdavidsaver, 2024-12-24:
Also, while I don't think it is relevant

```
   field(CALC,"a=0?0:b+1")
...
   field(CALC,"a=1?0:b+1")
```

In the second expression I would change `a=1` to `a!=0` to close a potential logic hole if the heartbeat PV were sometimes eg. `2`.  (unless this was intentional)

spreiter, 2024-12-26:
I tested it with _mdavidsaver:cp-link-rpro_, i.e.`R7.0.8.1-97-g53a943dba`. Run 80 failed this time, causing 4 of the heartbeat PVs to not work properly.
```
copy:PXD:O02S4:heartbeat:S:cur
copy:PXD:O10S3:heartbeat:S:cur
copy:PXD:O11S1:heartbeat:S:cur
copy:PXD:O12S2:heartbeat:S:cur
```

I attached the output of the IOC: [ioc-log.txt](https://github.com/user-attachments/files/18253500/ioc-log.txt)

After the starting error messages I do not see anything else.
```
scanOnce: Ring buffer overflow
dbCa.c failed to queue scanOnce
dbCa.c failed to re-queue scanOnce
errlog: lost 29 messages
```

The result looks similar when I tested `022b8d468`, so without the changes in #559.

And why is it called `scanOnce: dbProcess of ..` after init when it is part of callback? Or does it always says *Once* ?

ralphlange, 2024-12-26:
The `scanOnce` mechanism consists of a queue (whose size is fixed but can be set at IOC initialization) and a worker thread.

Many situations (e.g., incoming CA monitor updates for CPP links and incoming CA puts or a PINI setting) require a passive record to be processed, but can't do the processing in their own thread context because of execution time limitations / possible blocking.
In those cases, the request to process is put into the scanOnce queue, where the scanOnce worker thread will pick it up and process the requested record a bit later.

Overflowing the scanOnce queue means that records that are supposed to be processed are not. Often not a fatal situation, but that depends entirely on the application.

Newer versions of Base have a "watermark" feature that allows the current and highest use of the database's internal queues to be shown. (These values are also exposed to the iocStats Device Support.)

Your way out is increasing the queue size (from the startup script, before `iocInit`) until you stop seeing overflows.
And of course finding the root cause, i.e., what overloads the scanOnce queue. Note: At IOC startup, any PINI setting or CP input link will put a record in the scanOnce queue. If you have many external CP input links, network trouble or a remote IOC reboot can also cause a lot of records to be put in the scanOnce queue at the same time.

mdavidsaver, 2024-12-26:
> ... Note: At IOC startup, any PINI setting or CP input link will put a record in the scanOnce queue ...

@spreiter I note that all of the records in your `heartbeat_receiver.db` have `PINI` set.  I think all of these are unnecessary and can be left unset.

ralphlange, 2024-12-26:
> @spreiter I note that all of the records in your `heartbeat_receiver.db` have `PINI` set. I think all of these are unnecessary and can be left unset.

In case you set them PINI to avoid records being UDF/INVALID after startup, maybe setting UDFS is an option.

spreiter, 2024-12-27:
Earlier I also tried it with `PINI=NO`, but this did not improve the situation. I guess it might decrease the number of tasks put in the scanOnce queue, maybe it was not enough.

I run with https://github.com/epics-base/epics-base/commit/022b8d468 again and checked the log counting the previously mentioned messages. The first run pointed to `dbCa.c failed to re-queue scanOnce`, when the test fails, the second run also showed this message, when I works. But it could also be coincidence and the message is issued by another PV. In this run 6 heartbeat PVs did not work as expected.

```
[-] test   1 scanOnce error: overflows= 567, queue_fails= 647, requeue_fails=   9, scanOnceSetQueueSize=1000
[-] test   5 scanOnce error: overflows=  22, queue_fails= 667, requeue_fails=   0, scanOnceSetQueueSize=1000
[-] test   8 scanOnce error: overflows=   2, queue_fails= 122, requeue_fails=   0, scanOnceSetQueueSize=1000
[-] test  10 scanOnce error: overflows= 242, queue_fails= 299, requeue_fails=   5, scanOnceSetQueueSize=1000
[-] test  13 scanOnce error: overflows=  48, queue_fails= 138, requeue_fails=   0, scanOnceSetQueueSize=1000
[-] test  17 scanOnce error: overflows=   1, queue_fails=  12, requeue_fails=   0, scanOnceSetQueueSize=1000
[-] test  18 scanOnce error: overflows=  39, queue_fails= 112, requeue_fails=   0, scanOnceSetQueueSize=1000
[-] test  23 scanOnce error: overflows=  23, queue_fails= 134, requeue_fails=   0, scanOnceSetQueueSize=1000
[-] test  30 scanOnce error: overflows= 360, queue_fails= 210, requeue_fails= 280, scanOnceSetQueueSize=1000
[-] test  36 scanOnce error: overflows=  13, queue_fails= 517, requeue_fails=   0, scanOnceSetQueueSize=1000
[-] test  39 scanOnce error: overflows=  12, queue_fails= 642, requeue_fails=   0, scanOnceSetQueueSize=1000
[-] test  41 scanOnce error: overflows=  91, queue_fails= 120, requeue_fails=   0, scanOnceSetQueueSize=1000
[-] test  43 scanOnce error: overflows=   1, queue_fails=   1, requeue_fails=   0, scanOnceSetQueueSize=1000
[-] test  44 scanOnce error: overflows=  85, queue_fails= 199, requeue_fails=   1, scanOnceSetQueueSize=1000
```

I repeated the test multiple times and it seemed that every time the test fails re-queuing fails as well at least once. Sadly I was proven wrong. So there is no clear correlation.

I increased the `scanOnceSetQueueSize` to 10k and it looks like the issue is gone. Neither do I see a failure nor any scanOnce error reports. However I am only checking a few PVs in this test and I have the impression the problem remains hidden.

I also tried this approach with https://github.com/epics-base/epics-base/commit/6ac1e050b. It turns out good. This seems to be the effective configuration option.

The documentation states:
> The default is 1000. It should rarely be necessary to override this default.

The IOC does not provide too many PVs (4303 in total). 2326 of those have PINI=YES. Is there a drawback to increase the default?

----

> In case you set them PINI to avoid records being UDF/INVALID after startup, maybe setting UDFS is an option.

Yes, I think this may generally be good. Maybe this could be pointed out in the docs. I was not aware of this overhead of PINI. I will revise my database.

ralphlange, 2024-12-27:
I'm sorry - I just checked the source code and I was wrong regarding PINI.
PINI=YES processing at IOC startup is done directly from the main thread and does not use the `onceQueue`.
(see [lines 570ff. in `iocInit.c`](https://github.com/epics-base/epics-base/blob/169948967fe833a58ab88f20e0b86021f3d26f95/modules/database/src/ioc/misc/iocInit.c#L571))

Nevertheless, if you see overflows, increasing the queue size is the right fix.
The once queue is a ring buffer, a single entry consists of three pointers = 24 bytes. A slight memory increase is all to expect as a side effect. Not a big thing, especially on virtual memory platforms.

spreiter, 2024-12-27:
Thanks for checking.

I wonder why I did not observe this behavior in EPICS 3.14. Was the default queue size ever changed?

And to my understanding, the onceScan queue is filled by CA monitor updates. The worker "only" processes the PV (e.g. calculation, alarm, etc). But should not the next update on this CPP link add another entry in this queue?

mdavidsaver, 2024-12-27:
> ... Was the default queue size ever changed?

Not for quite some time 3e0445f2adcc873dc658f4576eee94da62e41b15 (circa 3.13).

> ... my change to prevent a queue overflow ...

For reference, the launchpad.net MR for this change to the Once queue and dbCa to introduce the `caLink::scanningOnce` counter.   https://code.launchpad.net/~epics-core/epics-base/dbscan-update/+merge/245680

mdavidsaver, 2024-12-27:
> ... should not the next update on this CPP link add another entry in this queue?

My change was to add a per-link counter `caLink::scanningOnce` so for each CP link, that record only need appear a single time in the Once queue.  The idea being to bound the Once queue size to the number of links instead of the number of links multiplied by the CA subscription update rate.  The individual counters are then allowed to saturate (at 5) instead of overflowing the shared Once queue.

For this reason, setting `scanOnceSetQueueSize` to a number greater than the number of CP/CPP links should prevent an overflow of the Once queue due to these links.  (the Once queue is also used for RPRO of busy async. records, and potentially other uses in external driver code, making it hard to produce peak usage...)

spreiter, 2024-12-27:
OK. So to cover the worst case the Once queue should be around 5 times the number of records with CP/CPP links, right?

During the PINI phase, when CA links are not yet established, what happens? Does the IOC simply ignore the processing or wait for the CA connection to be established, up to a timeout?

mdavidsaver, 2024-12-28:
> OK. So to cover the worst case the Once queue should be around 5 times the number of records with CP/CPP links, right?

`1x` should be sufficient.  `5` is the limit on the number of times each link will, after processing, automatically re-add that record to the Once queue.