epics-base/epics-base #583: Files not found when building base

dirk-zimoch, 2025-01-09:
Several make dependencies seem to be incorrect or incomplete.

In a "clean" environment, e.g. after `make distclean`, `make $EPICS_HOST_ARCH` fails with missing headers files in modules/libcom/test/:

```
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE   -DBUILDING_SHARED_LIBRARIES  -D_X86_64_  -DUNIX  -Dlinux
 -O3 -g   -Wall  -mtune=generic   -m64  -I. -I../O.Common -I. -I. -I.. -I../../../../include/compiler/gcc
-I../../../../include/os/Linux -I../../../../include   -c ../epicsLoadTest.cpp
../epicsLoadTest.cpp:19:10: fatal error: epicsInstallDir.h: No such file or directory
 #include "epicsInstallDir.h"
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
```

But `make host` completes without problems.

The same happens when `make` is called with any of the defined cross architectures, for example in my case `make yocto40-aarch64`. Again, it fails when compiling epicsLoadTest.cpp for the host architecture.

It seems that epicsInstallDir.h has not yet been created by the time libcom/test is built if `make` is called with the name of an architecture.

This part works if I use `make build.$EPICS_HOST_ARCH`, but later it fails in database/test/std/rec:

```
"../../../../../../bin/RHEL8-x86_64/msi" -D    -I. -I.. -I../O.Common -I../../../../../../db -o ../O.Common/simmTest.db
-S../simmTest.substitutions  > simmTest.db.d
/bin/sh: ../../../../../../bin/RHEL8-x86_64/msi: No such file or directory
Inflating database from ../simmTest.substitutions 
"../../../../../../bin/RHEL8-x86_64/msi"    -I. -I.. -I../O.Common -I../../../../../../db -o simmTest.db
-S../simmTest.substitutions 
/bin/sh: ../../../../../../bin/RHEL8-x86_64/msi: No such file or directory
make[5]: *** [../../../../../../configure/RULES.Db:498: ../O.Common/simmTest.db] Error 127
make[5]: Leaving directory '/home/zimoch/epics/epics-base-7.0-pure/modules/database/test/std/rec/O.RHEL8-x86_64'
```

So in this case, `msi` has not been installed yet (but it had already been built as modules/database/src/ioc/O.$EPICS_HOST_ARCH/msi).

The same happens with `make build` (no specific architecture).
On the other hand, `make install.$EPICS_HOST_ARCH` completes.

----

If VxWorks is in the cross architectures list, a simple `make` (without targets) or `make install` fails as well. In my case:

```
/opt/VxWorks/VxWorks6.9/gnu/4.3.3-vxworks-6.9/x86-linux2/bin/ccppc 
-isystem /opt/VxWorks/VxWorks6.9/vxworks-6.9/target/h   -DCPU=PPC604  -DvxWorks=vxWorks
-include /opt/VxWorks/VxWorks6.9/vxworks-6.9/target/h/vxWorks.h  -DBUILDING_libCom_API   -O2
-Wall -Werror-implicit-function-declaration      -mcpu=604 -mstrict-align    -fno-builtin   -I. -I../O.Common
-I. -I../osi/compiler/gcc -I../osi/compiler/default -I. -I../osi/os/vxWorks -I../osi/os/posix -I../osi/os/default
-I.. -I../as -I../bucketLib -I../calc -I../cvtFast -I../cppStd -I../cxxTemplates -I../dbmf -I../ellLib -I../env -I../error
-I../fdmgr -I../flex -I../freeList -I../gpHash -I../iocsh -I../log -I../macLib -I../misc -I../osi -I../pool -I../ring -I../taskwd
-I../timer -I../yacc -I../yacc -I../yajl -I../../../../include/compiler/gcc -I../../../../include/os/vxWorks -I../../../../include
-I/opt/VxWorks/VxWorks6.9/vxworks-6.9/target/h -I/opt/VxWorks/VxWorks6.9/vxworks-6.9/target/h/wrn/coreip
-c ../as/asTrapWrite.c
../as/asTrapWrite.c:28:23: error: dbChannel.h: No such file or directory
```

Again, it seems that not all header files are installed yet.

Without VxWorks, `make` without arguments completes.

dirk-zimoch, 2025-01-10:
`make $EPICS_HOST_ARCH` is broken since commit 615ba73.

anjohnson, 2025-01-10:
Confirmed, I ran `make distclean` then `make $EPICS_HOST_ARCH` and got this error on darwin-aarch64:
```
...
c++ -o iocshTest  -L/Users/anj/Software/epics/7.0/lib/darwin-aarch64       -arch arm64  -L/opt/homebrew/opt/readline/lib         iocshTest.o    -lCom   
c++          -DBUILDING_SHARED_LIBRARIES    -DUNIX  -Ddarwin      -O3 -g   -Wall      -arch arm64   -fno-common  -I. -I../O.Common -I. -I. -I.. -I../../../../include/compiler/clang -I../../../../include/os/Darwin -I../../../../include         -c ../epicsLoadTest.cpp
../epicsLoadTest.cpp:19:10: fatal error: 'epicsInstallDir.h' file not found
#include "epicsInstallDir.h"
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
gmake[4]: *** [../../../../configure/RULES_BUILD:268: epicsLoadTest.o] Error 1
gmake[4]: Leaving directory '/Users/anj/Software/epics/7.0/modules/libcom/test/O.darwin-aarch64'
gmake[3]: *** [../../../configure/RULES_ARCHS:62: darwin-aarch64] Error 2
gmake[3]: Leaving directory '/Users/anj/Software/epics/7.0/modules/libcom/test'
gmake[2]: *** [../../configure/RULES_DIRS:85: test.darwin-aarch64] Error 2
gmake[2]: Leaving directory '/Users/anj/Software/epics/7.0/modules/libcom'
gmake[1]: *** [../configure/RULES_DIRS:85: libcom.darwin-aarch64] Error 2
gmake[1]: Leaving directory '/Users/anj/Software/epics/7.0/modules'
gmake: *** [configure/RULES_DIRS:85: modules.darwin-aarch64] Error 2
```

Interestingly `make host` doesn't suffer the problem, it was added in Base-3.14.12.4 and in general I'd recommend using that instead, it's shorter to type and works even if the environment variable isn't set.

Looking at the build failure, there is a dependency in `modules/libcom/test/Makefile` which apparently gets applied when building the default make target but not for `make $EPICS_HOST_ARCH`:

```makefile
epicsLoadTest$(DEP): epicsInstallDir.h
```

This patch seems to solve that particular build failure, but it's a copy/paste fix and similar issues come up again later on:
```patch
diff --git a/modules/libcom/test/Makefile b/modules/libcom/test/Makefile
index a3893661f..339d269d7 100644
--- a/modules/libcom/test/Makefile
+++ b/modules/libcom/test/Makefile
@@ -345,6 +345,7 @@ rtemsTestData.c : $(TESTFILES) $(TOOLS)/epicsMakeMemFs.pl
        $(PERL) $(TOOLS)/epicsMakeMemFs.pl $@ epicsRtemsFSImage $(TESTFILES)
 
 epicsLoadTest$(DEP): epicsInstallDir.h
+epicsLoadTest$(OBJ): epicsInstallDir.h
 
 # use INSTALL_LOCATION instead of FINAL_LOCATION since test executables are not installed.
 epicsInstallDir.h: $(TOP)/configure/CONFIG_SITE*
```

I didn't think that would be the best solution and gave up on it before adding 9 similar lines to `modules/database/test/ioc/db/Makefile` and looked for a fix in the `configure` area instead. The Static Pattern rule on line 93 of `configure/RULES_DIRS` is what gets invoked by `make $EPICS_HOST_ARCH`.

The patch below gives the `$(ARCHS)` targets a separate rule which expand to `install.%`. However in `make $EPICS_HOST_ARCH` the current rule expands to the `$(dirArchTargets)` for each sub-directory in `DIRS`, so the implication is that one of those isn't working properly when it gets down to using `configure/RULES_BUILD`. Further investigation into that and why the specific commit broke the build is probably worth doing, but I'm out of time for that myself right now.

```patch
diff --git a/configure/RULES_DIRS b/configure/RULES_DIRS
index f47df4a53..ad976a581 100644
--- a/configure/RULES_DIRS
+++ b/configure/RULES_DIRS
@@ -90,7 +90,8 @@ $(DIRS) $(dirActionTargets) $(dirArchTargets) $(dirActionArchTargets):
 beforeActions = $(addprefix before-,$(ACTIONS))
 $(beforeActions):
 
-$(ARCHS) $(ACTIONS) $(actionArchTargets) :%: \
+$(ARCHS) :%: install$(DIVIDER)%
+$(ACTIONS) $(actionArchTargets) :%: \
     $(foreach dir, $(DIRS), $(dir)$(DIVIDER)%)
 
 .PHONY : $(DIRS) all host rebuild
```

dirk-zimoch, 2025-01-13:
The VxWorks build using `make` without arguments is broken since commit f6cd43b7.

dirk-zimoch, 2025-01-13:
That was a local change to `TARGET_CPPFLAGS` that I did for (experimantal) VxWorks 7 compatibility.