epics-base/epics-base #595: RTEMS "init" system

mdavidsaver, 2025-02-05:
In looking at #528 and other PRs adding to `posix/rtems_init.c`, what I see is the incremental and organic growth of what in the *nix world would be handled by an init system (a la. sysv scripts or systemd).  This early-ish startup code has already proven a maintenance headache, and further growth seems likely to continue this trend.

So I would like to propose an alternative design as a straw man.

In no particular order, the early startup tasks I see are:

- file system setup (in-memory, local, and remote)
- network setup
- time sync.  (aka. NTP client)

The potential inputs I see are:

- Compiled in in-memory FS images (or fragments of images)
- bootloader info (u-boot, motload, ...)
- DHCP response(s)

Design goals:

- Enable site customization
- Enable application specific customization
- ... preferably without C code changes
- Bound the complexity of rtems_init.c

Open questions:

- Treat DHCP as a one-shot, or attempt to handle address changes or lease expiration?
- Is the following too much for the simple iocsh or RTEMS shell?  (time for a "real" scripting language?)

mdavidsaver, 2025-02-05:
How to go about this...

From what I can see, the migration to libbsd brings a move toward configuration files.  I like this trend.

A search through `posix/rtems_init.c` turns writes to `/etc/hosts` and `/etc/dhcpcd.conf` already.  #528 would add `/etc/services`, `/etc/ntp.conf` and `/etc/leap-seconds`.  Also there is the computed path to the application `st.cmd`.

What I don't like, is seeing these files encoded within C files, both because it complicates editing, and because there is no easy, obvious, way for site or application customization.

This leads me to think about expanding `epicsRtemsFSImage` currently used for files with unit tests (or maybe `rtems_tarfs_load()` if it has learned how to `mkdir()` reasonably).

Instead of a single image, how about an ordered, extensible, sequence of image fragments?

(eg. as `static const` discoverable using `__attribute__((section("..."))))` iteration)

The idea being that Base could contribute a "default" set of files which might at least serve as a starting point.  Then support modules or applications could provide image fragments which overwrite some of these files.

Now how to accommodate dynamic information from bootloaders?

To my mind, this should be injected as environment variables explicitly,
and so as iocsh variables implicitly.

Then next question is how to inject these values into configuration files?

The "obvious" EPICS-y answer being macLib and/or MSI.

Specifically, I think the pieces which would need to be added are:

- Mechanism for compiling in file system fragments.
  - Need to establish conventional layout
- Code to translate bootloader config/info into environment variables.
  - Naming convention...
- Add IOC shell commands (or identify equivalent from RTEMS shell):
  - `mount` (generalize `nfsMount` to also handle `tftp` and others)
  - `cat`
  - `msi`
  - `dhcp`
  - `ntpd`
  - file/directory manipulations (eg. `mkdir`, `rm`, ...)
  - Teach iocsh to expand wildcard lists of file names.  (eg. `< /etc/boot.d/*.cmd`)

The process in `rtems_init.c` would become:

0. Set default values for some environment variables
1. Translate bootloader info into environment variables
2. Populate in-memory FS
3. `iocsh("/etc/boot.cmd")`
4. `main()`

So customization would be injected via `/etc/boot.cmd`, which would be responsible for initiating network configuration (if necessary), mounting file systems, and starting NTPd (if necessary).

kiwichris, 2025-02-05:
@mdavidsaver thank for you raising this.

Can I assume this is for RTEMS 5 and 6 systems?

LibBSD lets you use `/etc/rc.conf` to manage and configure network interfaces. The benefit of this approach is the reduction of code EPICS needs to manage networks and the ability to harden production system vs test systems without the need to rebuild the code.

The `rc.conf` approach is one I added a number of years ago written in C. It attempts to copy the syntax of FreeBSD so most of the existing FreeBSD documentation works. With suitable hardware you can have VLAN, bridges, specialized interface configurations and more. I would welcome additions for things packet filtering to provide localized firewalling.

kiwichris, 2025-02-06:
RTEMS now provides [LINKER_SETS](https://docs.rtems.org/docs/main/c-user/linker_sets.html) and the [SYSINIT](https://docs.rtems.org/docs/main/c-user/initialization/operations.html) interface to provide managed sequenced initialization in and around RTEMS.

This system is modeled on the FreeBSD system and similar to the constructor approach that exists in EPICS.

kiwichris, 2025-02-06:
> > Can I assume this is for RTEMS 5 and 6 systems?
> 
> Yes. `posix/rtems_init.c` implies RTEMS >= 5. (`score/rtems_init.c` is use for RTEMS 4.9/.10)

What about support for the legacy stack?

If it is needed (and I think it still is) a simplified `rc.conf` implementation could be added to it to help smooth over EPICS?
 
> > LibBSD lets you use /etc/rc.conf to manage and configure network interfaces ...
> > ... most of the existing FreeBSD documentation works ...
> 
> I am certainly all of reuse of existing mechanisms. Please assume I know nothing more about FreeBSD other than the name. 

Understood and it not assumed. The way to view this is RTEMS uses the FreeBSD networking sources and this approach aligns with it plus we do not have to maintain documentation or examples.

> I take it you refer to [`man rc.conf`](https://man.freebsd.org/cgi/man.cgi?rc.conf)? 

Yes.

> Can you quote/link some RTEMS specific example?

This is from a box I am working on at the moment. It is a Microzed dual core Zynq box:
```
rtems [/] # cat /etc/rc.conf
# Build config
hostname="spu2"
ifconfig_cgem0="DHCP rxcsum txcsum"
ifconfig_cgem0_alias0="ether 72:44:fa:b0:29:da"
dhcpcd_priority="200"
dhcpcd_options="--debug --nobackground --timeout 30"
```
And a KR260 (AMD K26) on the same network and application where the 2nd interface is being used:
```
 [/] # cat /etc/rc.conf  
# Build config
hostname="spu1"
ifconfig_cgem0="NOAUTO DHCP rxcsum txcsum"
ifconfig_cgem1="DHCP rxcsum txcsum"
ifconfig_cgem1_alias0="ether 00:0a:35:15:23:72"
dhcpcd_priority="200"
dhcpcd_options="--debug --nobackground --timeout 30 cgem1"
```

The same code runs and initializes both system yet they are very different boards.

mdavidsaver, 2025-02-06:
> What about support for the legacy stack?

Bothering about legacy and compatibility?  Isn't that supposed to be my job ;)

> If it is needed (and I think it still is) a simplified `rc.conf` implementation could be added to it to help smooth over EPICS?

Hopefully not.  I would wish this difference to be shoveled into a `/etc/boot.cmd` script, with a default contents selected per BSP.

kiwichris, 2025-02-07:
> > If it is needed (and I think it still is) a simplified `rc.conf` implementation could be added to it to help smooth over EPICS?
> 
> Hopefully not. I would wish this difference to be shoveled into a `/etc/boot.cmd` script, with a default contents selected per BSP.

Interesting. What would a `boot.cmd` look like? I ask as the legacy code support in EPICS is directly linked to globals in the legacy stack which are wired deep into the networking code. I am suggesting we step away from that and to a single piece of EPICS code that uses configuration files.

An example is https://github.com/epics-base/epics-base/pull/584 that resolves a compile problem by including a header in LibBSD that may be removed because there are no references it in LibBSD. I would be surprised and interested is an application linked with this code and LibBSD.

mdavidsaver, 2025-02-17:
> Interesting. What would a `boot.cmd` look like?

Hopefully quite similar to what would be found in the `/etc/rc.local` of an embedded Linux host.  Or a u-boot script.  eg.

```sh
ifconfig eth0 192.168.1.1/24
ntpdate 192.168.1.4
mount -t nfs 192.168.1.5:/iocs/myioc /ioc
```

> ... single piece of EPICS code that uses configuration files.

I think first of a command script since each IOC instance will already have such a script.

Strictly speaking, I think configuration files would be sufficient as long as there is a reasonable mechanism to provide per-instance configuration files.  eg. it should not be necessary to rebuild epics-base for each IOC executable.

As a convenience, I am thinking along the lines of templates to reduce boilerplate.  So the above might become a template for a site using static IP assignment might be:

```sh
ifconfig ${PRIMARY_IFACE} ${PRIMARY_IP4}
ntpdate 192.168.1.4
mount -t nfs 192.168.1.5:/iocs/myioc /ioc
```

Where `PRIMARY_IFACE` would have a BSP specific default value, and `PRIMARY_IP4` a value read from some local non-volatile storage.

For more complex configuration files, I think it makes sense to leverage the existing text processing tools in Base.  So eg. a shell command to expand macros into files.  eg. to expand an NTPd config to substitute in an IP address coming in from a DHCP response, and stored in an environment variable.

```sh
msi /etc/ntpd.conf.template /etc/ntpd.conf
```

anjohnson, 2025-02-17:
At APS we want to be able to set the target's IP address and st.cmd path in its nvram, preferably reading from the u-boot environment variables or MotLoad equivalents as the RTEMS-4.x code allowed us to do. If our techs need to replace a VME CPU board at 2am to bring the machine back up we don't want to have to wake up IT to change the IOC's MAC address or anything else in our DHCP/BOOTP database, nor should they to have to edit or rebuild anything in the IOC's boot directory.

I think @mdavidsaver's `${PRIMARY_IP4}` variable approach should allow this as long as the BSP gives access to the nvram environment and the `rtems_init.c` code imports those variables into the RTEMS global environment.

kiwichris, 2025-03-01:
> I think [@mdavidsaver](https://github.com/mdavidsaver)'s `${PRIMARY_IP4}` variable approach should allow this as long as the BSP gives access to the nvram environment and the `rtems_init.c` code imports those variables into the RTEMS global environment.

Exporting suitable data as environment variables is a good idea. I can see some issues, for example how addresses discovered by DHCP are handled so there will be some limitations?

I can see three parts being used to provide the solution:

1. A module of code to take the data from the various places that currently exists, for example u-boot parameters, EEPROM and NVM memory. The discovered data can then be used to create the environment variables. The current code is using a now obsolete header in RTEMS to set global variables present in the legacy stack. This cannot be made to work beyond the legacy stack.
2. A module of code and may be EPICS commands to take the environment variables to configure the networking stack built for the BSP.
3. A new `init` code module to join the pieces together.

Controlling the network stacks is a little more complex than the simplified Legacy stack approach Eric put together. The process of stack initialization is asynchronous and the time required can vary based on hardware, static vs dynamic addresses and local network conditions. It is not easy to directly control and set up the LibBSD stack. You have to use special sockets and a lot of complex C code. RTEMS avoided this problem by porting and maintained a few shell commands from FreeBSD to do this. However these commands are FreeBSD ones so different to Linux and Windows. The `rc.conf` solution in RTEMS simply takes the standard `rc.conf` format files and maps them to FreeBSD commands. @mdavidsaver rightly raised the point he does not know FreeBSD or the `rc.conf` standard and this will also be the case for a lot of EPICS users on RTEMS so direct access to the command is not better than `rc.conf`. We could provide a set of standard set ups for a lot of common cases that map to the network stack being used. This could be done in EPICS commands and scripted and the scripting can use environment variables created as part of the data capture process as command arguments. I also think there should be the ability to copy in to `/etc` a `rc.conf` file for those users who need specialized configurations not covered by the standard set.

This raises a few questions:

1. Should EPICS wait in `init` for a user specified period of time until a nonimated interface has media and a valid network address?
2. What would the list of standard configurations be?

The approach would provide support for the Legacy and LibBSD stack easing the move to LibBSD.

At a technical software level:
1. Could `init` be written in C++? I ask as @mdavidsaver's comment in the NTPD PR about C string code is valid and it has me wondering if C++ could be used to make the string handling simpler and easier to review? If c++17 is used the `std::filesystem` support could be used.
2. There is a priority mapping issue being discussed in RTEMS. I have documented the problem here https://users.rtems.org/t/scheduler-maximum-regression-in-rtems-6/78. We need to be careful when handling RTEMS priorities with EPICS mapped POSIX priorities. Mixing priority ranges can have some issues. 

I have some time available to work on this task. If we can agree on what is required I can start. I am so available to meet if this helps progress this task in a timely manner.

mdavidsaver, 2025-03-13:
> It is not easy to directly control and set up the LibBSD stack

Yup...  Thankfully, the scope of this decreases as more BSPs are successfully/completely ported.  I'm more open to "kludge" solutions in such cases where there is a clearly defined point at which they can be removed.

> However these commands are FreeBSD ones so different to Linux and Windows.

imo. this distinction is minor.  And I suspect the reality is that there will be enough RTEMS specific differences to make the whole process distinct.

> [@mdavidsaver](https://github.com/mdavidsaver) rightly raised the point he does not know FreeBSD or the `rc.conf` standard and this will also be the case for a lot of EPICS users on RTEMS so direct access to the commands is no better than `rc.conf`

This is not the message I wanted you to take away.  Having to learn esoteric configuration file formats is a known hazard of any sysadmin type role.  And I consider that a scriptable command set may well be better than a single config file.

> 1. Should EPICS wait in `init` for a user specified period of time until a nonimated interface has media and a valid network address?

Must be scriptable.

> 2\. What would the list of standard configurations be?

Must be scriptable.

> 1. Could `init` be written in C++?

Must be... you see where I am going  ;)

I have no objections to c++.  I write a lot of it.  Although Base currently tops out at c++98 until we can drop vxWorks and RTEMS 4.x.

Still, I would discourage you from jumping right in to writing a lot of c/c++ code.  One of our goals being to minimize the amount of code which epics-base developers need to maintain.

I think that strategic design of IOC/RTEMS shell commands could go a long way when combined with some build system machinery to assemble a ramfs image.

mdavidsaver, 2025-03-13:
> `rc.conf`

Actually.  The one concern I have with what seems like one Great Big Config file is how to handling patching only pieces of it.  I imagine that it would turn into something like:

```
hostname="$(HOSTNAME)"
ifconfig_$(PRIMARY_NIC)="$(IFCONFIG_$(PRIMARY_NIC))"
ifconfig_$(PRIMARY_NIC)_alias0="ether $(MAC_$(PRIMARY_NIC))"
```

which imo. starts to look like a sort of [print-and-scan](https://thedailywtf.com/articles/web_0_0x2e_1) situation.  aka. being a moderately inefficient intermediate format.

Still, something like the usual sort of `/etc/rc.conf.d/` mechanism would avoid the worst of this.