epics-base/epics-base #601: Archor links in dbdToHtml.pl

mdavidsaver, 2025-02-11:
Currently `dbdToHtml.pl` renders section headings like:

```html
<h3><a class='u'
name="Monitor-Parameters"
>Monitor Parameters</a></h3>
```

It would be handy if this archor tag also had `href="#Monitor-Parameters"` so that I could quickly select/copy the link URL with the ID.

My perl is weak, so I do not know if this change would be trivial, or practically impossible.  I do suspect that it would be one or the other.

anjohnson, 2025-02-27:
Our `EPICS::PodXHtml` module has apparently been broken for a while, but we haven't noticed because the `dbdToHtml.pl` script falls back to using `EPICS::PodHtml` in that case (earlier versions of Perl didn't come with both). `EPICS::PodHtml` derives from Perl's `Pod::Simple::Html` module, and that doesn't let us modify the output generated for section headings, but `EPICS::PodXHtml` derives from the Perl `Pod::Simple::XHTML` module, and that _can_ be wrangled to generate what you want.

Try PR #612

anjohnson, 2025-06-11:
Fixed by #612.

closed 2025-06-11 by anjohnson