epics-base/epics-base #602: multiline strings in iocsh files

hinxx, 2025-02-14:
It would be a good feature to have support for multiline strings in st.cmd / iocsh scripts.

Currently I noticed that when iocsh reads a script it does not like the line to be terminated with `\` (backslash). If it were able to concatenate the lines that end with backslash it would help the readability of very long lines in startup files.
Then this would be possible:

```
dbLoadRecords "db/dbSubExample.db",\
      "user=ahoj,  \
foo=bar ,\
  unused=hahhaha,    \
more=macros,\
  andmore=....,\
var=$(VAR1)\
"
```

Would something like this be acceptable to introduce or are there known issues and reasons why not have it?

anjohnson, 2025-02-14:
I'm not aware of any specific problem with doing this, but I haven't tried or looked very far into it. There is a fixed limit `MAC_SIZE` in macLib.h to the size of both macro names and their value strings, but they don't directly impact the length of command-lines. Including newlines in macro strings means that the `"\\\n"` combination might need checking for both when reading it in from readline and after the macros in the line are expanded.

Any addition like this would need a good set of tests to ensure that we catch all the corner cases we can think of, but I don't have a fundamental objection to the idea myself.

Is this something that you are volunteering to work on, or are you just making the suggestion for us to add to a list of Codeathon projects?

hinxx, 2025-02-14:
I've been hacking a part of code that reads a line now and stops at `\n`. I changed it such that `\` followed by `\n` gets ignored and the next line is just added to the previous one forming a longer single line. As a result the example above gets properly handled as if it were a single line. It works even if the word is split in half (silly I know), and the the `\` as an escape char in the string is still handled as before. 

I agree that there might be lots of corner cases to worry about and test. It might be worth specifying where and when exactly the multiline string would be acceptable; i.e. I feel like it is not needed for interactive mode, while it would serve well for files.

I can open up a PR with what I have for a start and we take it from there..

mdavidsaver, 2025-02-15:
> ... There is a fixed limit `MAC_SIZE` in macLib.h  ...

Can we can agree not to allow macros to be split across lines? 

 This would allow concatenation after macro expansion to reach arbitrary lengths.

```
# error
epicsEnvSet FOO "$(SOME\
NAME)"

# ok
epicsEnvSet FOO "$(SOME) \
$(OTHER)"
```

hinxx, 2025-02-19:
> It sounds like @hinxx initial implementation doesn't add the backspace or newline chars into the input buffer though, so I think your error example would be parsed as epicsEnvSet FOO "$(SOMENAME)".

Correct.

My current modifications are bound to `osdReadline()` to essentially detect the `\` and then in case it is followed by the `\n` just ignore both and continue to parse the next line of input into the line buffer. Expectation is that this will not have any impact on the current way of handling of the escaped chars. Also, any length limits from macro expansion currently imposed onto the single line would be in effect with the "concatenated" line.

We could enforce a `<space>\\n` char sequence to be the only valid sequence to avoid folks from splitting words as well as macro names and values where not desired.

I'll cleanup the change and submit the PR in the coming days with what I have so far.

closed 2025-06-20 by anjohnson