epics-base/epics-base #605: Ubuntu 24.04 / gcc-14 Werror build fails in pvaClientChannel.cpp

ralphlange, 2025-02-20:
In my fix-gha-config branch that bumps the Ubuntu builds to 22.04 / 24.04, the Werror build using Ubunto 24.04 / gcc-14 fails at a real warning in pvaClientChannel:

```c++
  ../pvaClientChannel.cpp: In member function ‘virtual void epics::pvaClient::PvaClientChannel::channelCreated(const epics::pvData::Status&, const epics::pvAccess::Channel::shared_pointer&)’:
  make[4]: Entering directory '/home/runner/work/epics-base/epics-base/modules/pvaClient/src/O.linux-x86_64'
  ../pvaClientChannel.cpp:182:59: error: ignoring return value of ‘constexpr std::__cxx11::basic_string<_CharT, _Traits, _Allocator> std::operator+(__cxx11::basic_string<_CharT, _Traits, _Allocator>&&, const _CharT*) [with _CharT = char; _Traits = char_traits<char>; _Alloc = allocator<char>]’, declared with attribute ‘nodiscard’ [-Werror=unused-result]
  g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE        -fdiagnostics-color -fdiagnostics-color -fstack-protector-strong -Wformat -Werror -Werror=format-security -Wno-error=deprecated-declarations -Wno-error=stringop-truncation -Wno-error=restrict -Wno-error=sizeof-pointer-memaccess -Wno-error=nonnull -Wno-error=dangling-pointer -Wno-error=format-overflow -Wno-error=stringop-overread -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -D_X86_64_ -DUNIX  -Dlinux      -O3 -g   -Wall     -std=c++20 -mtune=generic     -m64  -fPIC -I. -I../O.Common -I. -I. -I.. -I/home/runner/work/epics-base/epics-base/include/compiler/gcc -I/home/runner/work/epics-base/epics-base/include/os/Linux -I/home/runner/work/epics-base/epics-base/include -I/home/runner/work/epics-base/epics-base/include/compiler/gcc -I/home/runner/work/epics-base/epics-base/include/os/Linux -I/home/runner/work/epics-base/epics-base/include        -c ../pvaClientChannel.cpp
    182 |             + " status " +  status.getMessage() + " why??";
        |                                                           ^
  In file included from /usr/include/c++/13/string:54,
                   from /usr/include/c++/13/bits/locale_classes.h:40,
                   from /usr/include/c++/13/bits/ios_base.h:41,
                   from /usr/include/c++/13/ios:44,
                   from /usr/include/c++/13/ostream:40,
                   from /usr/include/c++/13/bits/unique_ptr.h:42,
                   from /usr/include/c++/13/memory:78,
                   from /home/runner/work/epics-base/epics-base/include/pv/event.h:12,
                   from ../pvaClientChannel.cpp:13:
  /usr/include/c++/13/bits/basic_string.h:3688:5: note: declared here
   3688 |     operator+(basic_string<_CharT, _Traits, _Alloc>&& __lhs,
        |     ^~~~~~~~
  cc1plus: all warnings being treated as errors
  make[4]: *** [/home/runner/work/epics-base/epics-base/configure/RULES_BUILD:268: pvaClientChannel.o] Error 1
```

https://github.com/ralphlange/epics-base/actions/runs/13394180716/job/37408907260#step:6:9599

Do you have an idea what's going wrong there? Some operator+() overloading magic fails.
To say it with the source code where the werror appears: "why??"

anjohnson, 2025-02-20:
```C++
    if(!status.isOK()) {
        string message("PvaClientChannel::channelCreated");
            + " status " +  status.getMessage() + " why??";
        throw std::runtime_error(message);
    }
```

There is a `;` at the end of the second line above, so the line after it isn't appending to that message string. This looks to be a possible fix:

```diff
diff --git a/src/pvaClientChannel.cpp b/src/pvaClientChannel.cpp
index c3f7e97..8e6e9f9 100644
--- a/src/pvaClientChannel.cpp
+++ b/src/pvaClientChannel.cpp
@@ -179,7 +179,7 @@ void PvaClientChannel::channelCreated(const Status& status, Channel::shared_poin
     }
     if(!status.isOK()) {
         string message("PvaClientChannel::channelCreated");
-            + " status " +  status.getMessage() + " why??";
+        message += " status " +  status.getMessage() + " why??";
         throw std::runtime_error(message);
     }
 }

```

mdavidsaver, 2025-02-20:
Any objections to transferring this issue to https://github.com/epics-base/pvaClientCPP ?

Also, wow.  I didn't realize that unary `+` was a thing with c/c++

anjohnson, 2025-02-20:
No objections, that's where it should go.